mirko/SonOTA #1: Sonoff Original Firmware

folny, 2017-05-29:
Hi

Is it possible to download the original firmware for sonoff with these links ?.

http://52.28.103.75:8088/ota/rom/xpiAOwgVUJaRMqFkRBsoI4AVtnozgwp1/user1.1024.new.2.bin
http://52.28.103.75:8088/ota/rom/xpiAOwgVUJaRMqFkRBsoI4AVtnozgwp1/user2.1024.new.2.bin

mirko, 2017-05-29:
The device adds some CGI params to get URL before issuing the GET request.
Unfortunately I don't have
a) the pcap anymore but just kept the downloaded images
b) any device running a firmware <1.5.5 for which I could once again initiate and sniff an upgrade

folny, 2017-05-29:
Thank you for your reply, please do not know where can i find the original firmware for sonoff device ?.

jalmeroth, 2017-05-31:
You will need the following HTTP-Headers to download the files:

> 
deviceid: "10000xxxxx"
ts: timestamp
sign: signed Request
Host: "dl.itead.cn"
User-Agent: "itead-device"

folny, 2017-06-01:
Hi

I have read some pages as it works well it is described in the attached link https://wiki.almeroth.com/doku.php?id=projects:sonoff but I could not correctly build the link for the download firmware please can you write me the correct wording http line to get the firmware put downloaded thank you

jalmeroth, 2017-06-01:
You can not construct a link for this. It requires the usage of a tool like curl or a script, where you can define HTTP-Headers. Also, the sign-parameter is based on your deviceid. You can re-use it to download other firmware as well, but I will not share my credentials. :)

mirko, 2017-06-01:
> You can not construct a link for this

Actually I think I did so. If I remember right I used a longish URL containing the the CGI params you mentioned and just set it up as a GET request.

The original firmware also contains the according string:

`GET %s?deviceid=%s&ts=%s&sign=%s HTTP/1.1`

mirko, 2017-06-01:
`http://52.28.103.75:8088/ota/rom/xpiAOwgVUJaRMqFkRBsoI4AVtnozgwp1/user1.1024.new.2.bin?deviceid=10000XXXXX&ts=DDDDDDDDD&sign=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX` works for me

jalmeroth, 2017-06-01:
You're right, it seems this changed since my investigation. 👍 

@mirko: Do you have an idea what the sign-parameter consists of? It is slightly different from what the app does:
```python
    def signature(self, message, secret=PROD_SECRET):
        logger.debug(message)
        message = bytes(message.encode('utf-8'))
        secret = bytes(secret.encode('utf-8'))

        signature = base64.b64encode(
            hmac.new(secret, message, digestmod=hashlib.sha256).digest())

        return signature.decode("utf-8")
```

mirko, 2017-06-01:
No idea, I just quick-checked if it's the sha256sum of the image (which is provided to the device beforehand via its WebSocket connection) but it doesn't seem to be.

folny, 2017-06-01:
Thank you, Mirko

I do not know where the problem is but when I opened a link so downloading the file does not work, so I want to ask or you would not get it and did not add bin files to your Github I would be grateful to you.

mirko, 2017-06-01:
I'm pretty sure I'm not allowed to redistribute those binary files.
On top I'm not really keen of having random 3rd binary files in my GitHub repos.
Also the file is device type specific (in this case "Sonoff Basic") and might be even generated on the fly taking my deviceid into account (although unlikely).
I'll leave the issue open, especially for figuring out how the sign parameter is generated.

folny, 2017-06-01:
Okay so if you could please send me the files to my email ? emil.schvarz@gmail.com

sillyfrog, 2017-10-06:
I know this is old, but I'm trying to restore the firmware so I can do testing of some updates to SonOTA (without having to buy more Sonoff's).

From my initial debugging I already had a backup of my `user1.1024.new.2.bin` and `user2.1024.new.2.bin`. I tried restoring them using the following command (and a bunch of other variants, including `user2.1024.new.2.bin` at `0x81000`, and different versions of `boot_*`):
```
esptool.py --port /dev/cu.usbserial-A50285BI write_flash -fs 8m 0x00000 bin/boot_v1.4\(b1\).bin 0x01000 ../user1.1024.new.2.bin 0xFB000 bin/blank.bin 0xFC000 bin/esp_init_data_default.bin 0xFE000 bin/blank.bin
```

Many combinations give me something such that the LED blinks 5 times, with a brief pause, 5 blinks etc. (This is from the end of https://wiki.almeroth.com/doku.php?id=projects:sonoff )

I have seen this with a stock device, and simply holding down the bottom puts it into AP mode (sometimes you need to hold it for 7 seconds a 2nd time). But with this after flashing, that's all I get 5 blinks, pause, 5 blinks.

If anyone happens to have any further ideas, please let me know and I'll give it a crack.

Cheers.

sillyfrog, 2017-10-06:
After further searching, the solution appears to be right under my nose at https://github.com/khcnz/Espressif2Arduino#flashing--running

Specifically, to backup:
```
esptool.py -p COM5 read_flash 0x0 0x100000 backup-0x00000.bin (note this is slow ~2 mins)
```

And to restore:
```
esptool.py -p COM5 -b 921600 write_flash 0x0 backup-0x00000.bin
```

Unfortunately I have already flashed all my devices, so can't do the initial backup, if someone does have a full backup they are willing to share with me, please let me know.

khcnz, 2017-10-09:
Sorry - the original firmware is locked to each device (there is a mac address check!). See details on the original tasmota thread (linked to from the sonta readme)

Geving, 2018-04-11:
I have two Sonoff Basic-units with the good ol' 1.5 firmware. I've dumped them both, and there are only one or two places where they differ when compared with a hex-editor. Their SSID is one of the things that are different. I suspect that, given some more time to look at it, we should be able to figure out what values need to be changed to make it work on a different device. Personally, I suspect that it's only the MAC and maybe a byte for checksum.

I also have ONE older TH-10, but no other device to dump and compare with.
As for newer devices, I have a S20, TH-16 and the POW (or whatever it's called). These are all with the newer "useless" firmware, and I'd LOVE to get my hands on an older version.

Based on what I've seen in the dumps, they do contain the wifi settings that the device is configured with, so that might be something to think about before distribution.

usamer, 2018-05-17:
First of all thank you very much for your efforts creating this very useful repository for "upgrading" Sonoff's over the air!

I have successfully flashed two basic Sonoff's with 1.55. My question is, if I buy some new devices they most likely will have >= 1.6 firmware version and there is no chance to downgrade the firmware to a lower version because it is bound to the individual mac address, right? They only way to upgrade those will be the hardware flashing method?

Many thanks in advance, just wanted to get the most recent statement on this.

Geving, 2018-05-18:
Yes, until someone is willing to do the work needed to figure out how to edit the firmwaredumps so that the MAC can be corrected before flashing, you're out of luck. I was thinking about doing that, but ended up flashing Tasmota firmware instead. Can't really say I regret that...

sillyfrog, 2018-05-18:
@usamer The issue now is, even if you have the firmware, there is no way in the app to tell it to downgrade to it (because if we could, we would just use that method to install SonOTA). So the only way is to flash it using serial, so if doing that, it's easiest to just flash Tasmota right away and not even worry about this :(

closed 2018-05-18 by sillyfrog

gsalvati, 2018-05-25:
Anyone could write another device original firmware into any other device? I have the same problem. 1 Sonoff with custom firmware without backup. I want to get back to Sonoff ecosystem.

Geving, 2018-05-27:
@gsalvati It's not the flashing part that's hard, but it's the content. Imagine that the firmware has some code like this at its begining:

```
addressA="string_hard_coded_in_FW_file"
addressB=getMACasStringFromChip()

if(addressA != addressB){
    while true{
        //Do nada!
    }
}
// Continue the boot sequence...
```
This means that any firmware you copy from one device to another, won't work (as seen from the users point of view).
It'll flash ok, but it won't boot as you expect it to.
Until someone can be bothered to figure out where the MAC is stored in the dumped firmware binaries (PS! It's very easy to find with a HEX editor...) and what is needed to alter it (I suspect a checksum must be recalculated or something?), there's no going back if you didn't backup your device before flashing it the first time.

Personally, I don't suspect it's too much work, but I also believe that most people capable of doing this will be the same people who would want to use a custom FW anyway. This results in very few people who actually have any incentive to give it a go.

0n3-70uch, 2019-02-18:
Hi, I wanna flash my sonoff back to stock firmware. But it seems impossible... The problem: I have no backup of the original firmware. 

For comparison, I have backed up three sonoff devices. Someone interested in reverse engineering?
Here is a image of the difference between the three firmware images:
[image]

Geving, 2019-02-18:
> Hi, I want to flash my devices back to the stock firmware. But it seems impossible... The problem: I have no backup of the original firmware.
> 
> For comparison, I have backed up three sonoff devices. Someone interested in reverse engineering?

Great start! Have you compared the varying data against the MAC-addresses of the devices?

0n3-70uch, 2019-02-18:
> Great start! Have you compared the varying data against the MAC-addresses of the devices?
Hi, yes. But cannot find a correlation. There are two more bytes with different content. But I think this is something like running time or boot count.

Here are some more data. From left to right (in the picture):
Sonoff Basic / 100053ff6d / cc:50:e3:02:dd:a3
Sonoff Basic / 100053ff07 / 80:7d:3a:32:50:0f
Sonoff Basic / 100053ff7b / 80:7d:3a:32:0b:c0

Link to the firmware dump:
https://drive.google.com/open?id=13RISTVmZCSaoJbWspJB8Ogy4vusFbpra

0n3-70uch, 2019-02-18:
PS: I can dump another 7 devices.

Hucksleyy91, 2019-02-22:
I’m interested in putting oem firmware from a sonoff th10/th16 to a sonoff sv :) as I want to install it in a 12v environment :)

kueblc, 2019-02-25:
> Hi, I want to flash my devices back to the stock firmware. But it seems impossible... The problem: I have no backup of the original firmware.
> 
> For comparison, I have backed up three sonoff devices. Someone interested in reverse engineering?
> Here is a image of the difference between the three stock firmware images:
> [image]

Have you considered the `flashChipId` as returned by [`ESP.getFlashChipId`](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#esp-specific-apis)?

> PS: I can dump another 7 devices. Would that be helpful?

Yes

0n3-70uch, 2019-02-26:
> > Hi, I want to flash my devices back to the stock firmware. But it seems impossible... The problem: I have no backup of the original firmware.
> > For comparison, I have backed up three sonoff devices. Someone interested in reverse engineering?
> > Here is a image of the difference between the three stock firmware images:
> > [image]
> 
> Have you considered the `chipId` and `flashChipId` as returned by [`ESP.getChipId` and `ESP.getFlashChipId`](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#esp-specific-apis)?
> 
> > PS: I can dump another 7 devices. Would that be helpful?
> 
> Yes

Hi, ok. I will check the ChipID next weekend... Then I also dump the other seven devices.

Hucksleyy91, 2019-04-07:
Hey any further finding on this :) 
In this video you can see ITEAD doing the sneaky using the sonoff TH firmware on a sonoff basic

0n3-70uch, 2019-04-08:
Hi, no sry... I flashed Tasmota. After some trouble with the official ITEAD server (disconnected and so on).

ledfreaker, 2019-08-01:
any progress?

programmer131, 2020-01-08:
> Hi, I want to flash my devices back to the stock firmware. But it seems impossible... The problem: I have no backup of the original firmware.
>
> For comparison, I have backed up three sonoff devices. [...]

share binary, i've loaded sonoff binary to nodemcu, it connects to wifi and able to control from ewelink app over LAN, but issue with server connection, maybe some ID issue, can you please share binaries?

CopterTec, 2020-01-08:
> share binary, i've loaded sonoff binary to nodemcu, it connects to wifi and able to control from ewelink app over LAN, but issue with server connection, maybe some ID issue, can you please share [...]

Hi, I deleted the files... sry.

0n3-70uch, 2020-01-08:
> share binary, i've loaded sonoff binary to nodemcu, it connects to wifi and able to control from ewelink app over LAN, but issue with server connection, maybe some ID issue, can you please share [...]

Hi, I deleted the files... sry.

Shamshala, 2020-01-08:
> share binary, i've loaded sonoff binary to nodemcu, it connects to wifi and able to control from ewelink app over LAN, but issue with server connection, maybe some ID issue, can you please share binaries?

Have you used binary from other Sonoff device or did you try to edit it? I wonder if the the highlighted sequence isn't just identifier which is logged to server once it tries to connect (bind to MAC address of the device).

0n3-70uch, 2020-01-08:
> > share binary, i've loaded sonoff binary to nodemcu, it connects to wifi and able to control from ewelink app over LAN, but issue with server connection, maybe some ID issue, can you please share binaries?
> 
> Have you used binary from other Sonoff device or did you try to edit it? I wonder if the the highlighted sequence isn't just identifier which is logged to server once it tries to connect (binded to MAC address of the device).

I have not done anything. Just power on and dump binary. No setup process, no wifi connection, no button usage. This may be a boot counter or something else.

programmer131, 2020-01-08:
> Have you used binary from other Sonoff device or did you try to edit it? I wonder if the the highlighted sequence isn't just identifier which is logged to server once it tries to connect (binded to MAC address of the device).

no editing, just uploaded one i got from internet. attached two LEDs and there is already a push button.

joiboi, 2020-07-01:
> `http://52.28.103.75:8088/ota/rom/xpiAOwgVUJaRMqFkRBsoI4AVtnozgwp1/user1.1024.new.2.bin?deviceid=10000XXXXX&ts=DDDDDDDDD&sign=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX` works for me

with this , did you also send any parameter for authentication? , i got my sign with dev id and ts, just shows as 401 error

programmer131, 2020-07-09:
here are sonoff binaries that i tested with nodemcu, i can use eWeLink app in LAN mode.
https://github.com/programmer131/sonoff_binaries
demonstration:
https://youtu.be/X_TUyoaxV3E

FrancYescO, 2020-07-09:
@programmer131 have you patched somehow the original bin?