mirko/SonOTA #2: Script improvement

ionciubotaru, 2017-06-01:
For users with less experience please add a script for raspberry pi 3:

1. start wifi as AP with SSID sonota and password 12345678
2.start python server on port 443 and download a compiled image of the latest tasmota version

The only configuration the user has to do is to select the AP and enter the password, and finally has a running version of Tasmota software

Or, better, provide a complete sd image for raspberry pi 3

mirko, 2017-06-01:
Well, it's certainly possible and I'm happy to see that.

However using the script has a few implications:
 - you still need to build different images for particular devices (mainly the Sonoff 4CH switch as well as the Sonoff Touch, as they have the ESP8285 instead of the ESP8266 built-in)
 - you might brick your device and therefore should be able to unbrick / (re)flash it via serial

Making it too easy I kinda fear that people who are not aware of the above shouldn't use that script anyway (at least as long as it's just a PoC and there a risks of (soft-)bricking).

Though, as said, you're more than welcome of course to do so!

ionciubotaru, 2017-06-01:
As I know @arendst solve the differences between 8266 and 8285 so you can use the same .bin for both
Unfortunately I am not so experienced in linux, that's why I request your help with a ready made raspberry image or, at leas,t all in one script.

You did a great job, keep up the good work

roondar, 2017-06-01:
Why did you close the issue ? You could tag it as an enhancement ? 
May be this weekend, I'will provide a docker image/virtualbox or other thing.

mirko, 2017-06-01:
> Why did you close the issue ? You could tag it as an enhancement ?
I'm sorry, I indeed only considered the issues being issues.
Will follow your advise and tag it appropriately.

roondar, 2017-06-01:
Very fast ! :-)
It'll be easiest  to do the job with latest Tasmota release.

mirko, 2017-06-03:
Short update: I pushed quite a few code changes yesterday, incl. provisioning support which obsoletes using the proprietary app and the DNS redirect. Also the necessity as running it as root is gone.
That way I think the script should be pretty usable for everybody who can install python dependencies and running such a script.

ionciubotaru, 2017-06-03:
Great news,
please upload the latest Tasmota bin1 and bin2, because this is another difficult part.
After first update is it possible to use the normal Tasmota web update procedure ?

mirko, 2017-06-03:
The Tasmota images can be built according to the instructions stated the Tasmota wiki with the adjustment mentioned in my blog post (as well as in the README of this very project).

ionciubotaru, 2017-06-03:
Hi @mirko 

You did a great job, you are just one step before complete - add a nice bash script to do all in one step:

```
#!/bin/bash
git clone https://github.com/mirko/SonOTA
#check if ssl certificate exist and create them if not
.............................
python3 sonota.py --wifi-ssid foobar --wifi-password ew4Ookie 10.23.42.5

```

For you is very easy to compile Tasmota's, but I'm sure that 50% of the users has problems finding boards.txt 
That's why @smadds serve already compiled bin files http://sonoff.maddox.co.uk/tasmota/sonoff.bin 
There is a link to your git in tasmota's wiki page https://github.com/arendst/Sonoff-Tasmota/wiki/Upload#sonoff-factory-ota-mechanism-experimental , so the easyer will be, the more users will benefit from it.

Sorry for my insistence

mirko, 2017-06-13:
If you have such a script, find it worth being added to the repo, add some proper documentation from where to fetch what kind of pre-built images, I'm more than happy to take a look at respective pull-requests.

roondar, 2017-06-19:
Hi Mirko, 

Do you know if ssl  still necessary  ?

mirko, 2017-06-19:
> Do you know if ssl still necessary ?

Could you please elaborate or provide some context?

roondar, 2017-06-19:
Yes, as you are using a different port than 443.
I suppose we can use a http url and not https.
So, if it's only http, openssl won't be needed ?

mirko, 2017-06-20:
It is HTTPS - thus HTTP over SSL/TLS.

roondar, 2017-06-20:
It s in order to deliver a bash script. 
I suposed https not needed anymore.

mirko, 2017-06-20:
The device is initiating the communication by sending an HTTP*S* request. I wonder where your suppositions come from.
Anyway, I'll close this ticket - feel free to propose a pull request for that enhancement.

closed 2017-06-20 by mirko