mirko/SonOTA #7: Build script

lgnd33, 2017-06-27:
(this might be cross-project enhancement) 
It would be really great if there would be a platfrom.io built target added to sonoff-tasmota project that would build images for SonOTA (with modified irom0_0_seg) , or provide a standalone build script that would build the images from source and put them to right place. 
I'm still struggling to do this even manually, so automated help would be more than welcome...

mirko, 2017-06-27:
There is no need for such a custom tasmota build anymore, as - thanks to @khcnz - we went for intermediate images[1].
This project includes respective build scripts you asked for.

https://github.com/khcnz/Espressif2Arduino

closed 2017-06-27 by mirko

khcnz, 2017-06-27:
Yes @ignd33 if you don't use the E2A codebase above, the bootloader won't be replaced, which means although you can build and install tasmota (or any other Arduino based build), you will not be able to do OTA upgrades. 

I'd recommend building the 2 E2A images following the guide on that repo - once you do that you can just install a stock sonoff image without needing any other build changes.

Note i haven't made platform.io instructions/configs yet for E2A - but should be pretty straight forwards by looking at the changes I have in the arduino folder. If you could post changes needed for platform.io i can contribute these to the repo.