mirko/SonOTA #8: Current Status of this solution?

BenWoodford, 2017-07-04:
Before I start fiddling with something that controls a 240V line, I thought it may be best to double check first. I'm not really a hardware kind of guy so soldering a header on isn't my idea of a fun afternoon, so this OTA solution is right up my street.

What's the current situation? Is it possible to flash a Sonoff OTA now? I just picked up two (possibly not iTad authentic, but hey, they were £10 for 2!) and I'm now looking to flash them as I didn't do enough research and didn't realise I'd need to flash them to get them working with Home Assistant. I just need to flash the MQTT-friendly firmware so that I can start using them with HASS really.

palitu, 2017-07-05:
I just tried it last night... didnt work for me, though i probably did it
incorrectly.

I ended up just flashing them with the esptool.py, and used a header just
pushed in (not soldered) whislt erasing the flash and updating the firmware.

blackrozes, 2017-07-05:
Perhaps someone who successfully used the script, could provide a datailed manual for people like me, which are only 50%-nerds. And more familiar with frontend development than flashing images things like that. That would be great!

lgnd33, 2017-07-05:
I'm trying to come to a sucessful deployment end-to-end (espessif2arduino +
sonfOTA + sonoff-tasmota) and am recording all the steps in a document.. If
and once I'm successul, I'll give it to Mirko so he can put it on SonOTA
Wiki, and I'll also send it to Sonoff-Tasmota guys, for E2E reference.
If anyone is faster, please do the same, to save us all some time ;)

blackrozes, 2017-07-05:
♥♥♥

BenWoodford, 2017-07-05:
Would be great to have, this certainly isn't my forte so a nice guide on getting it up and running so I can use it with HASS without having to solder would be great.

mirko, 2017-07-06:
> What's the current situation? Is it possible to flash a Sonoff OTA now?

Well, I'd say it's working :) I also updated the README a couple of days ago hoping it clarifies a few things.
Nevertheless I indeed see it's not a step-by-step guide, so I'm quite looking forward to someone providing such. Thanks for all your comments and contribution!

  mirko

cawede, 2017-07-06:
Hello Mirko, all,

I am quite new into that Arduino and Sonoff world. I took the following approach to get a S20 flashed:
1. compile two tasmata images with the OTA ROM1 and OTA ROM2 settings
2. used the esptool to convert the elf output format, resulting bin files start EA
3. copied the two bin files into the static below the place where sonota.py is located, renamed them to image_user1-0x01000.bin and image_user2-0x81000.bin respectively
4. started the script, connected to ITEAD SSID, provisioning is done, going back to home network and then there is a communication sequence (see below) going on between with the host and the S20 (I made some numbers unreadable to anonymize the trace)
5. there is an error 404 reported in the update sequence

-> any hint what could went wrong here?

Thx, 

Carsten

```
>> {
    "action": "update",
    "deviceid": „*********“,
    "apikey": „******************“,
    "userAgent": "app",
    "sequence": "1499288220698",
    "ts": 0,
    "params": {
        "switch": "off"
    },
    "from": "hackepeter"
}
>> {
    "action": "upgrade",
    "deviceid": „**********“,
    "apikey": „****************“,
    "userAgent": "app",
    "sequence": "1499288220714",
    "ts": 0,
    "params": {
        "binList": [
            {
                "downloadUrl": "http://192.168.67.40:8080/ota/image_user1-0x01000.bin",
                "digest": „**********“,
                "name": "user1.bin"
            },
            {
                "downloadUrl": "http://192.168.67.40:8080/ota/image_user2-0x81000.bin",
                "digest": „*************“,
                "name": "user2.bin"
            }
        ],
        "model": "PSA-B01-GL",
        "version": "23.42.5"
    }
}
2017-07-05 22:57:01,189 (DEBUG) << WEBSOCKET INPUT
2017-07-05 22:57:01,190 (DEBUG) << {
    "error": 0,
    "userAgent": "device",
    "apikey": „****************“,
    "deviceid": „*******“,
    "sequence": "1499288220698"
}
2017-07-05 22:57:01,190 (DEBUG) ~~~ device acknowledged our action request (seq 1499288220698) with error code 0
2017-07-05 22:57:01,647 (INFO) 200 GET /ota/image_user2-0x81000.bin?deviceid=************&ts=1270216262&sign=**************** (192.168.67.70) 12.30ms
2017-07-05 22:57:01,648 (DEBUG) << WEBSOCKET INPUT
2017-07-05 22:57:01,648 (DEBUG) << {
    "error": 404,
    "userAgent": "device",
    "apikey": „****************“,
    "deviceid": „****************“,
    "sequence": "1499288220714"
}
2017-07-05 22:57:01,648 (DEBUG) ~~~ device acknowledged our action request (seq 1499288220714) with error code 404
2017-07-05 22:57:53,119 (DEBUG) << WEBSOCKET INPUT
2017-07-05 22:57:53,120 (DEBUG) << {
    "userAgent": "device",
    "apikey": „********“,
    "deviceid": „************“,
    "action": "update",
    "params": {
        "switch": "off",
        "fwVersion": "1.5.2",
        "rssi": -38,
        "startup": "off"
    }
}

```

mirko, 2017-07-06:
Please do not use (modified) 3rd party Arduino projects  (e.g. Tasmota) being flashed via the Sonoff OTA mechanism.
Although it works (should at least), there will be no way of using the ArduinoOTA mechanism from upon there.
Use images built from https://github.com/khcnz/Espressif2Arduino project as intermediate ones, to replace the bootloader.
Either way, the way you're doing it flashing *should* work.
Can you provide me the image(s) you built so I can take a look, maybe there's sth. obviously wrong with them.

BenWoodford, 2017-07-06:
Dunno about everyone else, but all I'm after is a solder-free way to add Home Assistant support to my Sonoffs. Is that doable with this method?

mirko, 2017-07-06:
Home Assistant is totally unrelated to getting custom Arduino sketches onto the Sonoffs via its original OTA method.

BenWoodford, 2017-07-06:
More specifically I need a firmware with MQTT support on them, but after your previous comment now I'm not sure if that's possible using this method?

lgnd33, 2017-07-06:
It is possible - my use-case is the same, but this is only a part of the
puzzle.

I had luck of creating a PlatformIO environment for building OTA1 and OTA2
images with Espressif2Arduino project. I also added Mirko's script as
custom OTA loader, but am stuck with part of it, so I'm still debugging.
After this is done, Sonoff-Tasmota will be next logical (and pretty
strait-forward) step, after which you "just" have to configure it for
HASS...

I'm looking to automate all of this as I have 20+ sonoffs in my plan, so
doing it manually is painful (and less fun ;)
Stay tuned...

mirko, 2017-07-06:
There's plenty of information about which Arduino project supports which hardware and feature.
This project is to get 4rd party Arduino firmware onto Sonoff devices without opening them up and flashing via serial.

mirko, 2017-07-06:
I'd like to reference https://github.com/mirko/SonOTA/issues/10 where two guys had the same / similar issues which they eventually solved.

sillyfrog, 2017-09-23:
The current build now includes pre-built binaries so it should be a lot simpler to run. With that in mind, I'm closing this, please open a new ticket if you have any issues.

closed 2017-09-23 by sillyfrog