mirko/SonOTA #9: SSL Error when device is connecting to sonota webserver

kokhoor, 2017-07-05:
I have created a dummy cert earlier, using:

`openssl req -x509 -nodes -newkey rsa:2048 -keyout ssl/server.key -out ssl/server.crt -days 42235`

But when SONOFF SV tries to connect, I see SSL error as below:

~~ Waiting for device to connect
2017-07-05 13:14:02,654 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1748)
2017-07-05 13:14:11,660 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1748)
2017-07-05 13:14:45,717 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1748)
2017-07-05 13:14:49,306 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1748)
2017-07-05 13:15:17,768 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1748)

mirko, 2017-07-06:
To be honest I've no idea. Maybe asking on some tornado support channel? Could it be your sonoff has a different firmware (version) which doesn't use SSL/TLS at all? Just shooting in the dark...

kokhoor, 2017-07-06:
Hi mirko, thanks for response. Somehow the error happens on my OSX Sierra. I tried running it on Windows and it worked!

closed 2017-07-06 by kokhoor

iMarkus, 2017-09-28:
Hi,

I had the same problem. After installing the newest Python version 3.6.2 it worked for Mac OS High Sierra!

Cheers