mirko/SonOTA #10: Device answer: error 404

GhsHun, 2017-07-06:
Hi! 

First: Sorry for my english ;)
I create a user1 and a user2 image from source _eagle.flash.1m0.ld_ and copy them to the folder static/.
After that I run the script. It is what I see:

```
>> {
    "action": "upgrade",
    "deviceid": "100006dcbd",
    "apikey": "83e0dc72-4d5f-42bb-bef9-43871e386b90",
    "userAgent": "app",
    "sequence": "1499327137029",
    "ts": 0,
    "params": {
        "binList": [
            {
                "downloadUrl": "http://192.168.75.12:8080/ota/image_user1-0x01000.bin",
                "digest": "4f6b31fa2ca1a29d602b37040826ed62e9e357739cd0703c78cd12bdd3bcc5e9",
                "name": "user1.bin"
            },
            {
                "downloadUrl": "http://192.168.75.12:8080/ota/image_user2-0x81000.bin",
                "digest": "aaaa866e421d0eb844b8a6d0d5eef20cbda29cd307e69aac8b697c270cf34581",
                "name": "user2.bin"
            }
        ],
        "model": "ITA-GZ1-GL",
        "version": "23.42.5"
    }
}
2017-07-06 09:45:37,032 (DEBUG) << WEBSOCKET INPUT
2017-07-06 09:45:37,034 (DEBUG) << {
    "userAgent": "device",
    "apikey": "83e0dc72-4d5f-42bb-bef9-43871e386b90",
    "deviceid": "100006dcbd",
    "action": "query",
    "params": [
        "timers"
    ]
}
~~~ device sent action request,  acknowledging / answering...
~~~~ query
>> {
    "error": 0,
    "deviceid": "100006dcbd",
    "apikey": "83e0dc72-4d5f-42bb-bef9-43871e386b90",
    "params": 0
}
2017-07-06 09:45:37,951 (INFO) 206 GET /ota/image_user1-0x01000.bin?deviceid=100006dcbd&ts=1812158119&sign=5c9377401acaa5f49c2f65c7c539bf9e1e4d48902df3caada8a5bb4d2d852ce8 (192.168.
75.189) 1.00ms
2017-07-06 09:45:37,962 (DEBUG) << WEBSOCKET INPUT
2017-07-06 09:45:37,962 (DEBUG) << {
    "error": 404,
    "userAgent": "device",
    "apikey": "83e0dc72-4d5f-42bb-bef9-43871e386b90",
    "deviceid": "100006dcbd",
    "sequence": "1499327137029"
}
2017-07-06 09:45:37,962 (DEBUG) ~~~ device acknowledged our action request (seq 1499327137029) with error code 404
```

So, the question is: Is this correct to use the _eagle.flash.1m0.ld_ as the source of user1 and user2 image or should I use the _eagle.flash.1m64.ld_ file instead?

mirko, 2017-07-06:
Can you provide me the image(s) you built so I can take a look, maybe there's sth. obviously wrong with them.

GhsHun, 2017-07-06:
Here they are:
[image_user1-0x01000.bin.txt](https://github.com/mirko/SonOTA/files/1127423/image_user1-0x01000.bin.txt)
[image_user2-0x81000.bin.txt](https://github.com/mirko/SonOTA/files/1127427/image_user2-0x81000.bin.txt)

Please delete the .txt extension from the end of the files.

mirko, 2017-07-06:
Looks good to me - does it also happen with images build from the Espressif2Arduino sketch?
Maybe the file is too big?

GhsHun, 2017-07-06:
Ok. I used the original Tasmota files. I will replace them and try again.

GhsHun, 2017-07-06:
So, I check the linker script in the Espressif2Arduino sketch, and both are the same that I used. That's not the problem.

mirko, 2017-07-06:
There's still the potential issue of your images being too big, that's why I asked explicitly about trying it with Espressif2Arduino builds.

GhsHun, 2017-07-06:
Ok, I built an image width Espressif2Arduino linker script and its size is the same as before. 479.312 bite. 
I changed the WiFi settings in the the user_config.h and disabled the IR_REMOTE. Nothing else was modified. Should I change anything else?

cawede, 2017-07-06:
I think it is meant to build two Espressif2Arduino binaries (just compile from this sketch) with the respective linker scripts out of that project. You need to set URL_ROM_1 and 2 according to your needs. The resulting image size is in my build 259379 Bytes. Need to see this evening if they are flashing once I am close to my S20 again.

mirko, 2017-07-06:
> Ok, I built an image width Espressif2Arduino linker script and its size is the same as before. 479.312 bite.

This sounds far too much for a single Espressif2Arduino image.

GhsHun, 2017-07-06:
In this thread: [https://github.com/arendst/Sonoff-Tasmota/issues/476](url) the images are almost same size like mine:
[image]

My Arduino settings:
[image]

What is wrong?

mirko, 2017-07-06:
What images are you talking about?
I'm talking about those being created by the Espressif2Arduino Sketch, *which you're supposed to use by SonOTA*.

mihalski, 2017-07-06:
In that thread I am also having problems so I wouldn't go by my image sizes. I will be trying Espressif2Arduino now to see if that sheds any light on our problem.

GhsHun, 2017-07-06:
I want to flash Tasmota fw with SonOTA. It isn't possible? I use sonoff skatch. I downloaded it from [https://github.com/arendst/Sonoff-Tasmota](url)

mirko, 2017-07-06:
People, please read the README.
Way is: original ITEAD firmware -> intermediate Espressif2Arduino firmware (which on first run replaces the bootloader) -> non-modified Arduino images (e.g. Tasmota)

mihalski, 2017-07-06:
Well it's not working so I'd try the suggestion given to do it differently.. Trying to get it done now.

GhsHun, 2017-07-06:
Ok, I understand it. I didn't read it carefully, sorry!

mihalski, 2017-07-06:
I've not made any real progress.. But I've just noticed something that might be important.. esptool is being used on .elf images.. I would swear (but am not certain) that I have been doing so on .bin images.. If I have, that could be the problem.

GhsHun, 2017-07-06:
> I've not made any real progress.. But I've just noticed something that might be important.. esptool is being used on .elf images.. I would swear (but am not certain) that I have been doing so on .bin images.. If I have, that could be the problem.

Change the Build output folder in the Airduino. Watch this:
[https://www.kanda.com/blog/microcontrollers/avr-microcontrollers/find-arduino-hex-files-output-binaries/](url)

After that the elf files will be there in the folder.

mihalski, 2017-07-06:
I've got the images:
```
ls -l Espressif2Arduino.ino-*
-rw-r--r--  1 michal  staff  238304  6 Jul 22:22 Espressif2Arduino.ino-0x01000.bin
-rw-r--r--  1 michal  staff  238304  6 Jul 22:22 Espressif2Arduino.ino-0x81000.bin
```
I presume now I use Sonata as I did previously? Let's see how it goes.

GhsHun, 2017-07-06:
I have a question:
I read Espressif2Arduino readme. There it is described:
[image]
Where? In the Espressif2Arduino.ino file? In this file I found only this:

[image]
So, where is the URL_ROM_1?

mirko, 2017-07-06:
>  So, where is the URL_ROM_1?
There is none?! Rom 1 is fetched and flashed by the *original* Sonoff FW.

mihalski, 2017-07-06:
You wanna know what? It works!
I feel a strong urge to build my first ever docker container to automate this..

This is a triumph!
Thank you @mirko

mirko, 2017-07-06:
Congrats! Glad to hear that - happy about any contribution!

mihalski, 2017-07-06:
I do have one question though.. Is building Espressif2Arduino.ino-0x01000.bin totally redundant? The original Sonoff image is always at that location and thus only Espressif2Arduino.ino-0x81000.bin (and a TASMOTA or similar image) is actually required?

mirko, 2017-07-06:
> I do have one question though.. Is building Espressif2Arduino.ino-0x01000.bin totally redundant? The original Sonoff image is always at that location and thus only Espressif2Arduino.ino-0x81000.bin (and a TASMOTA or similar image) is actually required?

I'm not sure I get your question: The Sonoff original FW might either ask for, fetch and flash a user_1 or user_2 image - we can't control which one the Sonoff asks for. So we need both.
As a second (or, depending on whether the Sonoff asked for the user_1-image, the third) step Espressif2Arduino flashes the bootloader to the Arduino one to be compatible with the Arduino OTA mechanism.
As a third (fourth respectively) step the custom image (e.g. Tasmota) is fetched and flashed.

mihalski, 2017-07-06:
I think I erroneously made an assumption or two. Thanks for the step-by-step as it will be helpful when I try to design a way to automate this.

There is a disconnect between the direct flashing described in the Espressif2Arduino documentation and what SonOTA is trying to achieve that can be somewhat confusing.

GhsHun, 2017-07-06:
Ok, I flash the Espressif2Arduino fw.
```
2017-07-06 15:24:51,420 (INFO) 206 GET /ota/image_user1-0x01000.bin?deviceid=100006dcbd&ts=451230256&sign=ac4cb79830934433cd468dd32969de1a612633a493772349ed1db813b084914d (192.168.7
5.189) 1.50ms
2017-07-06 15:24:51,768 (DEBUG) << WEBSOCKET INPUT
2017-07-06 15:24:51,768 (DEBUG) << {
    "error": 0,
    "userAgent": "device",
    "apikey": "ab24d420-48b1-4f99-9b7f-2d90f222970e",
    "deviceid": "100006dcbd",
    "sequence": "1499347481720"
}
2017-07-06 15:24:51,769 (DEBUG) ~~~ device acknowledged our action request (seq 1499347481720) with error code 0

```

And what's the next step? My device led lightning continuously and nothing is happening.

GhsHun, 2017-07-06:
Ok. I'm stupid. It's WORKING!

Thank you @mirko

mirko, 2017-07-06:
Since all issues (and hopefully headaches) seem to be solved/gone by now, the documentation obviously is not complete / is misleading / you name it.
As from developer perspective things naturally seem clearer than they do to users, I'd highly appreciate if you could sum up the pitfalls and issues you encountered, maybe even coming up with a step-by-step tutorial which you think might clear things up for further folks.

mihalski, 2017-07-07:
I'm trying to get this to work with a new Sonoff Pow (one of my seems to have died this evening) but it doesn't seem to be joining the network after provisioning has been done:
** No ip address of the ITEAD DHCP range (10.10.7.0/24) is assigned to any of your interfaces, which means you don't appear to be connected to the IEAD WiFi network.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX)
** This application can be kept running.
.........~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10000b80c9",
    "apikey": "bae4de2b-52ad-42e5-ac3e-0ad9ce8558c7",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "SSID",
    "password": "password",
    "serverName": "192.168.1.41",
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
~~ Starting web server
~~ Waiting for device to connect
```
Just sits there waiting and I never see the Sonoff connect to my Wifi network.

Any ideas?

mihalski, 2017-07-07:
Used --legacy with root to get further:
```
~~ Waiting for device to connect
<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "error": 0,
    "reason": "ok",
    "IP": "192.168.1.41",
    "port": 443
}
2017-07-07 19:35:51,184 (INFO) 200 POST /dispatch/device (192.168.1.133) 1.75ms
2017-07-07 19:35:51,192 (INFO) 101 GET /api/ws (192.168.1.133) 0.30ms
2017-07-07 19:35:51,192 (DEBUG) << WEBSOCKET OPEN
2017-07-07 19:35:51,195 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:35:51,195 (DEBUG) << {
    "userAgent": "device",
    "apikey": "bae4de2b-52ad-42e5-ac3e-0ad9ce8558c7",
    "deviceid": "10000b80c9",
    "action": "register",
    "version": 2,
    "romVersion": "2.0.2",
    "model": "PSC-B01-GL",
    "ts": 844
}
~~~ device sent action request,  acknowledging / answering...
2017-07-07 19:35:51,195 (INFO) We are dealing with a PSC-B01-GL model.
~~~~ register
>> {
    "error": 0,
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "config": {
        "hb": 1,
        "hbInterval": 145
    }
}
2017-07-07 19:35:51,198 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:35:51,198 (DEBUG) << {
    "userAgent": "device",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "deviceid": "10000b80c9",
    "action": "date"
}
~~~ device sent action request,  acknowledging / answering...
~~~~ date
>> {
    "error": 0,
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "date": "2017-07-07T19:35:51.198Z"
}
2017-07-07 19:35:51,201 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:35:51,201 (DEBUG) << {
    "userAgent": "device",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "deviceid": "10000b80c9",
    "action": "update",
    "params": {
        "switch": "off",
        "power": "0",
        "fwVersion": "2.0.2",
        "rssi": -59,
        "staMac": "D0:27:00:16:FE:E2"
    }
}
~~~ device sent action request,  acknowledging / answering...
~~~~ update
>> {
    "error": 0,
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1"
}
>> {
    "action": "update",
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "userAgent": "app",
    "sequence": "1499420151201",
    "ts": 0,
    "params": {
        "switch": "off"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "userAgent": "app",
    "sequence": "1499420151201",
    "ts": 0,
    "params": {
        "switch": "on"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "userAgent": "app",
    "sequence": "1499420151201",
    "ts": 0,
    "params": {
        "switch": "off"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "userAgent": "app",
    "sequence": "1499420151201",
    "ts": 0,
    "params": {
        "switch": "on"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "userAgent": "app",
    "sequence": "1499420151202",
    "ts": 0,
    "params": {
        "switch": "off"
    },
    "from": "hackepeter"
}
>> {
    "action": "upgrade",
    "deviceid": "10000b80c9",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "userAgent": "app",
    "sequence": "1499420151205",
    "ts": 0,
    "params": {
        "binList": [
            {
                "downloadUrl": "http://192.168.1.41:8080/ota/Espressif2Arduino.ino-0x01000.bin",
                "digest": "a37b6962bc851e678a0774723bc9330a44633ae7ed4242140389d0ba6bb719f9",
                "name": "user1.bin"
            },
            {
                "downloadUrl": "http://192.168.1.41:8080/ota/Espressif2Arduino.ino-0x81000.bin",
                "digest": "745503eedcdb5bbf6bcd798636d42cfbd8a0c7ca626439e615c7801e169cde4e",
                "name": "user2.bin"
            }
        ],
        "model": "PSC-B01-GL",
        "version": "23.42.5"
    }
}
2017-07-07 19:35:51,209 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:35:51,209 (DEBUG) << {
    "error": 0,
    "userAgent": "device",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "deviceid": "10000b80c9",
    "sequence": "1499420151201"
}
2017-07-07 19:35:51,209 (DEBUG) ~~~ device acknowledged our action request (seq 1499420151201) with error code 0
2017-07-07 19:35:51,228 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:35:51,229 (DEBUG) << {
    "error": 404,
    "userAgent": "device",
    "apikey": "00791e2d-9903-45ce-902a-8b1e5fb622f1",
    "deviceid": "10000b80c9",
    "sequence": "1499420151205"
}
2017-07-07 19:35:51,229 (DEBUG) ~~~ device acknowledged our action request (seq 1499420151205) with error code 404
2017-07-07 19:35:51,231 (INFO) 200 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000b80c9&ts=686078705&sign=9a4e7053d10ad39ec0059201cf958fa925d19f9453a0432deebdf5ed6efe1531 (192.168.1.133) 11.69ms
```
These are the same firmware files I previously had success with on a basic Sonoff.
The files are available at the URL's in the debug. As mentioned by @mirko earlier this could be something to do with it not liking checksums?

mihalski, 2017-07-07:
Tried another Sonoff Pow. This time I started with legacy mode:
```
** No ip address of the ITEAD DHCP range (10.10.7.0/24) is assigned to any of your interfaces, which means you don't appear to be connected to the IEAD WiFi network.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX)
** This application can be kept running.
................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10000b8188",
    "apikey": "6b4189ec-834e-401f-9b8f-203f499bf87c",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "SSID",
    "password": "password",
    "serverName": "192.168.1.41",
    "port": 443
}
<< {
    "error": 0
}
~~ Provisioning completed
~~ Starting web server
~~ Waiting for device to connect
<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "error": 0,
    "reason": "ok",
    "IP": "192.168.1.41",
    "port": 443
}
2017-07-07 19:47:26,304 (INFO) 200 POST /dispatch/device (192.168.1.134) 0.68ms
2017-07-07 19:47:26,316 (INFO) 101 GET /api/ws (192.168.1.134) 0.35ms
2017-07-07 19:47:26,316 (DEBUG) << WEBSOCKET OPEN
2017-07-07 19:47:26,319 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:47:26,319 (DEBUG) << {
    "userAgent": "device",
    "apikey": "6b4189ec-834e-401f-9b8f-203f499bf87c",
    "deviceid": "10000b8188",
    "action": "register",
    "version": 2,
    "romVersion": "2.0.2",
    "model": "PSC-B01-GL",
    "ts": 744
}
~~~ device sent action request,  acknowledging / answering...
2017-07-07 19:47:26,319 (INFO) We are dealing with a PSC-B01-GL model.
~~~~ register
>> {
    "error": 0,
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "config": {
        "hb": 1,
        "hbInterval": 145
    }
}
2017-07-07 19:47:26,322 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:47:26,322 (DEBUG) << {
    "userAgent": "device",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "deviceid": "10000b8188",
    "action": "date"
}
~~~ device sent action request,  acknowledging / answering...
~~~~ date
>> {
    "error": 0,
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "date": "2017-07-07T19:47:26.322Z"
}
2017-07-07 19:47:26,325 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:47:26,325 (DEBUG) << {
    "userAgent": "device",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "deviceid": "10000b8188",
    "action": "update",
    "params": {
        "switch": "off",
        "power": "0",
        "fwVersion": "2.0.2",
        "rssi": -58,
        "staMac": "D0:27:00:17:00:60"
    }
}
~~~ device sent action request,  acknowledging / answering...
~~~~ update
>> {
    "error": 0,
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a"
}
>> {
    "action": "update",
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "userAgent": "app",
    "sequence": "1499420846326",
    "ts": 0,
    "params": {
        "switch": "off"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "userAgent": "app",
    "sequence": "1499420846326",
    "ts": 0,
    "params": {
        "switch": "on"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "userAgent": "app",
    "sequence": "1499420846326",
    "ts": 0,
    "params": {
        "switch": "off"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "userAgent": "app",
    "sequence": "1499420846326",
    "ts": 0,
    "params": {
        "switch": "on"
    },
    "from": "hackepeter"
}
>> {
    "action": "update",
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "userAgent": "app",
    "sequence": "1499420846326",
    "ts": 0,
    "params": {
        "switch": "off"
    },
    "from": "hackepeter"
}
>> {
    "action": "upgrade",
    "deviceid": "10000b8188",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "userAgent": "app",
    "sequence": "1499420846329",
    "ts": 0,
    "params": {
        "binList": [
            {
                "downloadUrl": "http://192.168.1.41:8080/ota/Espressif2Arduino.ino-0x01000.bin",
                "digest": "a37b6962bc851e678a0774723bc9330a44633ae7ed4242140389d0ba6bb719f9",
                "name": "user1.bin"
            },
            {
                "downloadUrl": "http://192.168.1.41:8080/ota/Espressif2Arduino.ino-0x81000.bin",
                "digest": "745503eedcdb5bbf6bcd798636d42cfbd8a0c7ca626439e615c7801e169cde4e",
                "name": "user2.bin"
            }
        ],
        "model": "PSC-B01-GL",
        "version": "23.42.5"
    }
}
2017-07-07 19:47:26,347 (DEBUG) << WEBSOCKET INPUT
2017-07-07 19:47:26,347 (DEBUG) << {
    "error": 404,
    "userAgent": "device",
    "apikey": "c00c6fbf-4a08-4aa2-9a60-9e3c2d06438a",
    "deviceid": "10000b8188",
    "sequence": "1499420846329"
}
2017-07-07 19:47:26,347 (DEBUG) ~~~ device acknowledged our action request (seq 1499420846329) with error code 404
2017-07-07 19:47:26,350 (INFO) 200 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000b8188&ts=1270216262&sign=c95071ef084c8d29a4a61a34ed9e7f345d09a437e48f5b7c1da648cb4fb35d31 (192.168.1.134) 10.95ms
```
I can't see any hints in these as to what could be going wrong.. Hopefully someone else can.

mirko, 2017-07-07:
Are you using builds from the Espressif2Arduino sketch?

mihalski, 2017-07-07:
As the filenames imply, yes.
Like I said.. These are the same builds that worked with the basic Sonoff.

I don't see why the basic Sonoff is happy to take them but the Sonoff Pow is not.

mirko, 2017-07-07:
> I don't see why the basic Sonoff is happy to take them but the Sonoff Pow is not.

Unfortunately, me neither.

mihalski, 2017-07-07:
Good idea.. I will try that next.. Might be a day before I get a chance to act on it..

On the bright side.. Unlike the latest basic Sonoffs which I have been unable to upgrade using an FTDI, the Sonoff Pow's still seem upgradable via FTDI.

mirko, 2017-07-07:
If it works with latest firmware, please open a separate issue-ticket stating that with your (previous) FW version it doesn't work, even when `--legacy` is specified.

wirephoto, 2017-07-15:
@mirko thanks for everything you have done to put this together. This will be an incredible boon to everyone working on DIY home automation.

@mihalski I would be very grateful if you followed through with the docker idea--I've never used docker before, but it looks well-documented. I have successfully flashed S20s via ftdi before, but right now I have a pile of six S20s and 2 Touchs that need to be flashed pretty soon and I don't want to take them apart and solder headers, etc. Unfortunately, I am not technically adept enough to follow the instructions in their current form to get it done OTA.

I'll keep watching. Good luck everyone, and thanks.

mihalski, 2017-07-15:
I hope to get the chance to work on the docker in a few days. I tried to get started on it but I am also new to docker and frankly creating a command line Arduino build environment inside a container is a lot harder than I anticipated.

Hopefully I'll be able to pull it off. I could easily just provide the firmware files to upload along with a modified sonota script, but that wouldn't be as transparent.

wirephoto, 2017-07-16:
I get that the ideal is a transparent setup that can be audited for security, but the truth is, the default firmware installed on the Sonoffs is sketchy enough security-wise (especially since it phones home) that a precompiled alternative is probably safer for most of us, and certainly more convenient.

mihalski, 2017-07-17:
OK. If anyone involved wants me to take this down I will do so immediately.

Here's is the simplest way to get the latest version of Tasmota onto your sonoff.
1. Download, extract, and cd into sonata directory.
2. Set up a Wifi network with the default SSID/password in the Tasmota firmware (i.e. indebuurt1/VnsqrtnrsddbrN). This is the network the default firmware will join. you can then change the Wifi settings via the web ui.
3. Prepare Sonoff by powering it using your FTDI (or other means).
4. On a PC with a wifi interface run "python sonota.py --wifi-ssid <YourWifiSSID> --wifi-password <YourWifiPassword> <YourIPAddress>.
5. The Sonoff light should be flashing if it is powered. Hold down the button until it flashes faster.
6. Once it is flashing quickly, follow the instructions SonOTA is giving you and join the ITEAD wifi network.
7. Wait until the Sonoff is detected, provisioned and rebooted onto the indebuurt1 Wifi network.
8. Find out the IP of the Sonoff via whatever means you have (should be listed somewhere on your router) and configure it via the web ui.

Done.
[sonota.zip](https://github.com/mirko/SonOTA/files/1151805/sonota.zip)

cputoaster, 2017-07-17:
Thanks a lot! I tried it with a new TH10, and am somehow stuck.

1) unzip works (macos). After this step, I have the sonota.py script but empty ssl / static directories
2) created the indebuurt1 wifi as a hotspot on my phone. Also joined my PC to it.
3) plugged it into mains
4) run sonota (see below)
5) hold button until it flashes quickly
6) join ITEAD wifi with my PC -> sonota does the GET / POST thingy (see below)
7) ITEAD disappears (sonoff probably booted)
8) PC joining indebuurt1 again, searching for ip with nmap, find nothing

any idea?

```
shinkansen:sonota ameyer$ python3 ./sonota.py --wifi-ssid cputoasters --wifi-password xxxx .
** No ip address of the ITEAD DHCP range (10.10.7.0/24) is assigned to any of your interfaces, which means you don't appear to be connected to the IEAD WiFi network.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX)
** This application can be kept running.
...............................................................................................................................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "1000080c1f",
    "apikey": "9fee9388-2c52-41d0-88f3-af0d2cd4e0b2",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "cputoasters",
    "password": "xxxxxx",
    "serverName": ".",
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
** The IP address of <serve_host> (.) is not  assigned to any interface on this machine.
** Please change WiFi network to $ESSID and make  sure %s is being assigned to your WiFi interface.
** This application can be kept running.
................................................
```

mihalski, 2017-07-17:
Your IP address can't be "."

Try this from a terminal:
```
ifconfig |grep inet
	inet 127.0.0.1 netmask 0xff000000 
	inet6 ::1 prefixlen 128 
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1 
	inet6 fe80::8b0:a7c:fd23:469b%en0 prefixlen 64 secured scopeid 0x4 
	inet 192.168.1.41 netmask 0xffffff00 broadcast 192.168.1.255
	inet6 fe80::4d:52e0:7149:a7ec%en1 prefixlen 64 secured scopeid 0x5 
	inet 192.168.1.51 netmask 0xffffff00 broadcast 192.168.1.255
	inet6 fe80::a0f7:3eff:fe5a:2e70%awdl0 prefixlen 64 scopeid 0x8 
	inet6 fe80::1f65:af06:21ea:ee36%utun0 prefixlen 64 scopeid 0x9 
	inet6 fe80::f233:9398:3f4:d4f1%utun1 prefixlen 64 scopeid 0xa 
```
For me the address is 192.168.1.41

That's the machine I am running this from and the IP address the Sonoff needs to download the firmware from.

cputoaster, 2017-07-17:
thanks! That got me one step further, now there is an error about the cert not being there. I can create it naturally, but was that not the point of the zip file, to have everything included? How about the flash binaries?

mihalski, 2017-07-17:
Umm.. isn't all that in there? Sec.

cputoaster, 2017-07-17:
thanks, now I get it to here:
```
shinkansen:sonota ameyer$ python3 ./sonota.py --wifi-ssid cputoasters --wifi-password xxxxxx 192.168.43.160
** No ip address of the ITEAD DHCP range (10.10.7.0/24) is assigned to any of your interfaces, which means you don't appear to be connected to the IEAD WiFi network.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX)
** This application can be kept running.
........~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "1000080c1f",
    "apikey": "9fee9388-2c52-41d0-88f3-af0d2cd4e0b2",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "cputoasters",
    "password": "xxxxxxx",
    "serverName": "192.168.43.160",
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
** The IP address of <serve_host> (192.168.43.160) is not  assigned to any interface on this machine.
** Please change WiFi network to $ESSID and make  sure %s is being assigned to your WiFi interface.
** This application can be kept running.
......~~ Starting web server
~~ Waiting for device to connect
```

with
 
```
shinkansen:~ ameyer$ nmap 192.168.43.1-255 -p 80

Starting Nmap 7.50 ( https://nmap.org ) at 2017-07-17 11:41 CEST
Nmap scan report for shinkansen (192.168.43.160)
Host is up (0.00036s latency).

PORT   STATE  SERVICE
80/tcp closed http

Nmap done: 255 IP addresses (1 host up) scanned in 22.86 seconds
shinkansen:~ ameyer$ ifconfig | grep en0 -C 5
	inet6 ::1 prefixlen 128
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1
	nd6 options=201<PERFORMNUD,DAD>
gif0: flags=8010<POINTOPOINT,MULTICAST> mtu 1280
stf0: flags=0<> mtu 1280
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	ether f4:0f:24:35:f2:06
--
	ether f4:0f:24:35:f2:06
	inet6 fe80::1c46:4ca4:ff69:bc6c%en0 prefixlen 64 secured scopeid 0x4
	inet 192.168.43.160 netmask 0xffffff00 broadcast 192.168.43.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
en1: flags=963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX> mtu 1500
shinkansen:~ ameyer$
```

mihalski, 2017-07-17:
Has the device connected to your wifi access point? That's a prerequisite for it to continue to the next step.

cputoaster, 2017-07-17:
I dont really know. In point 7) you say to connect to indebuurt1, which I did with the PC, but nmap does not show me any device on port 80 on the network. Or should I be connecting to my real wifi, "cputoasters"?

mihalski, 2017-07-17:
Can you not set your wifi access point to have a secondary SSID on the same subnet? That would be the easiest solution.

cputoaster, 2017-07-17:
I guess there is no difference if I just have my phone do a hotspot, no? I cannot change the main wifi easily and using guest access (to have a different SSID) would give me another subnet anyway.

cputoaster, 2017-07-17:
btw, when switching to the real wifi instead of indebuur1, I can see the device ip and an open port 80, but connecting with a browser does not do anything. Also, it does not seem to want to connect to the sonota web server (with the address of the real wifi of the PC as the serve_host)

```
...
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
** The IP address of <serve_host> (192.168.86.167) is not  assigned to any interface on this machine.
** Please change WiFi network to $ESSID and make  sure %s is being assigned to your WiFi interface.
** This application can be kept running.
.....~~ Starting web server
~~ Waiting for device to connect
```

mihalski, 2017-07-17:
No, your machine must be accessible to the sonoff from the network it joins or it will fail.

Delete your default wifi network so that it doesn't auto join that (you can put it back later) and set indebuur1 as your default so you go straight back to it after provisioning.

mihalski, 2017-07-17:
You know I am now so confused that I'm not 100% sure I am telling you the correct information :/

I need to take a break to look back over how all this works before I continue..

mihalski, 2017-07-17:
Oh, and if you create and join the indebuur1 wifi network you need to use the SSID/password for that and the ip that you get assigned from that network when running sonota.py.

cputoaster, 2017-07-17:
I think the maddox image is fine, it should use the config params that were set with the first POST call. But somehow the device is not calling back to the web server to get anything. What did you compile into the binaries as URL_ROM_2?

cputoaster, 2017-07-17:
btw, its a TH10, but I think its the same tasmota image.

mihalski, 2017-07-17:
Firstly, like I said at the very beginning and a few times since: this this is only known to work with the Sonoff Basic and DOES NOT work with the Sonoff Pow. If anything the Sonoff TH10 is more like the Pow than the Basic.

And secondly, the Maddox image is the same as the one provided in the Tasmota Github, just always the latest version. To my understanding both the images I have provided and the Maddox image must be available ON THE SAME NETWORK as the Sonoff joins and AT THE SAME TIME. Otherwise it WILL NOT WORK.

The images were compiled by following the Espressif2Arduino and SonOTA instructions.

cputoaster, 2017-07-18:
OK, thanks for the information. I tried compiling myself just to have the "custom" URL_ROM_2 to point to my PC ip address and now have a different problem that I dont think is related to the sonota script. Thank you for your help. I will for now wait for new developments and try again later.

spokkie2015, 2017-07-20:
@mihalski , I followed you method on the basic and this is the output: 

13066226-macair:sonota Barend$ python3 sonota.py --wifi-ssid indebuurt1 --wifi-password VnsqrtnrsddbrN 10.23.42.5
** No ip address of the ITEAD DHCP range (10.10.7.0/24) is assigned to any of your interfaces, which means you don't appear to be connected to the IEAD WiFi network.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX)
** This application can be kept running.
........................................................................................................................................................................................................................................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10000cb362",
    "apikey": "932b4f61-8ed1-4ef9-aa15-f7bf610b62bf",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "indebuurt1",
    "password": "VnsqrtnrsddbrN",
    "serverName": "192.168.8.81",
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
** The IP address of <serve_host> (192.168.8.81) is not  assigned to any interface on this machine.
** Please change WiFi network to $ESSID and make  sure %s is being assigned to your WiFi interface.
** This application can be kept running.
......~~ Starting web server
~~ Waiting for device to connect
<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "error": 0,
    "reason": "ok",
    "IP": "192.168.8.81",
    "port": 4223
}
2017-07-20 18:26:16,959 (INFO) 200 POST /dispatch/device (192.168.8.92) 1.65ms
2017-07-20 18:26:17,982 (INFO) 101 GET /api/ws (192.168.8.92) 0.77ms
2017-07-20 18:26:17,982 (DEBUG) << WEBSOCKET OPEN
2017-07-20 18:26:18,165 (DEBUG) << WEBSOCKET INPUT
2017-07-20 18:26:18,165 (DEBUG) << {
    "userAgent": "device",
    "apikey": "932b4f61-8ed1-4ef9-aa15-f7bf610b62bf",
    "deviceid": "10000cb362",
    "action": "register",
    "version": 2,
    "romVersion": "1.5.5",
    "model": "ITA-GZ1-GL",
    "ts": 263
}
~~~ device sent action request,  acknowledging / answering...
2017-07-20 18:26:18,165 (INFO) We are dealing with a ITA-GZ1-GL model.
~~~~ register
>> {
    "error": 0,
    "deviceid": "10000cb362",
    "apikey": "94333198-b682-4e11-ab6e-f5781f496d18",
    "config": {
        "hb": 1,
        "hbInterval": 145
    }
}
2017-07-20 18:26:19,361 (DEBUG) << WEBSOCKET INPUT
2017-07-20 18:26:19,361 (DEBUG) << {
    "userAgent": "device",
    "apikey": "94333198-b682-4e11-ab6e-f5781f496d18",
    "deviceid": "10000cb362",
    "action": "date"
[... 179 lines cut ...]
2017-07-20 18:26:29,094 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=1856224190&sign=8415a3b2c00d3f68e128cbd95fd874781a68e40d698941d195dca4e777449d0c (192.168.8.92) 1.78ms
2017-07-20 18:26:29,257 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=983059344&sign=70385a65ccc1cb362272f6448046e13b74452db403da5797374fe090948e32cb (192.168.8.92) 1.18ms
2017-07-20 18:26:29,416 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=1718458134&sign=f6904a2e20c7e1eb2a4445c1dcca02c19444008b66e523041520f471215ca7ef (192.168.8.92) 1.17ms
2017-07-20 18:26:29,518 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=1876268425&sign=74fe9f85cc807e82b88b55f779c63384986380c84576c8e5fea3b9980f476841 (192.168.8.92) 1.12ms
2017-07-20 18:26:29,671 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=1764841629&sign=f016c2b5042fabb84a82b2061bf80a179bc454900c98c9a293b9b44c1f3dff4b (192.168.8.92) 1.11ms
2017-07-20 18:26:29,838 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=398844030&sign=3323b8ca59eeddc0f2277593e4fcdcda7bd411fa5175013a23a2eff89dcaf11d (192.168.8.92) 1.20ms
2017-07-20 18:26:30,036 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=185252727&sign=d785df59d47879671a6ed2327a74a729d149197a4dafc02647d7c68eeffcba8b (192.168.8.92) 1.31ms
2017-07-20 18:26:30,174 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=1370429126&sign=7343780e17a9de43d396afd0435cce23fcb0e40e81cbad18d6be0c722ec0a940 (192.168.8.92) 1.11ms
2017-07-20 18:26:30,321 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=502141743&sign=85accd0fff0ef8cdf49105cbd6f8af5cf3f0d7943773021f6a26ddbc33df3906 (192.168.8.92) 1.15ms
2017-07-20 18:26:30,458 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=993687334&sign=4737dc03c73d82fce049a9f138b245cd29cfeacea4ab51947747462854620750 (192.168.8.92) 1.13ms
2017-07-20 18:26:30,632 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=15934104&sign=12fdbb8741e5b499513be45d81ee22a437f7f446f8cdb2a77ec72e755ee56d28 (192.168.8.92) 1.16ms
2017-07-20 18:26:30,734 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000cb362&ts=1363674760&sign=25e839608313df00e569193cf65542a9436edaf0e9f679dd076f662953cfc110 (192.168.8.92) 1.10ms
2017-07-20 18:26:31,100 (DEBUG) << WEBSOCKET INPUT
2017-07-20 18:26:31,101 (DEBUG) << {
    "error": 0,
    "userAgent": "device",
    "apikey": "94333198-b682-4e11-ab6e-f5781f496d18",
    "deviceid": "10000cb362",
    "sequence": "1500567979525"
}
2017-07-20 18:26:31,101 (DEBUG) ~~~ device acknowledged our action request (seq 1500567979525) with error code 0

My issue is that after nearly 30min it just stays like this with the green light on the basic flashing fast. I stopped the script, but the basic does not want to go into pairing mode again. I have done it with two of them now and same results with both. 

Any idea?

mihalski, 2017-07-20:
I would expect that it is now trying to join indebuurt1 with the password VnsqrtnrsddbrN.

I'm curious. How is it that the ip for the host you were running on was 10.23.42.5 and the URL's were http://192.168.8.81:8080/ota/Espressif2Arduino.ino-0x01000.bin and http://192.168.8.81:8080/ota/Espressif2Arduino.ino-0x81000.bin?

Doesn't seem to add up.

Here's my run:
```
python3 sonota.py --wifi-ssid MyWifi --wifi-password mywifipassword 192.168.1.41
** No ip address of the ITEAD DHCP range (10.10.7.0/24) is assigned to any of your interfaces, which means you don't appear to be connected to the IEAD WiFi network.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX)
** This application can be kept running.
...............................................................................................................................................................................................................................................................................................................................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10000bd80b",
    "apikey": "6a2ee380-81f7-4e55-8ace-f7d284f822f9",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "MyWifi",
    "password": "mywifipassword",
    "serverName": "192.168.1.41",
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
~~ Starting web server
~~ Waiting for device to connect
<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "error": 0,
    "reason": "ok",
    "IP": "192.168.1.41",
    "port": 4223
}
2017-07-17 16:54:22,707 (INFO) 200 POST /dispatch/device (192.168.1.139) 2.39ms
2017-07-17 16:54:22,884 (INFO) 101 GET /api/ws (192.168.1.139) 0.52ms
2017-07-17 16:54:22,884 (DEBUG) << WEBSOCKET OPEN
2017-07-17 16:54:22,891 (DEBUG) << WEBSOCKET INPUT
2017-07-17 16:54:22,891 (DEBUG) << {
    "userAgent": "device",
    "apikey": "6a2ee380-81f7-4e55-8ace-f7d284f822f9",
    "deviceid": "10000bd80b",
    "action": "register",
    "version": 2,
    "romVersion": "1.5.5",
    "model": "ITA-GZ1-GL",
    "ts": 744
}
~~~ device sent action request,  acknowledging / answering...
2017-07-17 16:54:22,891 (INFO) We are dealing with a ITA-GZ1-GL model.
~~~~ register
>> {
    "error": 0,
    "deviceid": "10000bd80b",
    "apikey": "2b982951-9bc3-4c37-8734-440eb9efacc6",
    "config": {
        "hb": 1,
        "hbInterval": 145
    }
}
2017-07-17 16:54:22,898 (DEBUG) << WEBSOCKET INPUT
2017-07-17 16:54:22,898 (DEBUG) << {
    "userAgent": "device",
[... 183 lines cut ...]
2017-07-17 16:54:27,685 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=1856224190&sign=22fa0dc90794367c2dfe1af9f179082b8bbeb2381b0396537a7253a55c860cd4 (192.168.1.139) 0.55ms
2017-07-17 16:54:27,789 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=983059344&sign=e742e28d08b3db9c761268740d9bb70dd4b30ed2dd240662896a905a9cb1ce80 (192.168.1.139) 0.55ms
2017-07-17 16:54:27,882 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=1718458134&sign=b24560ed6b6b73df04adf230a0cdff828c2eff48ec3cc1a2243d43e3dff05b0c (192.168.1.139) 0.55ms
2017-07-17 16:54:27,983 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=1876268425&sign=bdfbcb671bd96a37e778475b64e436f81f3e2f86ed82b7c2de92ea4a922d953d (192.168.1.139) 0.56ms
2017-07-17 16:54:28,053 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=1764841629&sign=e7634fec6ee3cf55c3e31af82f23e1252eead1b0474078d83aacd2650fcd2dbc (192.168.1.139) 0.56ms
2017-07-17 16:54:28,147 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=398844030&sign=4fb8de0362920908f76db6efc645d2254c77f0bad745d39baa49ff04458e2687 (192.168.1.139) 0.54ms
2017-07-17 16:54:28,239 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=185252727&sign=26421a5034cb61ff32eb21a0dd33389d2fd9f820d7b28abadfdf3e5dc986153f (192.168.1.139) 0.55ms
2017-07-17 16:54:28,336 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=1370429126&sign=fad20d45cd7e469737cfcfca72742ca13b1d25cef413bd6d417e0f2b7513dfaa (192.168.1.139) 0.56ms
2017-07-17 16:54:28,411 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=502141743&sign=3701ba9ce487d241a504cedcdcaab1129b498a15bedf3d39da4ad7582df4fb03 (192.168.1.139) 0.56ms
2017-07-17 16:54:28,499 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=993687334&sign=95f3a15161c0eef2f104916bf490050e3e2742650e575d6578e6a947be18ce79 (192.168.1.139) 0.56ms
2017-07-17 16:54:28,591 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x81000.bin?deviceid=10000bd80b&ts=15934104&sign=f5124eb297927660226460f8fe7e275522e6372dd7bcd867f1e7f04cee3f34c2 (192.168.1.139) 0.54ms
2017-07-17 16:54:28,926 (DEBUG) << WEBSOCKET INPUT
2017-07-17 16:54:28,927 (DEBUG) << {
    "error": 0,
    "userAgent": "device",
    "apikey": "2b982951-9bc3-4c37-8734-440eb9efacc6",
    "deviceid": "10000bd80b",
    "sequence": "1500274462909"
}
2017-07-17 16:54:28,927 (DEBUG) ~~~ device acknowledged our action request (seq 1500274462909) with error code 0
^C2017-07-17 17:00:17,108 (INFO) Quitting.
```
My wifi network has multiple SSID's so I can access it via the default as well as my own SSID/pass. Another thing that might be worth noting is that my Sonoff's Basics were not configured AT ALL. Never connected to them with the mobile app. Straight out of the box.

I've done 2 of them and still have 2 unconfigured for future testing if I ever figure out how to build a docker container that works.

[7 more replies]

closed 2017-09-23 by sillyfrog