mirko/SonOTA #11: Sonoff Touch

StSimmons, 2017-07-09:
Is there a way to get the Sonoff Touch to broadcast the ITEAD-XXXX network? AFAICT this script requires us to be n that network.

StSimmons, 2017-07-09:
So, I worked out that I could do this using `--legacy` and doing the old DNS redirects.

I successfully got the `Espressif2Arduino` flash working.

I then tried to use the Sonoff-Tasinoda (built locally using PlatformIO and the `sonoff-touch-4ch` config). I hosted it at http://192.168.1.1/sonoff.bin, I saw the sonoff try to download it (by watching the HTTP server logs). Sadly, it no longer boots.

Any ideas what might have gone wrong?

khcnz, 2017-07-10:
Any chance you also had a serial console hooked up at time of flash? or can hook up a serial console now?

khcnz, 2017-07-10:
I think i know what may be wrong - i was testing a slightly different configuration and noticed that sometimes the buffer downloading the new images doesn't fill up - and when it doesn't fill up in multiples of 4 bytes bad things happen (the flash doesn't write correctly). I have a fix in the works but for now you will have to flash it with serial cable sorry.

StSimmons, 2017-07-10:
Is this a bug in the Espressif2Arduino firmware? Or something else?

khcnz, 2017-07-10:
Yes in E2A

jleroux61, 2017-07-24:
I just received my sonoff 4ch. It has firmware v2.01. @StSimmons how did you connect to the device if the ssid is not broadcasting? For which command are you using the --legacy parameter?
I see videos on youtube where people see the ssid on their smart phone so I guess they stopped broadcasting in later releases.

mirko, 2017-07-29:
You could skip the provisioning vy SonOTA - e.g. if it doesn't create an AP named ITEAD-XXX - and do the initial provisioning via the ewlink app.

ageorgios, 2017-09-15:
What are the steps required if the device (the case is a Sonoff B1 Bulb) does not broadcast and ITEAD-* wifi? I provisioned it through ewlink app to my local wifi network. What next?

krokce, 2017-09-20:
On topic - just to confirm that I successfully flashed the "Sonoff Touch" (firmware version 1.6.2.) - smooth and easy. Thanks for the great job you guys!

andyfitter, 2017-09-23:
Any chance of a walkthrough of how you got the ‘touch’ working?

krokce, 2017-09-23:
@andyfitter Very much the same way it works for other Sonoffs. The only "not so ordinary" moment was the way Sonnof Touch behaved when putting it into pairing mode (when the switch starts to broadcast the ITEAD-### SSID). Namely after the first button press of 7 seconds the LED blink pattern was blink-blink-blink-**pause**-blink-blink-blink-**pause**, then you just press the touch button another 7 seconds when the LED blinking pattern changes to blink-blink-blink-blink-blink-blink... and here is when you should be able to see the ITEAD-### WiFi. Hope it helps...
Don't know if its worth mentioning - my (both) Sonoff Touch devices came with firmware 1.6.2 which I believe it's still the last one, and as mentioned - no issue flashing both of them with the current SonOTA build.

krokce, 2017-09-23:
@sillyfrog just want to confirm that this same method above worked for my Sonoff 4CH pro whcih started to advertise the ITEAD-### WiFi only after the second 7 second button press (same LED blinking pattern occurs). So you can put 4CH pro (original firmware version 2.0.1) on the list as confirmed to work with SonOTA - much more elegant way than soldering and grounding pin legs directly on the micro-controller...

sillyfrog, 2017-09-23:
Thanks all. It looks like we have a working combo, so I'll close this for now. Please open again if needed. Cheers.

closed 2017-09-23 by sillyfrog