mirko/SonOTA #12: Client provisioning timesout on windwos

khcnz, 2017-07-11:
When running on windows the POST timeouts. 

I think it relates to the newline \r\n in the JSON from dumps causing problems with parsing on the sonoff end. Additional research to follow - but adding this ticket for anyone else that encounters the same issue.

cowanjt, 2018-05-24:
I'm encountering the same issue. Is there a fix or work around for this?

sillyfrog, 2018-05-25:
This should not be an issue in the current implementation as many many people have used this on Windows since this was an issue. First thing I would suggest checking is ensuring _all_ firewalls are disabled, this is far and away the most common cause of issues - outside the current latest Sonoff firmware no longer allowing this, which is the starting to make this project redundant :(

closed 2018-05-25 by sillyfrog