mirko/SonOTA #14: config and setup questions.

tinstep, 2017-07-22:
Hi 
Im a little stuck on the process to get this installed. Im hoping someone can review my work and let me know this is correct - the Sonoff is connecting to my access point but not getting the bin files - do i have to do ssl?

This is what i have done.

1. I have SonOTA downloaded and extracted and created directory called "static".
2. I have Espressif2Arduino downloaded and extracted.
3. I have esptool installed and running
4. from Espressif2Arduino, I have added the boards and libraries.
5. I compile Espressif2Arduino, after changing "URL_QIO_ROM_2" in the sketch to "http://192.168.0.10/static/Espressif2Arduino.ino-0x01000.bin"
6. Choose "1M (Espressif OTA Rom 1)", in the Arduino menu.
7, Compile in IDE, move to build (/tmp) and Run "esptool.py elf2image --version 2 Espressif2Arduino.ino.elf"
8. once done, copy the file to the static directory in SonOTA.
9. I then repeat from step 6-8, but choosing "1M (Espressif OTA Rom 2)"
10. I now have Espressif2Arduino.ino-0x01000.bin , Espressif2Arduino.ino-0x81000.bin in my static directory.
11. In the root of the SonOTA folder i run 
`./sonota.py --wifi-ssid mywifiaccesspoint --wifi-password 1234567890 192.168.0.10`
12. fire up the sonoff device, connect to it with my phone and give it the wifi credentials for mywifiaccesspoint.

...What happens now? where does this get me and what do i need to do to run tasmota ?

khcnz, 2017-07-23:
Looking pretty good - you also need to set URL 3 in Espressif2Arduino (either use the maddox url or custom build your own version - using 1M spiff 0K as per the tasmota instructiuons

From there it should download and reflash automatically to tasmota.

tinstep, 2017-07-23:
Thanks for the info, I have got a lot further!

1. Added the "http://sonoff.maddox.co.uk/tasmota/sonoff.bin" url to URL_QIO_ROM_3
2. I got some ssl errors, so i created the ssl directory and ran the ssl command from the "read me" and a bunch of things happened, the relay clicked twice and the green light stayed on solidly for a while.

I get an error on the command line saying it is looking for   "(WARNING) [Errno 2] No such file or directory: 'static/image_user1-0x01000.bin'  

Does that mean I should have renamed "Espressif2Arduino.ino-0x81000.bin" to "image_user1-0x01000.bin" ? or are these separate images that are also needed? - added the cli output below.

~~~`
2017-07-23 12:49:34,817 (DEBUG) << WEBSOCKET INPUT
2017-07-23 12:49:34,818 (DEBUG) << {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "action": "update",
    "params": {
        "fwVersion": "1.5.5",
        "switch": "off",
        "staMac": "5C:CF:7F:AF:56:C6",
        "rssi": -72,
        "startup": "off"
    },
    "userAgent": "device"
}
~~~ device sent action request,  acknowledging / answering...
~~~~ update
>> {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "error": 0
}
>> {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "params": {
        "switch": "off"
    },
    "from": "hackepeter",
    "action": "update",
    "ts": 0,
    "sequence": "1500778174818",
    "userAgent": "app"
}
>> {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "params": {
        "switch": "on"
    },
    "from": "hackepeter",
    "action": "update",
    "ts": 0,
    "sequence": "1500778174818",
    "userAgent": "app"
}
>> {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "params": {
        "switch": "off"
    },
    "from": "hackepeter",
    "action": "update",
    "ts": 0,
    "sequence": "1500778174818",
    "userAgent": "app"
}
>> {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "params": {
        "switch": "on"
    },
    "from": "hackepeter",
    "action": "update",
    "ts": 0,
    "sequence": "1500778174819",
    "userAgent": "app"
}
>> {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "params": {
        "switch": "off"
    },
    "from": "hackepeter",
    "action": "update",
    "ts": 0,
    "sequence": "1500778174819",
    "userAgent": "app"
}
2017-07-23 12:49:34,819 (WARNING) [Errno 2] No such file or directory: 'static/image_user1-0x01000.bin'
2017-07-23 12:49:34,819 (WARNING) [Errno 2] No such file or directory: 'static/image_user2-0x81000.bin'
2017-07-23 12:49:34,830 (DEBUG) << WEBSOCKET INPUT
2017-07-23 12:49:34,830 (DEBUG) << {
    "apikey": "1fa2cfc2-915f-4ddc-b23b-cf1738cb31b8",
    "deviceid": "10000ca93d",
    "action": "query",
    "params": [
        "timers"
    ],
    "userAgent": "device"
}
`

khcnz, 2017-07-23:
Yes - either change the file name in the python script 
upgrade_file_user1 = "image_user1-0x01000.bin"
upgrade_file_user2 = "image_user2-0x81000.bin"

or change the name of the build output to match the above. re-run and it should work

tinstep, 2017-07-23:
Great, got further  ... 

now i get ...
cant see it on the network now either, may need to flash using serial - what are your thoughts ?
~~~
2017-07-23 14:25:03,523 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=248501142&sign=754a793fe8ec96421c35baf6a77aca6cf3ba9a9c45c1f4a67845d5ab5d07d458 (192.168.0.142) 1.09ms
2017-07-23 14:25:03,633 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1162346597&sign=d8f520dae19da33e5c8b691ad0f90244c853199dbadbaf2075878af81d854f9e (192.168.0.142) 1.28ms
2017-07-23 14:25:03,736 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1445357088&sign=f98f28af035c11ea5e86dc97b1bde6c17228466fec3c71ab05f7e7710d064841 (192.168.0.142) 1.17ms
2017-07-23 14:25:03,816 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1274341732&sign=3e95fc3283033b2c5715623cf8243771b4f5204ff95112d03d7b25a7363f14f2 (192.168.0.142) 1.12ms
2017-07-23 14:25:03,922 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1475831917&sign=126efee1b70c09401d74bb2801accd9fc3b5f150a1ff00cc1feb10635cc9e96d (192.168.0.142) 1.06ms
2017-07-23 14:25:04,022 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=2033644404&sign=8c8716827d12664e88acd833ea53f944f18f50c164ef4d6a3f3fac26861934a3 (192.168.0.142) 1.11ms
2017-07-23 14:25:04,126 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1860692743&sign=0b999e4336c341d5ffc55f354d4b10e9bcf3ec6c2db5d52119cf5a3ef367ccd0 (192.168.0.142) 1.25ms
2017-07-23 14:25:04,215 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1441680395&sign=69cb8f6d5dfdc036a0971619c89be51410dd4177710fd597ad4624d1484e100d (192.168.0.142) 1.38ms
2017-07-23 14:25:04,321 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1153542749&sign=275a2c1c7872e0aed02067d5964daca45b56d4c9251726bc878b7bddbc26c286 (192.168.0.142) 1.15ms
2017-07-23 14:25:04,415 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1333654407&sign=7a55182f3af1e02c9603af58a69494e030845e78c715cd77e714a5c06e24ead6 (192.168.0.142) 1.22ms
2017-07-23 14:25:04,520 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=383555936&sign=bbb0d5e475aca0e9e9b8fa5642840010f2199ba7894ba6a07056ae4335c70883 (192.168.0.142) 1.08ms
2017-07-23 14:25:04,606 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=10000ca93d&ts=1399374302&sign=417b61a41a14c5d3a09c0d4541f8fc8eada2ecd61930fe9c5767891eef314b38 (192.168.0.142) 1.14ms
2017-07-23 14:25:04,954 (DEBUG) << WEBSOCKET INPUT
2017-07-23 14:25:04,954 (DEBUG) << {
    "deviceid": "10000ca93d",
    "userAgent": "device",
    "apikey": "e01bcd70-8b6f-4a56-8d26-fdacde214b98",
    "error": 0,
    "sequence": "1500783898243"
}
2017-07-23 14:25:04,954 (DEBUG) ~~~ device acknowledged our action request (seq 1500783898243) with error code 0
2017-07-23 14:25:05,841 (WARNING) 404 GET /api?apikey=cbb34105da902ae83efe8bb78d203344&ma_username=nzcarrick&output=json&mode=queue&limit=5 (127.0.0.1) 0.90ms
2017-07-23 14:25:05,844 (WARNING) 404 GET /api?apikey=cbb34105da902ae83efe8bb78d203344&ma_username=nzcarrick&output=json&mode=history&limit=10 (127.0.0.1) 0.87ms
2017-07-23 14:25:06,903 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588605 (127.0.0.1) 0.87ms
2017-07-23 14:25:11,906 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588606 (127.0.0.1) 1.08ms
2017-07-23 14:25:16,905 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588607 (127.0.0.1) 1.08ms
2017-07-23 14:25:20,841 (WARNING) 404 GET /api?apikey=cbb34105da902ae83efe8bb78d203344&ma_username=nzcarrick&output=json&mode=queue&limit=5 (127.0.0.1) 0.75ms
2017-07-23 14:25:20,843 (WARNING) 404 GET /api?apikey=cbb34105da902ae83efe8bb78d203344&ma_username=nzcarrick&output=json&mode=history&limit=10 (127.0.0.1) 1.09ms
2017-07-23 14:25:21,904 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588608 (127.0.0.1) 0.92ms
2017-07-23 14:25:26,903 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588609 (127.0.0.1) 1.13ms
2017-07-23 14:25:31,904 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588610 (127.0.0.1) 0.85ms
2017-07-23 14:25:35,841 (WARNING) 404 GET /api?apikey=cbb34105da902ae83efe8bb78d203344&ma_username=nzcarrick&output=json&mode=queue&limit=5 (127.0.0.1) 0.85ms
2017-07-23 14:25:35,843 (WARNING) 404 GET /api?apikey=cbb34105da902ae83efe8bb78d203344&ma_username=nzcarrick&output=json&mode=history&limit=10 (127.0.0.1) 0.89ms
2017-07-23 14:25:36,903 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588611 (127.0.0.1) 1.13ms
2017-07-23 14:25:41,904 (WARNING) 404 GET /queue/?limit=1&start=0&output=json&apikey=cbb34105da902ae83efe8bb78d203344&_=1500531588612 (127.0.0.1) 1.19ms
~~~

khcnz, 2017-07-23:
What are the 404 requests coming from 127.0.0.1 about? are these unrelated?

Does the device respond a button push?

tinstep, 2017-07-23:
Im not sure where the 127.0.0.1 entries are coming from, I had to stop an unrelated service on port 8080 for this to work and - Im now thinking that systemd restarted that service automatically and buggered everything?

no, the button does not work anymore.

khcnz, 2017-07-24:
It looks like something odd was requesting pages from your python script. This process is still a little fragile so best to do it with a device you can capture the serial output for (until you get the first one working)

tinstep, 2017-08-03:
Yes correct, it was a plug-in in chrome, I've gone through 2 sonoffs, both are the the same state I think I'll wait until someone does a YouTube video, or condenses the instructions to a tutorial, where I don't screw it up. Thanks for your help

sillyfrog, 2017-08-23:
It looks like it has not completed the first step. I have packaged everything up I have done so others can use it here: https://github.com/sillyfrog/SonOTA

This update changes things slightly so it will download the pre-built files straight from your computer (rather than the internet), and I have found it to work will with my devices (a Basic's and a Dual).

PS: I have just put in a PR for this work to this project, but not sure how long it will take to get approved, so linking to my fork for the moment.

TristanMW, 2017-12-30:
Hi.
I just flashed the Sonoff Dual Switch which completed without errors and connected to my ceiling fan. (1 on the fan and 1 on the Light) I am connecting the sonoff via the homekit-node on Node-Red and the problem that I am facing now is that I cant control the fan separate to the light.
Is there a work around? 

When I open the sonoff console you can clearly see which switch is being activated
09:23:11 MQT: stat/%topic%/RESULT = {"POWER1":"OFF"}
09:23:11 MQT: stat/%topic%/POWER1 = OFF
09:23:11 MQT: stat/%topic%/RESULT = {"POWER2":"OFF"}
09:23:11 MQT: stat/%topic%/POWER2 = OFF

sillyfrog, 2018-01-28:
This now sounds like a Sonoff-Tasmota configuration issue - it's important you configure the device type correctly etc. I would suggest reading more at https://github.com/arendst/Sonoff-Tasmota/wiki and/or posting there now you have successfully flashed it.

Good Luck!

closed 2018-01-28 by sillyfrog