mirko/SonOTA #15: Sonoff RF does not connect to WLAN after flashing OTA

mnigbur, 2017-07-25:
So, i wanted to try the OTA flash for a new device, followed the instructions and everything worked like a charm, but the device can't connect to my wireless network. So i hooked it up to the FTDI and am getting this from the serial console:
(SSID and password are masked by purpose)

```
.........Done
Current Rom: 1
              Rom 1 Magic Byte: 0xEA
                                    Flash Mode: 2
                                                 Attemping to read itead Wifi credentials... Done
                        SSID: ****
                                        Password: *******************
                                                                     Connecting to Wifi..................................................................................................

```

It won't boot into firmware download mode via serial as well, so I'm kinda stuck now.
Already tried to change some WiFi settings (channels, security mode..) without any success. It's trying to connect forever.

Shouldn't it connect to "indebuurt1" after the flash anyways? This never happened, the second AP is still configured for this network.

Any Ideas?

mnigbur, 2017-07-25:
last output of the sonota.py:

```
2017-07-25 03:41:04,905 (INFO) 206 GET /ota/Espressif2Arduino.ino-0x01000.bin?deviceid=10000682fd&ts=1363674760&sign=b4097de9129df184c04005bd3b706725604384ceccd71273902ad8ade097072d (192.168.178.29) 2.11ms
2017-07-25 03:41:05,227 (DEBUG) << WEBSOCKET INPUT
2017-07-25 03:41:05,228 (DEBUG) << {
    "deviceid": "10000682fd",
    "error": 0,
    "userAgent": "device",
    "apikey": "530156ac-b622-4eef-a9f0-c97029a9d01f",
    "sequence": "1500979253388"
}
2017-07-25 03:41:05,228 (DEBUG) ~~~ device acknowledged our action request (seq 1500979253388) with error code 0

```

mnigbur, 2017-07-26:
Managed to put it into DFU Mode with shorting R21. Reflashed over serial and it's working fine. I'll try with another module later today.

maddhin, 2017-10-16:
did the OTA flashing with the second module work or is there a general problem with the RF?

Tontze, 2017-10-17:
I did try this same Sonoff RF yesterday and had similar results. It didnt connect anywhere. Im new on this, so i dont know how to debug. Flashing went fine through the process, but it never connected to anywhere.

maddhin, 2017-10-17:
I can actually confirm this. I tried to flash the RF and everything seem to have went fine (finalstage appeared and disappeared) but then the Sonoff RF seems to go in a state that "does nothing" - I was not able to find it being connected to the WLAN nor could I find a AP to connect to. 

Sadly, I closed the sonota.py window. I would recommend to write s log file so one can see if there were errors. I remember seeing some messages which didn't seem clean but everything went through very quick and in the end it said (something like) "success, you can close the window now".... :)

So, I assume it is bricked, or? Grr, that means I DO have to buy a USB TTL adapter...

maddhin, 2017-10-17:
just read the first post again: @Tontze: did you try to set up the  "indebuurt1" wifi and see if it tries to connect to that? That could be a possible scenario to "save" from manual flashing. Will try to test this tonight, too.

sillyfrog, 2017-10-17:
Pressing the button 4 times while the LED is flashing should put to into AP mode. See the home page of this project for more info and links. 

Cheers.

maddhin, 2017-10-17:
getting it into AP mode was the first thing I tried after I could not find it connected to my WLAN (nor seeing a Sonoff or other AP). But it does not work, no AP. I waited forever and used 2 laptops and 1 iPhone to see if there is something popping up :)
I will try the "indebuurt1" wifi later but doubt this will work if the AP is not coming up.

Tontze, 2017-10-17:
Ok, serial logger says its trying to connect to indebuurt1, which does not exist. But im not bale to get it to AP mode either ?

Tontze, 2017-10-17:
I made indebuurt1 hotspot with my phone. Sonoff RF is now connected to it and got ip address. 
 I can now connect to sonoff through my phone :)

maddhin, 2017-10-18:
I can confirm this. Once I set up indebuurt1, the RF connected and I were able to access it. I put in my wifi credentials but it did not connect to my wifi. Whether this is a general problem or due to a dumb mistake like a typo in the wifi password, I do not know. I used the WPS function and the RF connected to my wifi and is now accessible. 
1. the AP function (4 x button) still doesn't seem to work, at least I could not get it to show.
2. the flashing documentation should be revised indicating that a possible solution is to check for connection on the indebuurt1 wifi. In Arend's software it's documented but it is not obvious for the SonOTA as people expect to find the RF connected to the own wifi or AP. 
3. What should I do now? It looks like the flashing was successful but there seems to be something wrong with the wifi. Does it make sense to re-flash (OTA) or something? Or is the flashing 0/1 and reflashing wouldn't change? Or did the bootloader got corrupted or something like that?

Last but not least: many thanks for the OTA option, you saved me a lot of work manually flashing. Super!!!

sillyfrog, 2017-10-18:
If it's booting and connecting to the `indebuurt1` network, then the boot-loader is working because `Sonoff-Tasmota` must be running.

I found that the trick to get it into AP mode was to press the button at the same rate as the flashing LED (it must be done while it flashing), then the next reboot it was in AP mode (remember it's only in AP mode for 60 seconds). I'll try and update the docs to make it more clear when I get a chance (or approve a PR ;)  ) - maybe I should go back to my own build and default to AP mode (but there are potential security issues with that, which is why Sonoff-Tasmota does not default to that).

Good luck!

maddhin, 2017-10-18:
OK, then a possible issue is getting into AP mode per se. I thought I can do 4 x button any time and it stays in AP from then. I guess this should be in the - otherwise excellent - documentation. 

Will test this later. Many thanks for your explanations!

I think it is mainly a documentation issue. Being able to flash OTA is great. Whether `indebuurt1` or AP doesn't matter much. And connecting to the home wifi directly seems just luxurious haha

THANKS!

Electrocut, 2017-10-27:
For your information, I had the same problem with my S20 Socket (using lastest version, SHA1 [61b260f](https://github.com/mirko/SonOTA/tree/61b260f4b58a8732db36a7f45146ecb644572333))

./sonota.py went fine, but when the "FinalStage" SSID disapeered, the Sonoff S20 Socket didn't connect to my WiFi network.

I followed the instruction described above (press 4-5 times the button while the LED was flashing), and a sonoff-xxxx AP has appeared. I could connect to the AP from my computer, browse to http://192.168.4.1, and set the WiFi parameters.

Great :)

PhantomSage, 2017-11-02:
Worked on my RF Bridge on my first try.
What I did:
1: Start SonOTA
2: 7 second push to clear the unit
3: Follow the SonOTA sequence
4: Another 7 second push to make sure it is cleared
5: 4 presses relatively fast to get it into AP, connect & configure.

closed 2017-11-04 by sillyfrog