mirko/SonOTA #17: Sonoff POW not reconnecting to webserver

fusionfsn, 2017-08-09:
I've succesfully upgraded to Tasmota a stock wireless switch 10 minutes ago using SonOTA.
I plugged in a new sonoff POW, i can connect to the wifi network and the provisioning is being sent:

>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "mydeviceid",
    "apikey": "myapikey",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "myssid",
    "password": "mypassword",
    "serverName": "10.0.0.107",
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
~~ Starting web server
~~ Waiting for device to connect

then waits forever. The sonoff POW changes blinking pattern (2 fast blinks/2s pause) but nothing happens.
do i have to debug it with serial or there is something else i could try?

btw, thanks for this useful piece of software :)

DerFetzer, 2017-08-18:
Same here.
@fusionfsn Did you find any solution yet?

fusionfsn, 2017-08-25:
Not really. I've just flashed tasmota soldering the connector.
I have tried some debug but i'm unable to get anything after the boot messages. 

Honestly i don't even know if the original firmware is outputting something to the serial or not. 
Other 3 brand new POWs are waiting to be flashed, if someone gives me some advice on how to debug via serial i'll be glad to post the results

sillyfrog, 2017-08-30:
Sounds similar to an issue here: https://github.com/sillyfrog/SonOTA/issues/1

I have a fork that has pre-built binary files, but it also works slightly differently in that your computer hosts all the binary files (rather than reaching to the public internet).

In theory you should be able to just download and run my fork, and update things. I would be curious if you have a similar issue to these people, and if updating the original Sonoff firmware helps. (Probably best to update the linked issue if doing this on my fork, as it has not been merged in yet, and I'm tweaking based on this feedback).

zobelhelas, 2017-10-01:
I ran into the same issue. I found out, that i need to upgrade to the latest Sonoff upstream firmware first, connecting to their cloud and upgrading via cell phone app. Afterwards, using SonOTA worked like a charm.

fmaule, 2017-10-03:
thanks for the hint. 
how do you exactly use SonOTA after you connect the sonoff through the app? 
I've also tried that but can't get the sonoff in AP mode afterwards therefore i can't send the provision

sillyfrog, 2017-10-03:
Pressing the button for 7 seconds should get you into AP mode again (sometimes you need to hold it for a 2nd time for 7 seconds), and you should be able to flash using SonOTA.

echox, 2017-10-06:
I got the same problem with the Sonoff POW which seems to be related to https://github.com/mirko/SonOTA/issues/21

So this is the setup/discovery I made:

Got two Sonoff POW via Amazon today. Sadly I can't find any datecode or serial which could help to identify the age.
Only thing is a barcode with 160810001 which could be 10/08/16 but this is highly speculative, anyway:

Same problem like mentioned in the links bevore, 404 Debug Message, Timeouts and no "FinalStage" SSID.
I checked the DHCP Server log and saw a assigned ip for the sonoff device.
I also captured some network data on my computer which was interesting: 

Endless ACK requests to open a TCP connection for port 8081. This port is closed and therefore the connection request is terminated via RST, ACK.

This is the DEBUG log from the device:
```>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10000bb90d",
    "apikey": "88bf5043-233c-42d0-861e-299bb95af5d6",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "some-ssid",
    "password": "some-password",
    "serverName": "192.168.1.10",
    "port": 8443
}
<< {
    "error": 0
}
```

This made me wonder, maybe the Sonoff POW did ignore the `port:8443` parameter.
To test this I changed `DEFAULT_PORT_HTTP = 8080` to `8081` in sonata.py and I got one step further.

Now I see some websocket connections and the transfer of the firmware in the default log, e.g.:
```
2017-10-06 19:45:13,483 (INFO) 200 GET /ota/image_user2-0x81000.bin?deviceid=10000bb90d&ts=1994719310&sign=ac5bc413fd91c386605d0d8a9fbfd7c483c52f0ddce07b0c02d3f6e81eb6eee0 (192.168.1.130) 18.89ms
```

Occasionally the switch sends it sate to the "server" but still no "FinalStage" SSID :-/
I tried the Master branch and the D-OUT Branch (81a775c1c4aee86d2be4b314daa8b26270b04ac3).

zobelhelas, 2017-10-06:
@echox: IIRC the latest firmware version from Sonoff is 2.0.4. Did you try to upgrade (at least) one of them to the latest version?

echox, 2017-10-06:
I've updated the Firmware from 2.0.2 to 2.0.4 via the offical EWELink app.
After that the process of flashing the Sonoff POW works smoothly.
```.......No longer on "FinalStage" SSID, all done!```

echox, 2017-10-06:
(Sorry didn't saw your comment. I wasn't aware of the 2.0.2 version before creating a account with email, etc. for the EWELink app. As mentioned, after upgrading to 2.0.4 everything works fine. Thanks.)

sillyfrog, 2017-10-07:
Glad that worked, should you come across any older versions again, there is a `--legacy` flag that would be worth giving a go, it hosts things on the default ports as well for the older devices that did not use the custom port field during the provisioning step.

Cheers.

closed 2017-10-07 by sillyfrog

rajid, 2018-01-04:
I was just trying to do this initial setup  of a new sonoff device and here's what worked for me.
1) press the reset button for 7 seconds.  This makes the green light flash but not in a continuous flashing. so...
2) press the reset button for 7 seconds again.  This results in the green light flashing continuously.
3) set your wifi to the newly created "*LEAD*" named wifi, password "12345678".
4) send the aforementioned "http://10.10.7.1/device" URL and save the output.
5) Now I sent (using Postman) a POST to http://10.10.7.1/ap", with body:
{
  "version": 4,
  "ssid": "my-wifi-name",
  "password": "my-wifi-password",
  "serverName": "my-server-ip",
  "port": 8080
}
(Note, the lack of quotes around the port number and around version number, but quotes around other parts.  I selected port 8080 because I already have a server running on port 80.)
This results in the reply of:
{
    "error": 0
}
and now I see the device on my local wifi trying to communicate with my server at 8080 (which I don't have up yet, so it'll just have to chill for a little while!)

NiekN, 2018-02-17:
@rajid

i also try POSTMAN but i cannot get it working

i get 
Could not get any responce
there was an error connectinmg to http://10.10.7.1/ap
why this might have happend
-the server coul not send a responce
-self-signed SSL certificate are being blocked
-prox config incorrect
-request time out

i used same settings, only a headers:application/json

what is wrong?

haimkastner, 2018-05-13:
@NiekN  
you shuld notice that you need to fill The HTTP/1.1 request fileds in POSTMAN
For example if body is:
`{"version":4,"ssid":"Nxxx","password":"0xxxxxxxxx","serverName":"192.168.1.65","port":1081}`
The Headres need to be:
**Content-Type** : application/json
**Accept** : application/json
**Connection** : close
**Content-Length** : 91
**host** : 10.10.7.2