mirko/SonOTA #19: Did this bricked my sonoff?

krtschmr, 2017-08-24:
i connected with ewelink and my sonoff was DHCP 192.168.1.221. control with ewelink was fine but i need programatical access.

i ran the script

[image]

now i have no access to it anymore. won't connect into my wifi. the default wifi *indebuurt1* does indeed exist but nobody connects into aswell.

if i click the button long time, the green light blinks an i can connect into it's wifi. my PC then gets 10.10.7.2 as IP address but i cant access .7.1

am i misunderstanding the usage?

sillyfrog, 2017-08-26:
It sounds like it has only completed the first step (there are a number of steps). I have tried to outline what needs to happen a bit further in my fork here: https://github.com/sillyfrog/SonOTA/

This also has pre-built binaries so you can just clone and run, and it should work for you.

krtschmr, 2017-08-27:
@sillyfrog i'll try yours via live-stick today. the windows thing kinda not working and very difficult with phyton. running linux via VM also not working due the IP stuff

sillyfrog, 2017-08-27:
Not that I have tested it, but it should work on Windows, and from the screen shot, it looks like you have all of the dependencies installed OK. I would be curious as to what was going wrong as Python is very cross platform.

sillyfrog, 2017-10-06:
Things look to be OK now from most OS's with the latest enhancements, so closing this ticket.

closed 2017-10-06 by sillyfrog