mirko/SonOTA #20: Sonoff that does not advertise ITEAD- WiFi

ageorgios, 2017-09-15:
What is the procedure for sonoff devices (like B1) that do not advertise a WiFi SSID ITEAD-*?

One can use dnsspoof with a hosts file
`192.168.X.X eu-disp.coolkit.cc`

> https://null-byte.wonderhowto.com/how-to/hack-like-pro-spoof-dns-lan-redirect-traffic-your-fake-website-0151620/
> dnsspoof -i <interface> -f <hostsfile>

Can I use the SonOTA when the sonoff device is connected to my local network?
If so what is the command?

Thank you.

sillyfrog, 2017-09-21:
If the DNS spoof works, you should be able to run sonota with the `--legacy` mode (so it's on the standard ports), and it should work in theory (but is untested).  If it does, if you have a chance to put a page up on the Wiki for others that would be great.

It may also need to skip the first step (as SonOTA won't be configuring it, you'll be using the EWeLink app to do that), so if you comment out ` stage1()` on line 624 in `sonota.py`, that should get it going at stage2.

Let me know if it works, and I can also add a command line option to start at stage 2 automatically.

ageorgios, 2017-09-21:
I am willing to test the procedure you said.

I will do:
1) comment line 624
2) dnsspoof -f hosts (hosts = 192.168.??? eu-disp.coolkit.cc) 
3) python sonota.py --legacy

the dns redirect that should happen is the ip of the machine that runs SonOTA?

sillyfrog, 2017-09-21:
Hmm, that’s not connecting at all. Are you able to run dnsspoof on your router. 
I tried doing DNS spoofing some time ago and they only way I could make it work was to run the DHCP and DNS server on the router (which for me is a Linux bad anyway (from memory, I may have also been trying some arp spoofing as well). 
If possible the ideal debug stage now is to do a tcpdump on both the host running Sonota and the router for the traffic coming from the IP allocated to the Sonoff.

ageorgios, 2017-09-22:
I used this method for DNS spoofing:
https://www.youtube.com/watch?v=_O_IeyQj2fA&t=94s

When I got time I will try the tcpdump. What should I check?

rajil, 2017-09-24:
I also did a dns spoof on B1 and was able to flash the device. 
```
2017-09-23 21:07:15,506 (DEBUG) << WEBSOCKET OPEN
2017-09-23 21:07:15,517 (DEBUG) << WEBSOCKET INPUT
2017-09-23 21:07:15,517 (DEBUG) << {
    "userAgent": "device",
    "apikey": "xxx",
    "deviceid": "1000109276",
    "action": "register",
    "version": 2,
    "romVersion": "2.0.2",
    "model": "PSF-BLB-GL",
    "ts": 744
}
----
2017-09-23 21:07:21,847 (INFO) 206 GET /ota/image_user2-0x81000.bin?deviceid=1000109276&ts=1840531613&sign=66738ef420611776bf46b8fad27e8a59215fde35f913adeff5e4721714828c8e (172.16.1.176) 0.96ms
```

I connected to the FinalStage SSID, and it eventually disappeared. I can see on the router that the bulb has an arp entry with hostname ESP_xxxx, but it does not respond to ping. It seems to be dead.

Is there any way to revive this?

rajil, 2017-09-24:
I did not realise that B1 had changed into an AP, and was broadcasting the Sonoff SSID. I was able to connect to it and update to the latest Tasmotta. One litte issue i have is that it is not connecting to the primary WIFI, instead connecting to the secondary one. Otherwise B1 can be flashed by SonOTA!

sillyfrog, 2017-09-24:
Thanks @rajil  - I was just going to suggest that when you replied :) 
I have put that on the wiki as well with a note about DNS spoofing. Was your DNS server on the same host as your DHCP and gateway? I had some weird issues when I was first messing with it, but all my Sonoff's are now flashed so I can't test again. Using `--no-prov' is of course the best thing to do rather than commenting lines out (that will simplify things for you @ageorgios) - I have put a now in the help to clarify this.

@ageorgios If we can get the DNS spoofing working like we want, this shows it should work :) 

Cheers.

rajil, 2017-09-24:
Yes, both DHCP and gateway are on a pfsense router.

ageorgios, 2017-09-24:
I installed a new DNS server (dnsmasq) to make sure the DNS resolves ok.
I run
`sudo ./sonota.py --legacy --no-prov
`
but same happens!!
```
Select IP address of the WiFi interface:
    0: 192.168.178.54
Select IP address [0]: 0
WiFi SSID: MYWIFISSID
WiFi Password: MYWIFIPASSWORD

Using the following configuration:
	Server IP Address: 192.168.178.54
	WiFi SSID: MYWIFISSID
	WiFi Password: MYWIFIPASSWORD
Starting stage2...
~~ Starting web server (HTTP port: 8080, HTTPS port 443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..........
```
 What do I do wrong?

rajil, 2017-09-24:
@sillyfrog, I can also also confirm that BN-SZ01 also works. The instructions were the same as B1. Start pairing with eWeLink and then computer takes over,
```
2017-09-24 09:27:26,751 (DEBUG) << {
    "userAgent": "device",
    "apikey": "xx-yy",
    "deviceid": "1000119cc5",
    "action": "register",
    "version": 2,
    "romVersion": "1.6.2",
    "model": "PSF-B01-GL",
    "ts": 744
}
```

sillyfrog, 2017-09-26:
@ageorgios, my only suggestion is to use the Pi as your DHCP server and gateway as well as @rajil did. I had something similar (from memory, so not 100% sure), were that was the only way I could inspect the traffic. I ended up setting up a spare WiFi AP I had just so I could get inline to mess with the traffic when doing the initial reverse engineering (before finding this project).

sillyfrog, 2017-09-26:
@rajil Thanks for the update, I have put that on the Wiki as well!

jptrsn, 2017-11-09:
I'm not able to get this to work with my Sonoff B1. I'm spoofing eu-disp.coolkit.cc to my machine on 192.168.2.100, and using wireshark, I can see the sonoff hitting it with a request, but the sonota.py output never logs anything received, and I never see FinalStage as an available wireless network.

I also tried spoofing all calls to *.coolkit.cc to the machine running sonota, and logged the output. Attached is the text output, and the pcapng file is at [this link](https://drive.google.com/open?id=1gqx-vpWWzB8fZo4bgRLrWONwIFEqF-8H).

[sonoff_b1.txt](https://github.com/mirko/SonOTA/files/1458779/sonoff_b1.txt)

sillyfrog, 2017-11-13:
@jptrsn Can you include your debug_######.log file as well? Looking at the traffic dump something appears to be dropping the link, so please also ensure all firewalls are turned off.

oglodyte, 2017-11-15:
Worked well for me to flash B1.  Couple of things: 
- if in US need to spoof us-disp.coolkit.cc. (I simply added "address=/us-disp.coolkit.cc/192.168.X.X" to my tomato router DNS server config
- had to run "sudo ./sonota.py --legacy --no-prov --serving-host 192.168.X.X"  - it wanted that serving-host param
Hope this helps someone.

dony71, 2017-12-03:
I have openwrt chaos calmer router and sonoff basic, trying to flash from host 158.168.1.53
Anybody can tell me what I should do to spoof DNS on this router?
I add below in /etc/config/dhcp, but no SSID 'finalstage' 
What am I missing?

config domain 
    option name 'us-disp.coolkit.cc'
    option ip '158.168.1.53'

oglodyte, 2017-12-03:
@dony71  - I use tomato not openwrt, but I'd guess you just need  to add
`address=/us-disp.coolkit.cc/192.168.1.53`
to /etc/dnsmasq.conf

dony71, 2017-12-03:
@oglodyte your suggestion is working. thanks

czyz, 2017-12-06:
I managed to flash the Sonoff B1 using the instructions provided above. DNS spoofed by adding the line mentioned by 'oglodyte' above (plus a few more general ones like coolkit.cc and coolkit.cn to be sure) to dnsmasq.conf in my Almond Plus, and then running the command stated by ageorgios.

Then afterward I compiled the latest development branch of sonoff-Tasmota (5.10.0a) which supports configuring the module as a Sonoff B1. I uploaded the minimal firmware to the bulb, then after that loaded I uploaded the full firmware. It's now working!

ahzazou, 2017-12-20:
so if you comment out stage1() on line 624 in sonota.py  can you attach modified sonata.py as i am not able to understand what should i do clearly

sillyfrog, 2017-12-21:
@ahzazou In the current release, you can do the same thing by running it with `--no-prov` this will skip the stage1 step.

bennnnnnnn, 2018-01-05:
Hi guys,

I'm unable to flash mine. I've followed the steps by @oglodyte, to no avail.

I've done a packet capture, it seems the B1 is doing a DNS lookup and retrieves the spoofed response, but still connects to a (hardcoded?) IP address: 52.28.157.61.

Packet capture in attach: Sonoff B1: 10.9.8.70, DNS/GW: 10.9.8.254, Laptop running SonOTA: 10.9.8.4
[packetcapture.zip](https://github.com/mirko/SonOTA/files/1606976/packetcapture.zip)
Firmware of the B1 is 2.0.3

Does anyone have any ideas?

Thanks!
Ben

JacoFourie, 2018-01-05:
Just want to say a big thank you to all who worked on this. I just flashed 4 units. 2 X T1 Gang 2 units and 2 X T1 Gang 3 units. I saw that the version of firmware on the units according to the app was 2.0.1.
I flashed all 4 without any problems. Thanks a lot. You saved ma a lot of soldering and effort.

oglodyte, 2018-01-07:
@bennnnnnnn I may be having similar problem trying to flash RF Bridge.  Looks like it is trying to connect to my spoofed server few times, but it does not result in firmware update. Then it connects to AWS server IP (also w/out DNS query) and works with WEeLink app. 

Sonoff RF Bridge firmware is 1.1.0 (latest today). Attached log and packet capture. RF Bridge 192.168.5.150. SonOTA server 192.168.5.3
[log&pcap.zip](https://github.com/mirko/SonOTA/files/1609527/log.pcap.zip)

also out of ideas, but hope @sillyfrog can see something there.

laDanz, 2018-01-10:
same issue here as @bennnnnnnn.

Maybe it checks the server certificate?

I notice, that on my spoofed server there is just a "Client Hello" and a "Server Hello, Certificate, Server Hello Done" as response over TLSv1.2 (three times).
While on the real IP there is also a "Client Key Exchange, Change Cipher Spec, Encrypted Handshake Message" (from B1) and a "Change Cipher Spec, Encrypted Handshake Message" as response from server.
So maybe the client (B1) is rejecting the server certificate?

tyjtyj, 2018-01-15:
@oglodyte, can you try generate self sign certificate from here 
http://www.selfsignedcertificate.com/ with server name "eu-disp.coolkit.cc" and replace the crt/key in SSL folder ?

laDanz, 2018-01-15:
@tyjtyj i tried locally with new self signed certificate(from your mentioned site), seems it didn't made a difference...

tyjtyj, 2018-01-16:
Thx @laDanz, I hate the fact that the ssl error shown above means there is a cert install on the client device(the sonoff) which will match with the server. 

I can only think of spoofing the ip address 52.28.157.61 itself which might not worry as it client cert still dont match the server cert but worth a try

laDanz, 2018-01-16:
@tyjtyj spoofing the ip wasn't working for me either: still not a successful SSL handshake:

[image]

tyjtyj, 2018-01-29:
Hi,
I tried to clone all i can from the cert hoping something works 

this is the cert/crt
`
-----BEGIN CERTIFICATE-----
MIICdjCCAiCgAwIBAgIJAMykI5dz1pOCMA0GCSqGSIb3DQEBBQUAMF0xCzAJBgNV
BAYTAmNuMQswCQYDVQQIEwJnZDELMAkGA1UEBxMCc3oxEDAOBgNVBAoTB2Nvb2xr
aXQxCzAJBgNVBAsTAml0MRUwEwYDVQQDDAwqLmNvb2xraXQuY2MwHhcNMTgwMTI5
MjMxNzE3WhcNMjgwMTI3MjMxNzE3WjBdMQswCQYDVQQGEwJjbjELMAkGA1UECBMC
Z2QxCzAJBgNVBAcTAnN6MRAwDgYDVQQKEwdjb29sa2l0MQswCQYDVQQLEwJpdDEV
MBMGA1UEAwwMKi5jb29sa2l0LmNjMFwwDQYJKoZIhvcNAQEBBQADSwAwSAJBAMfX
JjdHoctLuvgaKWFArGV+yGabeXY+vm1BmeQrjz1ZfHd6qZzder2WZ1DyeMU0ESjI
jt64urluVDiOPRYKv0kCAwEAAaOBwjCBvzAdBgNVHQ4EFgQUVyIbr1CarrViRyAP
taBVqPPJXEwwgY8GA1UdIwSBhzCBhIAUVyIbr1CarrViRyAPtaBVqPPJXEyhYaRf
MF0xCzAJBgNVBAYTAmNuMQswCQYDVQQIEwJnZDELMAkGA1UEBxMCc3oxEDAOBgNV
BAoTB2Nvb2xraXQxCzAJBgNVBAsTAml0MRUwEwYDVQQDDAwqLmNvb2xraXQuY2OC
CQDMpCOXc9aTgjAMBgNVHRMEBTADAQH/MA0GCSqGSIb3DQEBBQUAA0EAburZJJic
L/qkgrYnZEW0Ogb1JzbGEr7T7Zmhj1U1CREk+MnGVeeZF9VE3h5qHKlcEYvQ5eNE
GTzI05VI0wm7Mw==
-----END CERTIFICATE-----
`

This is the key
`
-----BEGIN RSA PRIVATE KEY-----
MIIBOwIBAAJBAMfXJjdHoctLuvgaKWFArGV+yGabeXY+vm1BmeQrjz1ZfHd6qZzd
er2WZ1DyeMU0ESjIjt64urluVDiOPRYKv0kCAwEAAQJBALvn0rhR85FLwR9+wf//
Jts8jkTk1h3YO4hSEdjZGJPOf17m325oZOKOAtrnJXG1kYkuxOh/Sqj6y6tfGKiQ
800CIQD/SvyN7ARloTMYksI/Zn6tbvxvR0hkXYJxKin8MpnkRwIhAMhk2DahVzny
fPeKXzMMN3nfk0JSAd2gOXQFAx8vodfvAiBWUHYfTeMWsW/jKQuAABlobhPvU7yI
Io2N0WGsoTci1QIhALeJZV3jt3gjV8GAwmCGt+ivlLgRVBQgQ363eHx/q8ORAiAi
7bP/8xwSFpJdRZ6n155MqW2hEYZR7gRzNTTcEbUWdA==
-----END RSA PRIVATE KEY-----
`

sillyfrog, 2018-01-30:
Based on the last few posts, it sounds like #58 unfortunately... :(

closed 2018-03-14 by ageorgios

halaszj, 2018-08-08:
Any update on this, like, I am on a mac, I have a router that I was going to use with DDWRT, and I have the ATT router which is useless, so that is the reason for the DDWRT. Is there any step by step on how to spoof in this configuration? I went into my MAC settings and set the dns to the DDWRT, and was able to ping us-disp.coolkit.cc and eu just in case, no problem, on the DDWRT i used dnsmasq and did address=/us-disp.coolkit.cc/192.168.1.XXX
address=/eu-disp.coolkit.cc/192.168.1.XXX
from there, I did a telnet session, made sure the dnsmasq.conf and even updated the hosts file to point everything to my mac.

I this command

sudo ./sonota.py --legacy --wifi-ssid mywifi --wifi-password mypassword --serving-host 192.168.1.XXX

I never see the ITEAD ssid pop up, nothing, so, I guess I am at step 1. Can anyone lend a hand here, I really dont want to solder a header on all this bulbs

TomDII, 2018-10-04:
Hey all, I have a B1-v2 with 2.0.3 on it in the US.  I have been hitting a wall with this since last night...

My pi-hole is masking...
```
C:\Windows\system32>ping us-disp.coolkit.cc
Pinging us-disp.coolkit.cc [192.168.1.47] with 32 byt
Reply from 192.168.1.47: bytes=32 time=252ms TTL=128
```
"sonota.exe --legacy --no-prov" just does the ..... then the message and .... over and over. 

I removed my bulb from the app and re-added it with doing the on-off-on-off-on with 2 sec pause between and it just adds it to the app and starts working from the itead cloud.  

Looking for blue skies (No clouds) without solder being I want to get a few more of these if it works out.  

Thanks in Advance, 
Tom

tonycwhite, 2018-11-19:
does anyone have any update on this issue, Like a chump, i spent the day learning to spoof / masq, then eventually when it didnt work, i scrolled down to find security cert issues. :-/ If anyone has a way ahead, it would be appreciated, as i have a b1 with a lifted pad, stopping me soldering.

federom, 2019-01-19:
I have the same issue as Tom, The bulb is in pairing mode, i ran the command, but nothing happens