mirko/SonOTA #22: pip

fragolinux, 2017-09-23:
1st: very well done, all working flawlessly!

2nd: i followed the guide, installing the exact python version suggested, on windows: 3.5.0, adding it to path... then went in the folder where i uncompressed the zip of the repository, and running:
pip3 install --user -r requirements.txt

it gave error about something to be updated, and suggested pip... so i run this:
python -m pip install --upgrade pip

and then again the line before, and all fine now, procedure went smoothly :)
pip3 install --user -r requirements.txt

had to took out the fixed ip from my wifi connection, though, and continue all in dhcp... better doing this BEFORE starting the procedure... i use fixed ip to speed up my wifi connection to router

sillyfrog, 2017-09-23:
To confirm, the install did not work until you upgraded `pip`? I have seen it give warnings before, but not actually stop (I have rarely use Windows, so want to double check).

I'll add it to the README if it's needed.

Thanks!

fragolinux, 2017-09-24:
i uninstalled everything and redid the install... i used python-3.5.0-amd64-webinstall.exe, on 1st screen i choose to install it for every user and checked "add to path", then i went in the folder where was SonOTA and this is the full transcript of what happened, exactly as yesterday (i addedd the ### lines to divide the command i gave)... as you see, in last step, if i run your install command it's clean, while on 1st run, with old pip, it complains... i don't know if it's a halting step or not, but as i prefer not to brick my stuff, i preferred to go the safest path and have my system with no errors before starting your script :)

```
###################################################################################################
PS C:\Users\mrshark\Desktop\SonOTA-master\SonOTA-master> pip3 install --user -r requirements.txt
Collecting httplib2>=0.10.3 (from -r requirements.txt (line 1))
  Using cached httplib2-0.10.3.tar.gz
Collecting netifaces>=0.10.6 (from -r requirements.txt (line 2))
  Using cached netifaces-0.10.6.tar.gz
Collecting tornado>=4.5.1 (from -r requirements.txt (line 3))
  Using cached tornado-4.5.2.tar.gz
Installing collected packages: httplib2, netifaces, tornado
  Running setup.py install for httplib2
  Running setup.py install for netifaces
    Complete output from command c:\users\mrshark\appdata\local\programs\python\python35\python.exe -c "import setuptools,tokenize;__file__='C:\\Users\\mrshark\\AppData\\Local\\Temp\\pip-build-t_c5gzwg\\netifaces\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record C:\Users\mrshark\AppData\Local\Temp\pip-px2b9fy4-record\install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_ext
    building 'netifaces' extension
    error: [WinError 2] Impossibile trovare il file specificato

    ----------------------------------------
Command "c:\users\mrshark\appdata\local\programs\python\python35\python.exe -c "import setuptools, tokenize;__file__='C:\\Users\\mrshark\\AppData\\Local\\Temp\\pip-build-t_c5gzwg\\netifaces\\setup.py';exec(compile(getattr(tokenize, 'open',open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record C:\Users\mrshark\AppData\Local\Temp\pip-px2b9fy4-record\install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with errorcode 1 in C:\Users\mrshark\AppData\Local\Temp\pip-build-t_c5gzwg\netifaces
You are using pip version 7.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
###################################################################################################
PS C:\Users\mrshark\Desktop\SonOTA-master\SonOTA-master> python -m pip install --upgrade pip
Collecting pip
  Using cached pip-9.0.1-py2.py3-none-any.whl
Installing collected packages: pip
  Found existing installation: pip 7.1.2
    Uninstalling pip-7.1.2:
      Successfully uninstalled pip-7.1.2
Successfully installed pip-9.0.1
###################################################################################################
PS C:\Users\mrshark\Desktop\SonOTA-master\SonOTA-master> pip3 install --user -r requirements.txt
Requirement already satisfied: httplib2>=0.10.3 in c:\users\mrshark\appdata\roaming\python\python35\site-packages (from -r requirements.txt (line 1))
Collecting netifaces>=0.10.6 (from -r requirements.txt (line 2))
  Using cached netifaces-0.10.6-cp35-cp35m-win_amd64.whl
Collecting tornado>=4.5.1 (from -r requirements.txt (line 3))
  Using cached tornado-4.5.2-cp35-cp35m-win_amd64.whl
Installing collected packages: netifaces, tornado
Successfully installed netifaces-0.10.6 tornado-4.5.2
###################################################################################################
PS C:\Users\mrshark\Desktop\SonOTA-master\SonOTA-master> pip3 install --user -r requirements.txt
Requirement already satisfied: httplib2>=0.10.3 in c:\users\mrshark\appdata\roaming\python\python35\site-packages (from -r requirements.txt (line 1))
Requirement already satisfied: netifaces>=0.10.6 in c:\users\mrshark\appdata\roaming\python\python35\site-packages (from -r requirements.txt (line 2))
Requirement already satisfied: tornado>=4.5.1 in c:\users\mrshark\appdata\roaming\python\python35\site-packages (from -r requirements.txt (line 3))
###################################################################################################
```

sillyfrog, 2017-09-26:
That is worth doing looking at that error (not getting the wheel), I'll update the readme now.

Cheers!

closed 2017-09-26 by sillyfrog