mirko/SonOTA #24: Stuck in FinalStage - Sonoff TH16, MacOS

hutchwilco, 2017-09-27:
Stuck in FinalStage AP mode currently. 
By my reading there are no glaring errors - see output below. Any advice? Cycling power returns to FInalStage, button appears to do nothing, I don't seem to have an OTA recovery option - serial it is then? Or can I step through the script to pickup at FinalStage again?
How long should I expect the flashing to take place during FinalStage? I think I gave it a good 5 minutes, but it never connected to the SSID again. 

```
Select IP address of the WiFi interface:
    0: 10.0.1.10
Select IP address [0]: 0
WiFi SSID: SSID_WHAT_I_OWN
WiFi Password: PASSWORD_WHAT_GETS_ME_IN

Using the following configuration:
	Server IP Address: 10.0.1.10
	WiFi SSID: SSID_WHAT_I_OWN
	WiFi Password: PASSWORD_WHAT_GETS_ME_IN
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
......................................^[[C.......~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100006b46f",
    "apikey": "0f01e839-97a1-4abd-9c5d-ff4deb5050fe",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "SSID_WHAT_I_OWN",
    "password": "PASSWORD_WHAT_GETS_ME_IN",
    "serverName": "10.0.1.10",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (10.0.1.10) is not  assigned to any interface on this machine.
** Please change WiFi network to applepomis and make sure 10.0.1.10 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
...............~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "error": 0,
    "reason": "ok",
    "IP": "10.0.1.10",
    "port": 8443
}
2017-09-27 21:42:33,485 (INFO) 200 POST /dispatch/device (10.0.1.52) 2.13ms
2017-09-27 21:42:33,669 (INFO) 101 GET /api/ws (10.0.1.52) 0.84ms
2017-09-27 21:42:33,669 (DEBUG) << WEBSOCKET OPEN
2017-09-27 21:42:33,678 (DEBUG) << WEBSOCKET INPUT
2017-09-27 21:42:33,678 (DEBUG) << {
    "userAgent": "device",
    "apikey": "0f01e839-97a1-4abd-9c5d-ff4deb5050fe",
    "deviceid": "100006b46f",
    "action": "register",
    "version": 2,
    "romVersion": "2.0.4",
    "model": "PSA-BHA-GL",
[... 209 lines cut ...]
}
2017-09-27 21:42:40,246 (DEBUG) ~~~ device acknowledged our action request (seq 1506501753698) with error code 0
......

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..........

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.........

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
2017-09-27 21:43:35,666 (INFO) 200 GET /ota/image_user2-0x81000.bin (192.168.4.1) 7100.75ms
No longer on "FinalStage" SSID, all done!
2017-09-27 21:43:59,653 (INFO) Quitting.

```

apathyuk, 2017-10-01:
I got further with my T1 but now its stuck in Final Stage.   Is there a way to tell the script to pick off at that point?

sillyfrog, 2017-10-03:
@apathyuk When I get a chance I'll make it easier to select which stage to start at. At the moment you can use `--no-prov` and that'll get you to stage 2, and (in theory) when you connect to "FinalStage", it should move to stage 3 (the FinalStage). Especially if running on Windows, it's important to ensure that your Firwall is turned off during this process as inbound connections will be blocked otherwise.

@hutchwilco This line suggests the download completed:
`2017-09-27 21:43:35,666 (INFO) 200 GET /ota/image_user2-0x81000.bin (192.168.4.1) 7100.75ms`

I'm not sure what happened after that as it should just work - I would try a full power cycle of the Sonoff and see what you get after the reboot.

hutchwilco, 2017-10-05:
@sillyfrog thanks yes, I figured the download was fine. A full power cycle just left the sonoff creating the FinalStage SSID. Of course the server script had completed and there was no way (that I knew) to start the server at the FinalStage part of the script again.
In the end I flashed it via serial, will try again for the next hardware.

sillyfrog, 2017-10-05:
@hutchwilco Let me know how you go - I think what happened (and this appears to be very rare!), the active image has in the "user2" space (ie: the second half of the flash). The idea is it should have downloaded user1 (which it did), then it would be in FinalStage mode, and and get user2 (which it also did).

The issue then is the script saw that FinalStage had gone away, so it exited (and there is no easy way to get back to stage 3). I'll have a think as to the best way to address that (with out having to start at stage 3 again - which I'm looking to do anyway).

The most common case in the Sonoff gets user2 straight up, so there is no need to hang around for 2 instances of FinalStage - hence not something I have seen before. (And working on this is tricky as once the Sonoff firmware is replaced, I have not found a way to restore it, so each full run requires a new Sonoff).

Thanks for getting back to me, and if you do see thing again, let me know and hopefully I'll be able to come up with a solution.

closed 2017-10-05 by sillyfrog

sillyfrog, 2017-10-06:
Confirming that I have made some changes to fix this now in the `DOUT-mode` branch. This will likely become master shortly, should you come across this again.