mirko/SonOTA #25: Sonoff 4CH Pro

iMarkus, 2017-09-28:
I see in the wiki 4CH is compatible. Is 4CH Pro also working?

https://github.com/mirko/SonOTA/wiki

sillyfrog, 2017-09-30:
I’m not sure if that has been tried, it’s not something I have myself, but worth giving it a try (worst case it’s just a manual flash anyway) and if it works we can update the Wiki :)

Cheers.

iMarkus, 2017-10-03:
As soon as I receive mine I will give it a try. So far thanks for the great work!

Regards

iMarkus, 2017-10-09:
Hi @sillyfrog I can confirm 4CH Pro is working fine. The firmware update to 2.0.1 was needed using official app. You can update the wiki ;)

Thank you so much!

sillyfrog, 2017-10-09:
Awesome, thanks for the feedback. Wiki updated.

jscurtu, 2017-10-15:
@iMarkus 
I think I messed up my 4ch pro .. It is stuck on FinalStage and nothing else is possible.
I was on the newest firmware.
Ordered me a new one and wanted to be sure I get it right this time, other than the latest firmware is there anything else to watch out for?

sillyfrog, 2017-10-15:
Can you send the logs of what’s happening?

If you need to start again, run it with no-prov then connect to the FinalStage SSID.

jscurtu, 2017-10-15:
thx, I tried that with no-prov, no success .. it will not continue. It does nothing and just shows me the info  its connected to FinalStage.

sillyfrog, 2017-10-15:
So there is no progress output?

Also ensure all firewalls are off. If you get asked about public/private or similar, set it to private (but turn off all firewalls is the real solution).

jscurtu, 2017-10-15:
The only progress is the dots , then it repeats .. 
I tried it on Linux and macOS... No firewall active

sillyfrog, 2017-10-15:
@jscurtu Can you do a `tcpdump` on the WiFi interface after connecting to the `FinalStage` SSID and send that through?

The other things to check are:

- Do you have the correct IP address after connecting to `FinalStage`
- Can you ping the Sonoff while connected (on `192.168.4.1`)
- Can you browse to http://192.168.4.2:8080/ota/image_arduino.bin (from on the same host, using `wget` or `curl` should be fine, this is just to make sure the web server is working as expected).

iMarkus, 2017-10-16:
@jscurtu I tried it first with delivered firmware and it did not work (404 error). After this I updated the firmware with EWeLink app and started the script again. Thats all.

OS: Mac OS High Sierra

iMarkus, 2017-10-19:
@sillyfrog just wanted to inform you that i received a second 4ch pro yesterday and went through the following steps:

- turn on
- hold any button for > 7 seconds
- hold any button for > 7 seconds (ap mode)
- connect to itead wifi using smartphone
- add new device using EWeLink app
- update firmware to 2.0.1
- hold any button for > 7 seconds
- hold any button for > 7 seconds  (ap mode)
- start awesome SonOTA script

I never turned off the device and the script finished after a few minutes successfully 👍

After this:

- connect to sonoff wifi
- change wifi settings
- connect to device using browser
- change device type to 4ch pro

Thats all 🥇

jscurtu, 2017-10-19:
@sillyfrog I will get to this probably this evening

jscurtu, 2017-10-20:
@iMarkus one question, which power source did you use (input)? 
The DC 5-24V or the AC 90-250V ?

iMarkus, 2017-10-20:
@jscurtu AC

jscurtu, 2017-10-20:
@sillyfrog 
Today I flashed a fresh 4CH Pro and it worked without a issue.. 

For the one that got fried here are the requested information:

Do you have the correct IP address after connecting to FinalStage
- Yes
[image]

Can you ping the Sonoff while connected (on 192.168.4.1)
- Yes
[image]

Can you browse to http://192.168.4.2:8080/ota/image_arduino.bin (from on the same host, using wget or curl should be fine, this is just to make sure the web server is working as expected).

- No, connection refused.. 
[image]

here is the tcpdump

[dump.zip](https://github.com/mirko/SonOTA/files/1402991/dump.zip)

sillyfrog, 2017-10-20:
@jscurtu This (should) be easy to fix, it looks like the web server is not actually running on port 8080 for some reason when on FinalStage.

Can you please run it with `--no-prov` connect to the `FinalStage` SSID again, wait a minute or so, then send the output? I'm going to expect there is some sort of Python traceback, and that should hopefully point to the cause of the issue.

That test `curl` is what we need to get working, a very quick and easy test as well which is nice.

jscurtu, 2017-10-20:
@sillyfrog  send the output of the terminal? Or a new tcpdump?

jscurtu, 2017-10-20:
@sillyfrog this is what I get after 2 - 3 minutes.

python3 sonota.py --no-prov
Starting stage2...
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
........

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
..............................

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
..

jscurtu, 2017-10-20:
@sillyfrog for what ist worth, I just did a curl on the web server while it was trying .. 

then the status changed .. 

python3 sonota.py --no-prov
Starting stage2...
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
...Sending file: /ota/image_arduino.bin
2017-10-20 23:29:12,922 (INFO) 200 GET /ota/image_arduino.bin (192.168.4.2) 4.67ms
......

But it will stay like that.. the other terminal gave me this the content of the firmware .:
So it looks like the web server is running .. but not uploading

sillyfrog, 2017-10-20:
So that looks like it's listening on port 8080 OK now, can you do a `tcpdump` again please after connecting to FinalStage, I'm hoping to see the SYN packets coming in, but this time actual get an ACK back (rather than the RST of the original tcpdump, when the curl didn't work).

wiktorschmidt, 2017-10-28:
Might be worth updating the wiki to indicate that it does NOT work with v2.0.0. Upgrade to v2.0.1 is required through the official app. Currently the wiki is not clear on this

closed 2017-11-04 by sillyfrog

jscurtu, 2017-11-04:
Yea, I tried flashing over serial, the thing is fried.. nothing to do here..

sillyfrog, 2017-11-04:
Bummer, I have seen some other people that have had bad flash. Maybe worth a warranty claim?

taylormia, 2018-05-20:
I just got a 4CH Pro and added it to Ewelink. It shows Firmware 2.0.0. How do I upgrade the firmware to 2.0.1?

sillyfrog, 2018-05-21:
@taylormia In the eWeLink app, under Settings for the device you can upgrade.

taylormia, 2018-05-21:
I have had my 4ChPro online for several hours but the Ewelink app does not show the 2.0.1 firmware upgrade. My sonoff is accessible to/from the intenet and my Ewelink is on the same wifi network as the sonoff. Any ideas why I don't see the 2.0.1 upgrade? Here is a screen shot of the Setting screen

[image]

taylormia, 2018-05-22:
I finally got the option in Ewelink to upgrade my 4CH Pro to firmware 2.0.1 after almost 3 days. Must have been a Sonoff server problem. Successfully updated to 2.0.1. Then used SonOTA to do the flashing of Tasmota to my device. The Sonoff.exe method is very easy and works flawlessly. Key thing to remember is to make sure ALL the firewalls are turned off in Windows 10 Defender for FinalStage to successfully complete.

giovannigenna, 2018-08-21:
@msound76  have u resolved ? I'm stuck on FinalStage. Same as @jscurtu... The HTTP server reply on 8080 port if i do a wget, but the device 192.168.4.1 doesn't send the request....

msound76, 2018-09-05:
@giovannigenna unfortunately I'm still stuck on FinalStage ... nothing to do here. I'm waiting for a serial/usb adapter to flash Tasmota over serial hoping this will work