mirko/SonOTA #26: Flashing a Sonoff T1 and questions on the process

apathyuk, 2017-09-29:
Hello

I am trying to flash a Sonoff T1 and getting stuck.

Am I right in thinking the process is as follows:

1) Connect PC to HOMENET
2) Run script and answer 3 questions (local IP and HOMENET creditials)
3) Put Sonoff in WiFi AP mode and connect to ITEAD wifi AP
4) Script configs the Sonoff with the OTA settings including connecting to HOMENET
5) PC needs to reconnect to the HOMENET but this time must be on a hardcoded IP of 192.168.4.2
6) Sonoff pulls the OTA update from the script which is running a web server for this purpose
7) Reboot, happy days

I am getting as far as step 5 but the issue I am having is 192.168.4.2 is not routable on HOMENET

Questions:

a) Is there any reason why when passing the HOMENET details to the Sonoff, it cannot also configure the IP address of the PC so that it works on the existing network

b) Any reason why the last step of downloading the firmware cannot be done over the Internet -- it would be easy to create a portable hotspot for this purpose

Kind regards

James

apathyuk, 2017-09-30:
I'll give it a try and let you know. 

For your second point, why not add a parameter which defines what binary to use (local/remote)?

sillyfrog, 2017-10-03:
The "FinalStage" image has a pre-built binary, this has hard coded in it the URL to get stuff from (and with this build it also turns it into an AP rather than a client). It would technically be possible to write to an unused portion of Flash, and hope it's still there after a reboot etc, but the extra complexity is not something I wanted to invest my time in as it would make the image larger, and lead to the potential for more errors (and then need to flash via serial).

If you were to rebuild the binaries at https://github.com/khcnz/Espressif2Arduino and drop them in place in the `static` directory, you could get this for your setup easily (the checksums etc are all dynamic to allow this to happen easily).

Cheers.

closed 2017-10-03 by sillyfrog