mirko/SonOTA #28: Also stuck in FinalStage

tocklime, 2017-10-05:
Output of my run(s) here:
https://gist.github.com/tocklime/6d33a4f9d4a5acbd7e9c0a2e081544af

I'm stuck in a place where the FinalStage SSID is visible, But nothing seems to happen when I connect to it. 

I'm guessing the problem is that my home IP network is on 192.168.4.0, and FinalStage wants to use that too. 

But if it's got enough of a firmware on there to host the Finalstage network, and my PC is not connected to my home network when it's connected to FinalStage, that shouldn't be a problem, right?

Is there any useful diagnostics I can do?

This is a S20. I have another S20 and a T1 to flash like this, so I've got 2 more goes to diagnose this (if this one is going to need serial flashing).

sillyfrog, 2017-10-05:
My first guess would be a firewall is running when connected to the FinalStage network. On Windows 10 people have reported they need to mark the network as "Private" to open up the firewall (I'm guessing the OS here). But it maybe best to completely disable all firewalls for testing and see if that gets you any further.

Running it with --no-prov is the best way to get to the second stage, and then when you connect to the FinalStage SSID, it'll also change to final stage (I have it on my list to make that step easier).

PS: I would suggest removing that gist when you can as it includes the WiFi password (or modify it if you can).

tocklime, 2017-10-06:
Firewall it was. All sorted now, thanks.

I did censor those passwords at some point. Guess I then copy pasted the wrong thing. Thanks for the heads up.

closed 2017-10-06 by tocklime