mirko/SonOTA #30: Stuck in Final Stage

tocklime, 2017-10-08:
My device is stuck on Final stage again, this time the firewall is already off, and I get the same behaviour running it in Linux. I've attached a serial monitor, and this is what is output:

```
Initalizing.......................................................................................................Done
Current rom: 2
Rom 1 magic byte: 0xFF
Reflashing rom: 0
Configuring Wifi...Connecting...
.Connected
Done
	0.0.0.0
Flash Mode: 3
Flash Mode not recognized
```

It's a Sonoff T1. Any ideas?

sillyfrog, 2017-10-09:
Thanks for connecting the serial and sending the output - I'm pretty sure I do know what that is, and thinking about it, it's weird that I didn't see it in my testing!

I know what the fix will be, and it will required some changes to the `image_user*` files unfortunately, but as you have the serial connected, it should be an easy fix.

I have also pulled these files from the repo for the moment until I have a chance to sort it out.

I'll let you know when this is fixed. Cheers.

tocklime, 2017-10-09:
No worries. I've flashed the firmware by serial, so don't hurry on my account.

sillyfrog, 2017-10-09:
Hi,

I'm looking at the code now and I think the error is the issue (I'll sort that, it's not as dire as I thought it was!). To help me debug, had you done any serial flashing of this device previously? It looks like what was written to the flash had errors (which would be possible if there were other things happening on serial already).

Cheers.

tocklime, 2017-10-09:
Nope. From brand new in box, it went:

1. Connect to back box, power up and check local operation.
2. Run sonota.py. Forget to switch off firewall, so device is provisioned, but doesn't download any binaries.
3. Kill sonota.py, switch off firewall, re-run sonota.py with --no-prov. It asked me for wifi credentials anyway. I put in junk (This seems to be a windows only thing. Seperate issue, anyway).
4. First download goes through OK. Final Stage appears.
5. Connect to final stage. Nothing happens.
5. Much fiddling, and restarting of sonota.py, and trying it in linux, but I don't think I actually changed anything.
6. The serial log above.

sillyfrog, 2017-10-09:
Thanks for the detailed breakdown - in that case I'm not sure what happened :/

I have made some changes to the Espressif2Arduino portion so the output is a bit better to help with debugging (I'm about to push it). Please let me know if you see it again with the latest build should you get another one.

Cheers.

closed 2017-10-09 by sillyfrog