mirko/SonOTA #31: SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC

gdgeist, 2017-10-08:
Brand New SONOFF Basics(2) out of the box.  Using latest master branch.  Receive SSL error: (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2083) after First Stage succeeds.  SONOFF units do join the wifi and do attempt to proceed but this error is thrown.  Deleting the certificates generates general errors in stead and the use of self signed certificate/key  does not change the error message.
`.....................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "apikey": "myapikey",
    "accept": "post",
    "deviceid": "mydeviceid"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "password": "mypassword",
    "ssid": "myssid",
    "version": 4,
    "port": 8443,
    "serverName": "192.168.1.39"
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.1.39) is not  assigned to any interface on this machine.
** Please change WiFi network to nestegg and make sure 192.168.1.39 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
........~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.2017-10-08 14:38:59,059 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2083)
.2017-10-08 14:39:01,439 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2083)
...2017-10-08 14:39:07,822 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2083)
................2017-10-08 14:39:39,358 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2083)`

sillyfrog, 2017-10-09:
Can you confirm the version of Python and OS you are running on? It should be a normal download at this point.

The included certificates are just self signed, so unlikely to make much difference, it's not something I have seen before at this stage, but with the above I'll see if anything comes to mind.

gdgeist, 2017-10-09:
PEBKAC - manually installed Python3 from website instead of using Homebrew on OS X.  Installing via Brew of both openssl and python3 then pip to install the requirements resolved!!!!

closed 2017-10-09 by gdgeist