mirko/SonOTA #32: No "FinalStage"

serger2000, 2017-10-08:
Hello!
My Sonoff model TH16
Tried few times but no luck

Select IP address of the WiFi interface:
    0: 192.168.88.208
Select IP address [0]: 0
WiFi SSID: m5_home
WiFi Password: xxxxxxxx

Using the following configuration:
	Server IP Address: 192.168.88.208
	WiFi SSID: m5_home
	WiFi Password: xxxxxxxx
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
............................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100006af71",
    "apikey": "71ec9611-82ad-44c4-a915-120a084a7f11",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "m5_home",
    "password": "xxxxxxxx",
    "serverName": "192.168.88.208",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.88.208) is not  assigned to any interface on this machine.
** Please change WiFi network to m5_home and make sure 192.168.88.208 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
.......~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.......2017-10-08 22:59:25,125 (WARNING) 404 POST /inform (192.168.88.169) 5.43ms
.......................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.........2017-10-08 23:01:28,443 (WARNING) 404 POST /inform (192.168.88.169) 0.96ms
.....................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..........2017-10-08 23:03:31,505 (WARNING) 404 POST /inform (192.168.88.169) 0.96ms
....................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................

and so on
Please help)

thefathefa, 2017-10-08:
Did you check all possible firewalls on your PC which could block the
process?

serger2000, 2017-10-08:
Yes, but I'm using mac+airport

serger2000, 2017-10-08:
And I can see webserver on both ports from other devices at this wifi
[image]

sillyfrog, 2017-10-09:
That is weird as there are so many timeouts for the initial `/device` calls. I suspect that's the cause of the other issues - to start with, I would suggest doing a full restart of everything (WiFi, computer etc), and double checking if you have any software that may mess with traffic (eg: anti-virus, if you do, disable, and reboot your Mac).

serger2000, 2017-10-09:
Thanks for advise. Now I have something new)

Select IP address of the WiFi interface:
    0: 192.168.88.208
Select IP address [0]: 0
WiFi SSID: m5_home
WiFi Password: xxxxxxxx

Using the following configuration:
	Server IP Address: 192.168.88.208
	WiFi SSID: m5_home
	WiFi Password: xxxxxxxx
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
................................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100006af71",
    "apikey": "71ec9611-82ad-44c4-a915-120a084a7f11",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "m5_home",
    "password": "xxxxxxxx",
    "serverName": "192.168.88.208",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.88.208) is not  assigned to any interface on this machine.
** Please change WiFi network to m5_home and make sure 192.168.88.208 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
........~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Traceback (most recent call last):
  File "sonota.py", line 647, in <module>
    main()
  File "sonota.py", line 643, in main
    stage2()
  File "sonota.py", line 571, in stage2
    app.listen(DEFAULT_PORT_HTTP)
  File "/Users/tinkov/Library/Python/3.6/lib/python/site-packages/tornado/web.py", line 1943, in listen
    server.listen(port, address)
  File "/Users/tinkov/Library/Python/3.6/lib/python/site-packages/tornado/tcpserver.py", line 142, in listen
    sockets = bind_sockets(port, address=address)
  File "/Users/tinkov/Library/Python/3.6/lib/python/site-packages/tornado/netutil.py", line 197, in bind_sockets
    sock.bind(sockaddr)
OSError: [Errno 48] Address already in use
MacBook-Pro:SonOTA-master tinkov$

sillyfrog, 2017-10-09:
That last error `OSError: [Errno 48] Address already in use` suggest there is another service already running on port `8080` or `8443`, if you stop that service for the moment, you should be OK.

serger2000, 2017-10-09:
Everething same. Last errors where due to Unify Center.
Now Im back to previus log)
As I see at wifi explorer, after "Starting stage2..." itead ssid apearing several times, but as I understand it must have name FinalStage?

[image]

sillyfrog, 2017-10-09:
If you have stopped the service that was using the port, you should be able to just start again. If you are still having issue, please send through the logs again.
Good luck.

serger2000, 2017-10-09:
Yes, here it is%

Last login: Mon Oct  9 11:15:32 on console
MacBook-Pro:SonOTA-master tinkov$ sudo python3 sonota.py
Password:
Select IP address of the WiFi interface:
    0: 192.168.88.208
Select IP address [0]: 0
WiFi SSID: m5_home
WiFi Password: ********

Using the following configuration:
	Server IP Address: 192.168.88.208
	WiFi SSID: m5_home
	WiFi Password: ********
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
........................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100006af71",
    "apikey": "71ec9611-82ad-44c4-a915-120a084a7f11",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "m5_home",
    "password": "********",
    "serverName": "192.168.88.208",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.88.208) is not  assigned to any interface on this machine.
** Please change WiFi network to m5_home and make sure 192.168.88.208 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
.....~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................

and so on
no SSID FinalStage availible

sillyfrog, 2017-10-09:
Can you see the Sonoff connect to your network? If you have access to the DHCP server logs or similar you should be able to see it (this stage the stock Sonoff firmware is trying to connect to your computer).

The other thing you can try is running it with `--legacy` - unfortunately until it connects again you can't see the version (which would let us know if that's required). If that works, if you can send the output (which will include the romVersion), I can put on the wiki so others know what does and does not work.

Cheers.

serger2000, 2017-10-09:
Yes, 192.168.88.70 in list
I should try: sudo python3 sonota.py --legacy
?

[image]

sillyfrog, 2017-10-09:
Correct, try `sudo python3 sonota.py --legacy`. Good luck!

serger2000, 2017-10-09:
here we are, but no luck(

Select IP address of the WiFi interface:
    0: 192.168.88.208
Select IP address [0]: 0
WiFi SSID: m5_home
WiFi Password: *********

Using the following configuration:
	Server IP Address: 192.168.88.208
	WiFi SSID: m5_home
	WiFi Password: *********
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
.........~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100006af71",
    "apikey": "71ec9611-82ad-44c4-a915-120a084a7f11",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "m5_home",
    "password": "*********",
    "serverName": "192.168.88.208",
    "port": 443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.88.208) is not  assigned to any interface on this machine.
** Please change WiFi network to m5_home and make sure 192.168.88.208 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
........~~ Starting web server (HTTP port: 8080, HTTPS port 443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
........................<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "error": 0,
    "reason": "ok",
    "IP": "192.168.88.208",
    "port": 443
}
2017-10-09 13:39:29,870 (INFO) 200 POST /dispatch/device (192.168.88.70) 1.91ms
2017-10-09 13:39:29,883 (INFO) 101 GET /api/ws (192.168.88.70) 0.98ms
2017-10-09 13:39:29,884 (DEBUG) << WEBSOCKET OPEN
2017-10-09 13:39:29,887 (DEBUG) << WEBSOCKET INPUT
2017-10-09 13:39:29,887 (DEBUG) << {
    "userAgent": "device",
    "apikey": "71ec9611-82ad-44c4-a915-120a084a7f11",
    "deviceid": "100006af71",
    "action": "register",
    "version": 2,
    "romVersion": "2.0.1",
    "model": "PSA-BHA-GL",
    "ts": 733
}
~~~ device sent action request,  acknowledging / answering...
[... 178 lines cut ...]
        "currentTemperature": "26",
        "currentHumidity": "46",
        "sensorType": "AM2301"
    }
}
~~~ device sent action request,  acknowledging / answering...
~~~~ update
>> {
    "error": 0,
    "deviceid": "100006af71",
    "apikey": "bbea280e-3c1e-4f9c-b79f-f4c1568c6e57"
}
......
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..........^C2017-10-09 13:41:01,720 (INFO) Quitting.

sillyfrog, 2017-10-09:
The `"error": 404` is something that we've been trying to fix, and thought we had it sorted.

The `--legacy` has worked in the past with that version (v2.0.1, see here https://github.com/mirko/SonOTA/wiki ).

Now things are getting to the later stages, I'm keen to see if we can resolve this, if you can try running again (same options) on the git SHA `4ff55988b9d53eec461c71825fe9f76ce5f6a5e6` (to do this, go to the directory, and run `git checkout 4ff55988b9d53eec461c71825fe9f76ce5f6a5e6`, and to go back, run `git checkout master`).

This may mean that we need a hybrid of the older and newer flash versions - but I think I can make this automatic, it'll just require some testing - unfortunately none of the hardware I have has ever had this issue so it's a bit of testing what does and does not work. 

Let me know how you go.

PS: The fall back plan is to use the eWeLink app to update to the latest version, bit it would be great if that was not needed...

serger2000, 2017-10-09:
Difference was only in .bin files?

Log:
Select IP address of the WiFi interface:
    0: 192.168.88.208
Select IP address [0]: 0
WiFi SSID: m5_home
WiFi Password: ********

Using the following configuration:
	Server IP Address: 192.168.88.208
	WiFi SSID: m5_home
	WiFi Password: ********
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
........................~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100006af71",
    "apikey": "71ec9611-82ad-44c4-a915-120a084a7f11",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "ssid": "m5_home",
    "password": "********",
    "serverName": "192.168.88.208",
    "port": 443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.88.208) is not  assigned to any interface on this machine.
** Please change WiFi network to m5_home and make sure 192.168.88.208 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
........~~ Starting web server (HTTP port: 8080, HTTPS port 443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.......................<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "error": 0,
    "reason": "ok",
    "IP": "192.168.88.208",
    "port": 443
}
2017-10-09 14:45:38,017 (INFO) 200 POST /dispatch/device (192.168.88.70) 3.48ms
2017-10-09 14:45:38,044 (INFO) 101 GET /api/ws (192.168.88.70) 0.75ms
2017-10-09 14:45:38,044 (DEBUG) << WEBSOCKET OPEN
2017-10-09 14:45:38,055 (DEBUG) << WEBSOCKET INPUT
2017-10-09 14:45:38,056 (DEBUG) << {
    "userAgent": "device",
    "apikey": "71ec9611-82ad-44c4-a915-120a084a7f11",
    "deviceid": "100006af71",
    "action": "register",
    "version": 2,
    "romVersion": "2.0.1",
    "model": "PSA-BHA-GL",
    "ts": 470
}
[... 170 lines cut ...]
        "currentTemperature": "26",
        "currentHumidity": "45",
        "sensorType": "AM2301"
    }
}
~~~ device sent action request,  acknowledging / answering...
~~~~ update
>> {
    "error": 0,
    "deviceid": "100006af71",
    "apikey": "1681d06d-5863-40d4-a1ca-f6fcf959a3a0"
}
......
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..................^C2017-10-09 14:47:28,349 (INFO) Quitting.

sillyfrog, 2017-10-09:
Correct, they use a different flash mode (which was the best guess as to the errors). 

In that case you may need to update to the latest firmware in the app, then try again.

serger2000, 2017-10-09:
Done. Which bins to try?

sillyfrog, 2017-10-09:
The latest should be good. Do a pull first as I have made some more minor update in the last few hours as well. 

Cheers.

serger2000, 2017-10-09:
Everything great! Thanks! I was scaread a litle bit about usbttl flashing)

closed 2017-10-09 by serger2000

sillyfrog, 2017-10-09:
Awesome!
One day I’ll be able to replicate that 404 issue. 

Glad it’s sorted in the end.