mirko/SonOTA #33: How to directly upload the last Sonoff-Tasmota version?

hista, 2017-10-09:
Hi,

According to the readme, SonOTA uploads the v5.5.2, which is not the last one (v5.8.0i currently).
How to directly upload the last Sonoff-Tasmota version?

Best regards and thanks for your very convenient OTA tool!

thefathefa, 2017-10-09:
Ah, I had the same question:-)

Also, it is not clear to me if the Tasmota binary files are made available
somewhere or if they need to be compiled somehow? I know this is not
strictly a SonOTA question, but related to the OP, I think.

Thanks,

whuiscool, 2017-10-10:
Hi，

SonOTA is awesome！！！I've used it to flash my basic and 4CH two devices.
But I have the same question, Sonoff-Tasmota has released 5.8.0 version, but SonOTA now just 5.5.2 version, how do we use the new version in SonOTA?

Best regards and thanks for your very convenient OTA tool!

sillyfrog, 2017-10-10:
Thanks!

I have not updated the version (or build) of Sonoff-Tasmota as this one is now well tested during this OTA upgrade process. That said, I have been doing a bunch more playing around, and would be much more comfortable than I was previously to do this now - which I'll do when I get a chance.

The prebuilt binary is at `static/image_arduino.bin`, and you can put any built Arduino binary file there (with the exact same filename). That said, once you have run the SonOTA, you can also use the built in web UI in Sonoff-Tasmota to replace the image (either as part of it's config, or uploading via the web interface).

Sonoff-Tasmota has a page going through how to do the build - it's important you get all of the steps correct (the custom `boards.txt`, and selections in the Arduino IDE etc) to get a good working binary build. The good news is, if you have another ESP device (I have a WeMos D1 mini for example), you can easily test before doing it on the Sonoff.

Cheers.

closed 2017-10-10 by sillyfrog

sillyfrog, 2017-10-10:
Further to this, I forgot to mention the reason I don't use the binaries from https://github.com/arendst/Sonoff-Tasmota/releases (which you can use once configured to get the latest easily), is the default mode is WPS. This means that after it has been flashed, and there is no config, the Sonoff will try to connect to a WPS enabled AP - which I don't have (and probably many others) - so prefer it starting out as an AP you can connect to.

hista, 2017-10-10:
Your point is relevant.
The ideal solution would be to be able to upload/flash with an already configured firmware with our user settings.
How would it be possible?

sillyfrog, 2017-10-10:
@hista All you need to do is build your own image (with settings), and replace the `image_arduino.bin` file then run SonOTA - the script calculates all the checksums etc on the fly so it's simple to use whatever image you wish.

hista, 2017-10-10:
Yes I need to find a good tutorial to build it... and to wait for receiving my new sonoff S20 to test it because the previous one are already flashed :-)

sillyfrog, 2017-10-10:
And just testing now, what I have said about building and defaulting to AP mode is sorted with the current release of Sonoff-Tasmota. It will be in AP mode for 2 minutes after the reboot. I'm going to make some changes when I get a chance and make this the default included binary, with directions as to where to get the latest.

sillyfrog, 2017-10-10:
No... never mind... my test was bad... I will keep doing my own build so it's in AP mode by default...