mirko/SonOTA #34: Stuck on FinalStage ssid

brenosabino, 2017-10-09:
I'm trying to flash using a windows laptop with python 3.5 everything seemed ok but the program got stuck on the finalstage part.

I disabled all the firewalls rebooted the pc and tried again. It seemed to flash the image_arduino.bin file but I was still connected to the FinalStage ssid but with a 169.xxx ip instead of 192.168.4.2.

Also after rebooting the sonoff the led keeps flashing green, and keeps broadcasting the finalstage ssid. If I run the script again with --no-prov it flashes the file again the led stops blinking, it stays on the correct ip but the script just loops waiting for the ssid to go away.

sillyfrog, 2017-10-10:
I have had one other report of this where some WiFi drivers don't appear to disconnect when the AP goes away.

What you have done sounds like it should work.

Do you see the `sonoff-####` SSID in your list of available networks? (It's often easier to check on your phone, and give it a good 30-60 seconds for it to come up.) If you do see it, things have worked, if not - maybe try a power cycle of the Sonoff (and a reboot of the PC to be sure), and see if that helps. And if you are still having issues, if you can post the output I'll try and guess what maybe happening...

Cheers.

ilya-fedin, 2017-10-19:
I have same problem. I trying to flash Sonoff Basic at laptop with Arch Linux. There is no `sonoff-####` SSID, only FinalStage and Wi-Fi routers.

sillyfrog, 2017-10-20:
@ilya-fedin Weird - the only thing I can think that may cause that is if something is connecting to the open AP before you do (I have seen some older network things do that before), but glad it worked!

mirkorobocop, 2017-10-21:
I also have the same problem with sonoff first generation and im stuck on FinalStage
I tried --no-prev and set static ip unfortunately, this did not work. 
when I connect the device to a power outlet led blinks for 5 seconds and then shines continuously
After connecting to the ssid FinalStage I run python3 sonota.py --no-prev and then I'm in the upload file loop because network never disappears.
Do you have any ideas how to fix this? I'm using windows but I have also tried linux and it is the same problem

sillyfrog, 2017-10-22:
@mirkorobocop If you restart everything (both PC and Sonoff), do you still get the FinalStage SSID? And if so, can you try doing what you outlined again and see if that works. It's very weird as it's downloading the file, but not actually flashing it...

mirkorobocop, 2017-10-22:
@sillyfrog I did as you said but nothing changed

sillyfrog, 2017-10-23:
@mirkorobocop So the same output as well?

If so (ie: FinalStage comes up, it downloads, then does it again), I'm not sure what's happening, it sounds like it's unable to write to flash for some reason. It'll probably require manual flash unfortunately :(

If you do that (ie: solder a header etc), it would be great to get the Serial output for debugging (of what it's doing now before doing the manual flash) and trying to see if this is likely to happen again and/or how we can maybe fix it.

Good luck!

rummyr, 2017-11-02:
I had this .. it kept rebooting back into "FinalStage".
TLDR: ran the --no-prov from windows instead of a RaspberryPi, setup indebuurt AP, configured, all is fine.

So, after having several non-working attempts to flash OTA I connected a serial console.
saw it was in "waiting"
power cycled the slampher, 
re-ran the python script with --no-prov 
connected my Windows box to "FinalStage" 
- and it flashed perfectly fine!

The 2 differences I can remember were:
1. Serial Console attached 
2. ran the python script from Windows instead of an Rpi3

I couldn't get the 4 button press thing to work, and I could see it was trying to connect to indebuurt, so I ended up setting up an "indebuurt" AP on my tablet as described and configuring it from there.

Now it's fully functional.

wejak, 2017-11-03:
I had exactly the same issue with Raspberry3 (latest Raspbian) and Sonoff basic. The sonoff started to download image_arduino.ino, but it never finished. I wiresharked the traffic and the TCP connection just died in a weird way. I retried with --no-prov dozens of times without any change in the behavior.

I even tried to serve the image from Nginx instead of Python, but the TCP trace was identical. The SYN/ACK handshake went fine, sonoff sent "GET image_arduino.ino", raspberry replied 200 OK and started to transfer binary, but after a few IP packets the connection crashed.

I didn't save the wireshark traces, but I do have three more factory-fresh sonoffs waiting.

Switching to Windows and re-running with --no-prov did the job instantly. I never connected serial to the device.

rlust, 2017-11-14:
I used the newest release which got me to the Final _Stage, but now stuck there. Using MAC OSX.

sillyfrog, 2017-11-15:
@rlust Can you please post the logs. Please also ensure all firewalls are completely turned off.

rlust, 2017-11-15:
[debug_1510584891.log](https://github.com/mirko/SonOTA/files/1475808/debug_1510584891.log)
[debug_1510585577.log](https://github.com/mirko/SonOTA/files/1475809/debug_1510585577.log)

Here are 2 log files.

I am constantly getting the FinalStage AP

I have tried putting SOnOff into discovery phase by holding button for 7 seconds, and also pushing button 4 times

No Changes

SunLikeStar, 2017-11-17:
I am also facing the same issue. Stuck at finalstage SSID. Tried many times but same result. I 6 more sonoffs with me. Not sure if I should try it with another one. Any idea why this is happening.

sillyfrog, 2017-11-18:
@rlust Looking at the first log it looks like it didn't download the full image on the first try. Looking at the second log, as it suggests, try running it with `--no-prov`, and connecting to the FinalStage SSID.

hipohop, 2017-11-24:
I have pretty much the same. Log file is here. 
[debug_1511525591.log](https://github.com/mirko/SonOTA/files/1501890/debug_1511525591.log)

I'm lost what  to do now.
Any help is very much appreciated

smcallis77, 2017-11-24:
After much frustration I finally managed to solve this issue for me. I kept getting stuck on finalstage - no download of image_arduino.ino. Windows 10. Finally went into properties in the 'finalstage' network. Changed network profile from 'public' to 'private' and voila 11 seconds later I had a fully working sonoff basic switch. Not sure of the version and cannot check because they are shutdown at the moment, but, I purchased them about a week ago. When they power up again I'll add the version. Hope this helps somebody.

sillyfrog, 2017-11-25:
@ hipohop Can you please confirm you have completely disabled all firewalls. As per @smcallis77 , thIs is a very common issue, especially on Windows, as the 'FinalStage` SSID is considered a new network, and Windows will treat it as new, and default to blocking everything inbound (and that's what the logs suggest).

hipohop, 2017-11-25:
I can confirm that I have turned off Antivirus and Windows Firewall on my windows 10. I'm not sure if  the network profile for 'FinalStage' was public or private. 
I'm lost in how to rerun the entire procedure or where it point of "reentry" in to the procedure is?

sillyfrog, 2017-11-25:
@hipohop To restart from later in the process, run it with `--no-prov`, no need for SSID etc at that point. When you connect to the `FinalStage` SSID, it will figure it out, and go to stage 3.

hipohop, 2017-11-25:
Ok just started the whole process again and when connecting to "finalstage" changed from 'public' to 'private' and it seems to work as well. Thank you all for your help 
[debug_1511588156.log](https://github.com/mirko/SonOTA/files/1502832/debug_1511588156.log)

sillyfrog, 2017-11-25:
Great news! Thanks for letting me know.

closed 2017-11-25 by sillyfrog

ettiennegous, 2018-01-24:
Can confirm I am having the same issue with Rpi3, Will try windows instead

magnusoverli, 2018-02-07:
> To restart from later in the process, run it with --no-prov, no need for SSID etc at that point. When you connect to the FinalStage SSID, it will figure it out, and go to stage 3.

I cannot get it to start again from this step. If I run it using --no-prov, and connects to the FinalStage SSID nothing at all happens....? It just waits for the device to connect again, something that seemingly never happens...

jakobsnet, 2018-04-18:
I also flashed 6 Sonoff 4CH pro OTA. One now remain on FinalStage. I tried the WIFI Setting privat and public. And also startet Sonota.exe with --no-prov. Nothing happens. I guess the Sonota has to request to get the file to be sent agein. but I don't see the request. Any Idea?

magnusoverli, 2018-04-18:
When I had this issue I had to flash them using serial. That was the only way out for me.

Magnus

renanfporto, 2018-08-16:
I also had this issue flashing a sonoff dual v1.3.0, using the sonota.exe and Windows 10. After reading the comments, I figured out that the the windows firewall was only disabled for private connections. So I restarted both PC and Sonoff, disabled firewall for public and private connections, changed FinalStage to private (as @smcallis77 said) and finally ran the Sonota.exe again. Everything worked porfect and now a have a Tasmotized Sonoff. 
Thank you guys!