mirko/SonOTA #35: SSL error

jebo01, 2017-10-12:
I´m following the instructions here https://github.com/mirko/SonOTA and the first stage goes well. In the second stage i get the following error 
.2017-10-12 21:07:01,051 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2083)

I´m not at all an advance user but would be great if anyone could point me in the right direction.
Have upgrade sonoff to latest firmeware and working of a Mac.

sillyfrog, 2017-10-12:
Sounds like it's the same as #31 - does doing the same as this work?

gdgeist, 2017-10-12:
OS X default Python and Openssl cause the issue.

jebo01, 2017-10-13:
Thanks, yes same problem as in #31. I'll try this later on today.

ro-76, 2017-10-21:
I'm having the same problem. 

2017-10-21 22:03:43,877 (WARNING) error on read: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1976)

I had installed python manually, so I went and installed via homebrew, but I still get the error.

sillyfrog, 2017-10-22:
@ro-76 Can you please send the output of `/usr/bin/env python3 --version`, and how you are calling the script? I'm guessing it's still using the system python.

jebo01, 2017-10-22:
@ro-76 I had the same thing, I ran `brew doctor` and found some issues, after fixing those it all worked for me.

closed 2017-12-12 by jebo01