mirko/SonOTA #36: No FinalStage SSID on Sonoff POW

TommySharpNZ, 2017-10-14:
I've just managed to flash 3 Sonoff basics with no issues. Thanks again for a fantastic solution!

Now I'm trying to flash a POW and I'm going through the exact same steps but I'm just not getting a FinalStage SSID show up like with the basics.

Any ideas? The LED is flashing 2 shorts and then a pause.

.............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
...........

sillyfrog, 2017-10-14:
Can you please send through the full output from the run (remember to replace your WiFi password).

lennardk, 2017-10-14:
I just tried my first SonOTA flash, and it was on a POW as well. I had the same problem, but I noticed the POW connecting to port 8081 of the server. On a whim I redirected port 8081 like below, and this worked.

`sudo iptables -t nat -I PREROUTING -d <server_ip> -p tcp --dport 8081 -j REDIRECT --to 8080`

sillyfrog, 2017-10-14:
Sounds like it maybe older firmware. The `--legacy` option will do something similar. Glad you got it sorted!

TommySharpNZ, 2017-10-15:
I still have an issue so can we re-open this issue please?

fat-fred, 2017-10-16:
I have also an issue. With the --legacy i come to the finale stage with the POW. but here it hangs.

TommySharpNZ, 2017-10-18:
Using the eWelink app and upgrading my POW to 2.0.4 resolved my issues!
I didn't have to use legacy or anything.....
Although did have to press and hold the button twice to get the itead SSID

sillyfrog, 2017-10-20:
@fat-fred Can you send through the full output if you are still having issues? Also, please ensure you have turned off all firewalls, this has been an issue an number of times when stuck on the FinalStage SSID (as it's a new network, by default it's "public", and so blocks everything).

psidlo, 2017-10-21:
I had the same problem. First I saw POW was trying to connect to 8081 port so I used --legacy option. Then POW did not find right file on sonota web server. After upgrade to 2.0.4 issue was resolved.

Here is output of SonOTA: https://pastebin.com/sTTQbya2

sillyfrog, 2017-10-22:
Glad an update fixes it, weird it's happening on this one so reliably on the older release. It sounds like it's likely the same as #21 (so going to assume it's the same for the moment). I have not been able to replicate this to date, but have a Sonoff that has had the issue shipped to me so I can look, and try and figure out a fix.

closed 2017-10-22 by sillyfrog

soytoledano, 2017-11-06:
Hi,

first of all, thanks for this fantastic work!

It seems I have the same issue with the Sonoff POW ver 2.0 with Python 3.7.0a2 using Windows 7. I´ve disabled Windows Firewall and the antivirus (just in case), but The FinalState SSID doesn´t come up and the Sonoff seems to go to "normal mode" - Blue led blinking (same than when I power it up). I´ll try to Upgrade the POW to 2.0.4 to see if that also resolves the issue (need to install eWelink app...).

2017-11-06 21:22:17,011: DEBUG: Current IPs: ['192.168.1.130']
2017-11-06 21:22:39,459: INFO: Using the following configuration:
2017-11-06 21:22:39,459: INFO: 	Server IP Address: 192.168.1.130
2017-11-06 21:22:39,459: INFO: 	WiFi SSID: JAZZTEL_bdc5
2017-11-06 21:22:39,459: INFO: 	WiFi Password: ************
2017-11-06 21:22:39,459: INFO: Platform: win32
2017-11-06 21:22:39,506: INFO: ** Now connect via WiFi to your Sonoff device.
2017-11-06 21:22:39,506: INFO: ** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
2017-11-06 21:22:39,506: INFO: To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
2017-11-06 21:22:39,506: INFO: ** This application should be kept running and will wait until connected to the Sonoff...
2017-11-06 21:23:32,905: DEBUG: Current IPs: ['10.10.7.2']
2017-11-06 21:23:32,920: DEBUG: ~~ Connection attempt
2017-11-06 21:23:32,920: DEBUG: >> HTTP GET /10.10.7.1/device
2017-11-06 21:23:32,920: DEBUG: << {
2017-11-06 21:23:32,920: DEBUG:     "deviceid": "10000b77c1",
2017-11-06 21:23:32,920: DEBUG:     "apikey": "289a1f0a-f496-41b3-8270-fbceb92fe43b",
2017-11-06 21:23:32,920: DEBUG:     "accept": "post"
2017-11-06 21:23:32,920: DEBUG: }
2017-11-06 21:23:32,920: DEBUG: >> HTTP POST /10.10.7.1/ap
2017-11-06 21:23:32,920: DEBUG: >> {
2017-11-06 21:23:32,920: DEBUG:     "version": 4,
2017-11-06 21:23:32,920: DEBUG:     "ssid": "JAZZTEL_bdc5",
2017-11-06 21:23:32,920: DEBUG:     "password": "************",
2017-11-06 21:23:32,920: DEBUG:     "serverName": "192.168.1.130",
2017-11-06 21:23:32,920: DEBUG:     "port": 8443
2017-11-06 21:23:32,920: DEBUG: }
2017-11-06 21:23:33,014: DEBUG: << {
2017-11-06 21:23:33,014: DEBUG:     "error": 0
2017-11-06 21:23:33,014: DEBUG: }
2017-11-06 21:23:33,014: INFO: ~~ Provisioning completed
2017-11-06 21:23:33,014: INFO: Starting stage2...
2017-11-06 21:23:33,029: INFO: ** The IP address of <serve_host> (192.168.1.130) is not assigned to any interface on this machine.
2017-11-06 21:23:33,029: INFO: ** Please change WiFi network to JAZZTEL_bdc5 and make sure 192.168.1.130 is being assigned to your WiFi interface.
2017-11-06 21:23:33,029: INFO: ** This application should be kept running and will wait until connected to the WiFi...
2017-11-06 21:23:45,251: DEBUG: Current IPs: ['192.168.1.130']
2017-11-06 21:23:45,251: INFO: ~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
2017-11-06 21:23:45,251: INFO: ~~ Waiting for device to connect
2017-11-06 21:23:45,282: INFO: *** IMPORTANT! ***
2017-11-06 21:23:45,282: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 21:23:45,282: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 21:23:45,282: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 21:23:45,282: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 21:23:52,601: WARNING: 404 GET / (fe80::d54f:dea6:ef55:e466%11) 0.00ms
2017-11-06 21:23:52,632: WARNING: 404 GET / (fe80::d54f:dea6:ef55:e466%11) 0.00ms
2017-11-06 21:23:52,694: WARNING: 404 GET / (192.168.1.130) 0.00ms
2017-11-06 21:23:52,710: WARNING: 404 GET / (192.168.1.130) 0.00ms
2017-11-06 21:24:46,362: INFO: *** IMPORTANT! ***
2017-11-06 21:24:46,362: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 21:24:46,362: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 21:24:46,362: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 21:24:46,362: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 21:25:47,639: INFO: *** IMPORTANT! ***
2017-11-06 21:25:47,639: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 21:25:47,639: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 21:25:47,639: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 21:25:47,639: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 21:26:48,901: INFO: *** IMPORTANT! ***
2017-11-06 21:26:48,901: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
[... cut ...]
2017-11-06 21:29:52,436: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 21:30:53,682: INFO: *** IMPORTANT! ***
2017-11-06 21:30:53,682: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 21:30:53,682: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 21:30:53,682: INFO: This server should automatically be allocated the IP address: 192.168.4.2.

Thanks for the help.

sillyfrog, 2017-11-07:
@soytoledano I would suggest checking your DHCP server logs and make sure the song off is connecting to your Wi-Fi network. It may also be worth trying the `--legacy` option.

soytoledano, 2017-11-07:
Hi @sillyfrog, thanks for the help. I tried the --legacy option and nothing changed. Then I updated the sonoff version to 2.0.4 (I had 2.0.2) and the program got to the "FinalStage", but when I tried to connect to it, windows couldn´t connect, not sure why, but I suspect was a windows issue, cause I could connect from the phone. I restarted windows and now I can connect to FinalStage (and get the right IP).
Now my problem is that the sonoff can´t be restarted (holding the button does nothing) and even if I unplug it, when I plug it back, it powers up in the "finalStage" state, cause I can see the SSID.
I was thinking to run the sonora.py function but "hack" it to jump directly to the FinalStage to sync the program with the sonoff. Is there an easy way to do that in the program? I´m not too familiar with phyton. Or is there a way to get the sonoff out of that state?

Thanks,
Daniel

soytoledano, 2017-11-07:
Ok, I just bypassed the stage1 and it went to the "finalStage" as I wanted. It seems it did what it´s supposed to do and finish (finalstage SSID is gone=, but not sure if the sonoff is working fine. There is no led on and It´s not connected my SSID. Pushing or holding the button doesn´t seem to do anything. If I umplig it and plug it again, the blue led blinks and the goes of.  Hope I haven´t bricked it...

Log:

Current IPs: ['192.168.1.130']
Select IP address of the WiFi interface:
    0: 192.168.1.130
Select IP address [0]: 0
WiFi SSID: JAZZTEL_bcd5
WiFi Password: kr2gq66syfva
Using the following configuration:
	Server IP Address: 192.168.1.130
	WiFi SSID: JAZZTEL_bcd5
	WiFi Password: ************
Platform: win32
Starting stage2...

~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
...........Current IPs: ['192.168.4.2']
The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program

Sending file: /ota/image_user2-0x81000.bin
...200 GET /ota/image_user2-0x81000.bin (192.168.4.1) 7255.01ms
.......Sending file: /ota/image_arduino.bin
....200 GET /ota/image_arduino.bin (192.168.4.1) 8548.82ms
........Current IPs: ['192.168.1.130']
No longer on "FinalStage" SSID, all done!
Quitting.

soytoledano, 2017-11-07:
Hi again, I just found out that it´s actually working. The button is working, although it´s not very intuitive in my opinion... I was able to Start Wifi manager providing an Access Point with IP address 192.168.4.1 with 4 short presses (I guess). I thought it didn´t work cause the leds are off until the wifi is configured.

Fantastic program!! Thanks a lot for your work, so simple and quick to flash it... (if no issues faced hehe)

marine1988, 2018-01-21:
hi there im stuck in final stage how can you by pass til final stage?

fysmd, 2018-01-27:
Hi all,
I'm having same issues with Sonoff POW but got it working - two options:
OSX, firewall disabled, confirm web server reachable on 8080 from another host (didn't test ssl)

POWs shipped with v2.0.2 of their sofware and I could deploy using:
` sudo python3 ./sonota.py --legacy --slowstream`

I tried upgrading one using Ewelink software to v2.0.4 and that one would deploy using:
 `sudo python3 ./sonota.py`

Hope this helps!

soytoledano, 2018-01-28:
Hi, you just need to comment out the stage2

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

sillyfrog, 2018-01-28:
If you are stuck on the FinalStage SSID, there is a `--no-prov` option to take you to stage2/3, so when you join the FinalStage SSID, it will start service that content right away.

flavio20002, 2018-03-21:
Just updated a brand new Pow (from Amazon Italy). Got stucked 3 times with normal procedure. Added legacy option. After 2 long minutes the Relay "clicked". After another 3 minutes the websocket got a request update and stucked. So i installed Ewelink, update to 2.0.4 and successfully flashed without any option. Thanks to everybody!

obstler42, 2018-04-16:
I tried to follow the instructions here, as I also have a sonoff pow on fw 2.0.2 that fails to OTA flash. but how can I upgrade the firmware to 2.0.4? Installed ewelink, added the pow in there, but it says "Firmware version: 2.0.2   Latest Version" -- and I can't do anything else there.

How does one get ewelink to upgrade it?

flavio20002, 2018-04-16:
It happened to me with the second pow I've bought. Some versions can't be upgraded. You have two choices...try with legacy and DNS simulate, otherwise, open it and flash with an usb to serial adapter. I've chosen the second one, and my device is still dead (I made hundreds of attempts, but I just get failures). Keep trying...

cirolaferrara, 2018-04-27:
Installed ewelink, added the pow in there, but it says "Firmware version: 2.0.2 Latest Version"

henrikwils, 2018-10-07:
My POW shipped with 2.0.4, but I'm stuck as well, and never get to Final Stage, using a Mac with firewall disabled. I noticed that firmware 2.6.0 is available, but the only entry in the wiki with 2.6.0, says it's not working, so I'm reluctant to try that.

I might try to see if I can get it working using a Windows PC.

sillyfrog, 2018-10-07:
@henrikwils Can you please attach logs, I maybe able to figure out what's happening

henrikwils, 2018-10-07:
> @henrikwils Can you please attach logs, I maybe able to figure out what's happening

Thanks, but I got it working now. I think a weak wifi connection was the culprit :)

sillyfrog, 2018-10-12:
@henrikwils It's unlikely on the latest firmware, basically if there are no lines with `200 GET` in them, then it's failing to connect to the web server (because of SSL).

tiagofontoura, 2018-10-23:
good night, I'm trying to install via sonota.exe, but it's always in the stage where it leads to the final stage, details:
windows 10
model: sonoff touch US
version of the pod: 1.8.1 (latest version)
firewall disabled
running sonota.exe as admin
I am looking forward to help
follows the log.
[debug_1540254458.log](https://github.com/mirko/SonOTA/files/2503949/debug_1540254458.log)