mirko/SonOTA #38: Email adress invalid

JacoFourie, 2017-10-19:
Hi. I sent you an email but it keeps on coming back. Not sure if it is wrong on the wiki.
I must say thank you sooo much for this. I have flashed 3 Sonoff Basics and 2 Touch.
It worked with some issues but I managed to get them all done.  The script would get stuck at the FinalStage SSID part. It would sit there and I can not kill the script. Only way is to close the window.
The I have to restart the windows PC as it will say that it is still busy with the FinalStage every time you start the script.

One of the touch units is non responsive after an update. It did do all its relay click etc but now it wont bot up. How can I get the unit back ?

Thanks again for all the work. It saved me a lot of time as I did 2 sonoffs the old way and it takes a lot longer.

sillyfrog, 2017-10-20:
Re the touch, do you have the output of the run?

There have been some others with issues with the Touch and doing upgrades (even just using the eWeLink app) - so I think there is a quality issue with the Flash on some of them.

That said, the first thing to try is to do a normal reset and try and start the process again. Importantly, it's good to keep the output to try and debug as well.

Also, if it did get past the FinalStage (sometimes the PC can reconnect), it could be setup, so it has "Sonoff-Tasmota" installed already (and just needs to be configured).

Good luck!

JacoFourie, 2017-10-20:
Good news is you where correct. The Touch did update. I hit the button 4 times to put it into AP mode and configured it to connect to my WiFi network. I have ordered a couple of T1 units. I hope to flash them this way as well.  Thanks for all the hard work.

closed 2017-11-04 by sillyfrog