mirko/SonOTA #39: Syntax error

ro-76, 2017-10-21:
Please be patient with a python newbie. I followed all the steps and installed the dependencies and requirements, but when I run the script I get this error:

File "./sonota.py", line 499
print(".", end="", flush=True)
^
SyntaxError: invalid syntax

Please help!

I'm using OSX and installed the latest python 3.6. I dragged the sonota.py file onto the launcher for the 3.6 app. I've also tried on 2 different macs and get the same error. 

Thanks

sillyfrog, 2017-10-21:
Hmm, I have not tried using the GUI before to run it. If you open Terminal, then ‘cd’ to the folder with Sonota.py, then type ‘./sonota.py’, it should get you going. 

If not let me know and I can be more specific tomorrow when I’m in front of a Mac.

ro-76, 2017-10-21:
Thanks for that. I tried also running it from terminal but with the same result.

closed 2017-10-21 by ro-76