mirko/SonOTA #41: Sonoff 1 Channel Inching Relay Module (ESP 8285)

boraygowda, 2017-10-23:
Hi,

Is there a different process for ESP8285 based boards? I've seen a separate item in the board manager on Arduino IDE, however, in the SonOTA OTA update procedures laid down, there's nothing specific for 8285.

Will the documented process work equally well for 8285 based boards like the one I mentioned?
Or is there anything different I must do?
So far I've not tried for the fear of bricking/overwriting the stock firmware.

Would really appreciate any advice/recommendations.

Thank you.
BG.

sillyfrog, 2017-10-23:
The list of units where this has worked is listed on the Wiki. It should work, and the builds should support both, but everything that I know about is there. However that said, this project is still in beta, and could have problems, so worse case there's always a chance you'll need to manually flash anyway.

I have some notes on backing up the firmware here: https://github.com/mirko/SonOTA/issues/1#issuecomment-334724811

If it does work, please let me know and I can update the wiki (or feel free to update the wiki yourself).

Good luck.

boraygowda, 2017-10-23:
Hi, Thank you so much for your guidance.
Unfortunately it didn't work.  I went through the steps until it connects to "FinalStage", does the upload and disconnects. For what it's worth, here's the logs:  https://pastebin.com/kJrDHu5J

I then rebooted the Sonoff and now it doesn't connect to my router.
The relay also goes on and off every second. Just weird behavior.

As a next step I will try to flash the Sonoff-Tasmota through the USB programmer and see if that works.

Thank you so much for your help. I appreciate your time, patience and all the work you've done here.

-BG.

sillyfrog, 2017-10-23:
Hi, That log looks good! :) 

It'll probably be in the default Tasmota configuration (looking to join via WPS, and trying the default WiFi networks). see here for more information (including button presses etc): https://github.com/mirko/SonOTA#running

boraygowda, 2017-10-23:
Sorry, I jumped the gun and posted before I could try a few things...

I was able to get the Sonoff to join the router and I can see the configuration page.

1. Then I selected the device type to be "1 Channel".
2. Saved configuration and rebooted.
3. Copied and pasted this URL to set the pulse time to 1 second:
_http://IPSONOFF/cm?cmnd=PulseTime2011_
(The above is from _https://github.com/arendst/Sonoff-Tasmota/issues/453_)
4. Saved configuration and rebooted.
5. Clicked the outer button (Mode Setting/Toggle button) to set the relay in Self-Lock mode (where switch state is persistent)
6.  Now when I press the on/off button, it switches off automatically after 1 sec.  THis behavior is also confirmed when triggered via the web page.

Phew! Now the Sonoff is behaving in a useable relay like way. I will now add a device handler on the Smartthings and see if I can control it via SmartThings app.

I can't thank you enough for standing by and helping me out.

Thank you so much!
BG.

sillyfrog, 2017-10-23:
Awesome! Glad it worked.

I have also updated the Wiki.

Enjoy!

closed 2017-10-23 by sillyfrog

sfanicola, 2017-10-31:
guys
I got almost BG's same problem, unfortunately mine doesn't seem to solve alone!
My sonoff 1Channel is stuck in FinalStage status. Same as above the relay also goes on and off every second. I can stop it by pressing first button.
I tried to restart sonota.py with --no-prov but it simply doesn't work.

sillyfrog, 2017-11-01:
To confirm, you have the finalstage in SSID? Are you able to connect to it? If so, can you please send through the logs. The most common reason there is an issue at this stage is because of a firewall running when connected to the final stage SSID, so it can never downloaded the image.

sfanicola, 2017-11-01:
Sorry but I don't have the complete log, just this:

iMac:SonOTA-master nicola$ python3 sonota.py --no-prov
Starting stage2...
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed
Once "FinalStage" has gone away, you can stop this program
.......................

I don't think it's a firewall problem because I don't have one.
FinalStage is available and I'm connected to it.
Thanks in advance for your support.

boraygowda, 2017-11-01:
The firewall was my issue. I'm on Windows7 and when joining FinalStage, the firewall window popped up, but went hiding under a bunch of other windows, so I missed it.  To my luck, I saw the firewall dialog box and I hadn't killed the running script.  So once I allowed permission the script continued and ultimately completed successfully.  The documentation specifically addresses this issue (thanks to the person who documented this).

Are you on Windows?  If so, you will most likely have the firewall that came with the OS.
I don't have any external firewall either, but use the one that came with the OS.

When you re-run the script from a clean start, please do collect the logs. It will be very helpful to troubleshoot.

sfanicola, 2017-11-01:
I’m on Mac. I can’t rerun the script from the start... the relay is stucked on FinalStage

adamrobinson361, 2017-11-10:
Hi @boraygowda,

Please can you expand on what you did to get from the intermittent relaying to being able to connect.

I'm in the same situation you were in.

Regards,

Adam

sfanicola, 2017-11-10:
To solve the intermitten on/off I had to remove the IC like in the photo below
[https://user-images.githubusercontent.com/9368955/28057156-21e35406-65d4-11e7-88b1-a32e8a1379b1.png](url)
After that I flashed the board with arduino, because the script simply didn't go on

boraygowda, 2017-11-13:
Adam,

When it started I simply reset the board with a long press (I forget how many seconds), and restarted the process.

BG.

ryanhunt, 2017-12-14:
I found that this command: `http://IPSONOFF/cm?cmnd=PulseTime2011` didn't work for me. What I had to do was (from the Sonoff WebUI): 

Main Menu -> Console -> and then in 'Enter Command' I typed: `PulseTime 30` this seemed to work for me.