mirko/SonOTA #42: Unable to install requirements: issue with netifaces

papaiatis, 2017-10-23:
System: Linux Mint 18.2 64 bit
Python installed: 3.5.2
PIP3 installed: 9.0.1

```
pip3 install --user -r requirements.txt 
Requirement already satisfied: httplib2>=0.10.3 in /home/atis/.local/lib/python3.5/site-packages (from -r requirements.txt (line 1))
Collecting netifaces>=0.10.6 (from -r requirements.txt (line 2))
  Using cached netifaces-0.10.6.tar.gz
Collecting tornado>=4.5.1 (from -r requirements.txt (line 3))
  Using cached tornado-4.5.2.tar.gz
Installing collected packages: netifaces, tornado
  Running setup.py install for netifaces ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-r6gnu8lp/netifaces/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-o75pi5lg-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_ext
    checking for getifaddrs...found.
    checking for getnameinfo...found.
    checking for IPv6 socket IOCTLs...not found.
    checking for optional header files...netash/ash.h netatalk/at.h netax25/ax25.h neteconet/ec.h netipx/ipx.h netpacket/packet.h linux/irda.h linux/atm.h linux/llc.h linux/tipc.h linux/dn.h.
    checking whether struct sockaddr has a length field...no.
    checking which sockaddr_xxx structs are defined...at ax25 in in6 ipx un ash ec ll atmpvc atmsvc dn irda llc.
    checking for routing socket support...no.
    checking for sysctl(CTL_NET...) support...no.
    checking for netlink support...yes.
    will use netlink to read routing table
    building 'netifaces' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DNETIFACES_VERSION=0.10.6 -DHAVE_GETIFADDRS=1 -DHAVE_GETNAMEINFO=1 -DHAVE_NETASH_ASH_H=1 -DHAVE_NETATALK_AT_H=1 -DHAVE_NETAX25_AX25_H=1 -DHAVE_NETECONET_EC_H=1 -DHAVE_NETIPX_IPX_H=1 -DHAVE_NETPACKET_PACKET_H=1 -DHAVE_LINUX_IRDA_H=1 -DHAVE_LINUX_ATM_H=1 -DHAVE_LINUX_LLC_H=1 -DHAVE_LINUX_TIPC_H=1 -DHAVE_LINUX_DN_H=1 -DHAVE_SOCKADDR_AT=1 -DHAVE_SOCKADDR_AX25=1 -DHAVE_SOCKADDR_IN=1 -DHAVE_SOCKADDR_IN6=1 -DHAVE_SOCKADDR_IPX=1 -DHAVE_SOCKADDR_UN=1 -DHAVE_SOCKADDR_ASH=1 -DHAVE_SOCKADDR_EC=1 -DHAVE_SOCKADDR_LL=1 -DHAVE_SOCKADDR_ATMPVC=1 -DHAVE_SOCKADDR_ATMSVC=1 -DHAVE_SOCKADDR_DN=1 -DHAVE_SOCKADDR_IRDA=1 -DHAVE_SOCKADDR_LLC=1 -DHAVE_PF_NETLINK=1 -I/usr/include/python3.5m -c netifaces.c -o build/temp.linux-x86_64-3.5/netifaces.o
    netifaces.c:1:20: fatal error: Python.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-r6gnu8lp/netifaces/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-o75pi5lg-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-r6gnu8lp/netifaces/

```

Google doesn't help me with this error message. 
Is Linux supported at all? Because only Windows is mentioned in the README.

Any idea would be great.

papaiatis, 2017-10-23:
I confirm that the initial setup steps works on Windows 7 64 bit.
Seems Linux isn't supported well.

sillyfrog, 2017-10-24:
It looks like it's missing the Python and header files, normally this is in a Python-dev package.

I would suggest searching for this: "fatal error: Python.h: No such file or directory"

Glad it worked on Windows, good luck!

closed 2017-10-24 by sillyfrog