mirko/SonOTA #43: FinnalStage SSID never shows up

christiamdc, 2017-10-23:
Hi.

First thank you for the effort for developing this tool. 

I know a similar issue has been reported but i still want to share the following information:

-I tried SonOff basic and SonOff S20 in two different wifi networks with all the requirement but I get stuck waiting for the FinalStage SSID to show up but this never happen. After this I can still control the SonOffs with ewelink app but only in the wifi network, not anymore from internet. Ewelink app shows the message "Control by Lan".

I don't know if this info could help but I share it just in case.

Thanks

sillyfrog, 2017-10-24:
Thanks for the info, are you able to send the output from a run through as well please? (remember to remove your WiFi password). I can then take a look and try and see what it maybe.

christiamdc, 2017-10-24:
Logs:

C:\Users\christiam.munoz-del-\SonOta\SonOTA-master>python sonota.py
Select IP address of the WiFi interface:
    0: 192.168.1.17
Select IP address [0]: 0
WiFi SSID: BettyCl
WiFi Password: XXXXXX

Using the following configuration:
        Server IP Address: 192.168.1.17
        WiFi SSID: BettyCl
        WiFi Password: XXXXXX
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default pas
sword is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light wi
ll start flashing rapidly.
** This application should be kept running and will wait until connected to the
Sonoff...
.........~~ Connection attempt
>> HTTP GET /10.10.7.1/device
timed out
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "1000113dc3",
    "accept": "post",
    "apikey": "4439b735-6a9e-4503-b94a-b37fe46fddbf"
}
>> HTTP POST /10.10.7.1/ap
>> %s {
    "version": 4,
    "port": 8443,
    "ssid": "BettyCl",
    "password": "XXXXXX",
    "serverName": "192.168.1.17"
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.1.17) is not  assigned to any interfa
ce on this machine.
** Please change WiFi network to BettyCl and make sure 192.168.1.17 is being ass
igned to your WiFi interface.
** This application should be kept running and will wait until connected to the
WiFi...
...........~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should conne
ct to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiF
i network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Addres
s you were allocated, please ensure no other device has connected, and reboot yo
ur Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should conne
ct to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiF
i network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Addres
s you were allocated, please ensure no other device has connected, and reboot yo
ur Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should conne
ct to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiF
i network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Addres
s you were allocated, please ensure no other device has connected, and reboot yo
ur Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should conne
ct to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiF
i network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Addres
s you were allocated, please ensure no other device has connected, and reboot yo
ur Sonoff.
........

sillyfrog, 2017-10-24:
There does not appear to be any incoming HTTP connections so my guess would be that there is a firewall running on the host. Can you check that all firewalls have been disabled and see how you go?

strasharo, 2017-10-24:
Same happened for me and the issue was that my iptables firewall was up. Flushed it and it all went fine.

christiamdc, 2017-10-24:
I turn off the firewall and try several times.. it works after the fifth time, I don't know why it failed the first four attempts.

Thank you for the support!

closed 2017-10-25 by sillyfrog