mirko/SonOTA #45: Sonoff S20 Wifi Settings - Device will reboot ....

zibous, 2017-11-04:
Hi,

I try to update the firmware with your script but the device can not find Wi-Fi "indebuurt1"  
and I can not change the settings. 

An update of the firmware with the USB adapter is not  possible because the 
TX pin is not working on the connector strip.

How can I set WLAN?

Alternative would be an update of the firmware with the USB interface, 
but I would need to know where I could still connect TX.

```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_2E9208, GroupTopic sonoffs) Version 5.8.0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-4616...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-4616...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute
00:00:33 APP: Restarting
---
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_2E9208, GroupTopic sonoffs) Version 5.8.0
00:00:00 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-4616...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-4616...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute
00:00:33 APP: Restarting
```

zibous, 2017-11-04:
Hi,

Sorry. solved. Have host computer set up a hotspot and so I could access the device and make the settings.

[image]

Super software, everything works fine without the devices open and without soldering.

So have updated 3 Sonoff S20 with the Tasmota software.
Many Thanks.

sillyfrog, 2017-11-10:
Glad it worked!

I'm thinking I may change the default included image so that it starts in AP mode, as this is not the first time people have had this type of issue.

closed 2017-11-10 by sillyfrog