mirko/SonOTA #47: Sonoff Touch not reaching FinalStage

djvandijk, 2017-11-06:
I've got a Sonoff Touch T1 UK 3 channel switch which is stuck after provisioning stage 2. I however don't see the FinalStage SSID coming up. Down below is the debug log. I also attached 2 images of the PCB itself. If anyone can point out where I can find GND/V3/etc so I can flash it using a cable that would also help
[image]
[image]

:

2017-11-06 19:38:55,175: DEBUG: Current IPs: ['192.168.1.64', '169.254.80.80', '172.16.80.1']
2017-11-06 19:38:59,114: INFO: Using the following configuration:
2017-11-06 19:38:59,114: INFO: 	Server IP Address: 192.168.1.64
2017-11-06 19:38:59,114: INFO: 	WiFi SSID: florimel
2017-11-06 19:38:59,114: INFO: 	WiFi Password: *******************
2017-11-06 19:38:59,114: INFO: Platform: win32
2017-11-06 19:38:59,182: INFO: ** Now connect via WiFi to your Sonoff device.
2017-11-06 19:38:59,182: INFO: ** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
2017-11-06 19:38:59,182: INFO: To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
2017-11-06 19:38:59,182: INFO: ** This application should be kept running and will wait until connected to the Sonoff...
2017-11-06 19:47:44,776: DEBUG: Current IPs: ['169.254.80.80', '172.16.80.1']
2017-11-06 19:47:53,392: DEBUG: Current IPs: ['169.254.80.80', '172.16.80.1', '169.254.45.211']
2017-11-06 19:48:01,940: DEBUG: Current IPs: ['10.10.7.2', '169.254.80.80', '172.16.80.1']
2017-11-06 19:48:01,978: DEBUG: ~~ Connection attempt
2017-11-06 19:48:01,978: DEBUG: >> HTTP GET /10.10.7.1/device
2017-11-06 19:48:02,009: DEBUG: << {
2017-11-06 19:48:02,009: DEBUG:     "deviceid": "1000199951",
2017-11-06 19:48:02,009: DEBUG:     "apikey": "e6ce914e-f026-4a6f-9059-c7e1bcbc8f37",
2017-11-06 19:48:02,009: DEBUG:     "accept": "post"
2017-11-06 19:48:02,009: DEBUG: }
2017-11-06 19:48:02,009: DEBUG: >> HTTP POST /10.10.7.1/ap
2017-11-06 19:48:02,009: DEBUG: >> {
2017-11-06 19:48:02,009: DEBUG:     "version": 4,
2017-11-06 19:48:02,009: DEBUG:     "ssid": "florimel",
2017-11-06 19:48:02,025: DEBUG:     "password": "*******************",
2017-11-06 19:48:02,025: DEBUG:     "serverName": "192.168.1.64",
2017-11-06 19:48:02,025: DEBUG:     "port": 8443
2017-11-06 19:48:02,025: DEBUG: }
2017-11-06 19:48:02,210: DEBUG: << {
2017-11-06 19:48:02,210: DEBUG:     "error": 0
2017-11-06 19:48:02,225: DEBUG: }
2017-11-06 19:48:02,225: INFO: ~~ Provisioning completed
2017-11-06 19:48:02,225: INFO: Starting stage2...
2017-11-06 19:48:02,309: INFO: ** The IP address of <serve_host> (192.168.1.64) is not assigned to any interface on this machine.
2017-11-06 19:48:02,309: INFO: ** Please change WiFi network to florimel and make sure 192.168.1.64 is being assigned to your WiFi interface.
2017-11-06 19:48:02,309: INFO: ** This application should be kept running and will wait until connected to the WiFi...
2017-11-06 19:48:19,000: DEBUG: Current IPs: ['192.168.1.64', '169.254.80.80', '172.16.80.1']
2017-11-06 19:48:19,000: INFO: ~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
2017-11-06 19:48:19,000: INFO: ~~ Waiting for device to connect
2017-11-06 19:48:19,069: INFO: *** IMPORTANT! ***
2017-11-06 19:48:19,069: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 19:48:19,069: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 19:48:19,069: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 19:48:19,069: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 19:48:28,260: DEBUG: << HTTP POST /dispatch/device
2017-11-06 19:48:28,260: DEBUG: >> /dispatch/device
2017-11-06 19:48:28,260: DEBUG: >> {
2017-11-06 19:48:28,260: DEBUG:     "error": 0,
2017-11-06 19:48:28,260: DEBUG:     "reason": "ok",
2017-11-06 19:48:28,275: DEBUG:     "IP": "192.168.1.64",
2017-11-06 19:48:28,275: DEBUG:     "port": 8443
2017-11-06 19:48:28,275: DEBUG: }
2017-11-06 19:48:28,275: INFO: 200 POST /dispatch/device (192.168.1.67) 15.67ms
2017-11-06 19:48:29,188: INFO: 101 GET /api/ws (192.168.1.67) 0.00ms
2017-11-06 19:48:29,188: DEBUG: << WEBSOCKET OPEN
2017-11-06 19:48:29,204: DEBUG: << WEBSOCKET INPUT
2017-11-06 19:48:29,204: DEBUG: << {
2017-11-06 19:48:29,204: DEBUG:     "userAgent": "device",
2017-11-06 19:48:29,204: DEBUG:     "apikey": "e6ce914e-f026-4a6f-9059-c7e1bcbc8f37",
2017-11-06 19:48:29,220: DEBUG:     "deviceid": "1000199951",
2017-11-06 19:48:29,220: DEBUG:     "action": "register",
2017-11-06 19:48:29,220: DEBUG:     "version": 2,
2017-11-06 19:48:29,220: DEBUG:     "romVersion": "2.0.1",
2017-11-06 19:48:29,220: DEBUG:     "model": "PSF-B04-GL",
[... 314 lines cut ...]
2017-11-06 19:51:27,065: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 19:51:27,065: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 19:52:27,329: DEBUG: ~~ websocket close
2017-11-06 19:52:29,699: INFO: *** IMPORTANT! ***
2017-11-06 19:52:29,699: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 19:52:29,699: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 19:52:29,715: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 19:52:29,716: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 19:52:31,790: DEBUG: Current IPs: ['169.254.80.80', '172.16.80.1']
2017-11-06 19:52:42,295: DEBUG: Current IPs: ['192.168.1.64', '169.254.80.80', '172.16.80.1']
2017-11-06 19:53:32,334: INFO: *** IMPORTANT! ***
2017-11-06 19:53:32,350: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 19:53:32,350: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 19:53:32,350: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 19:53:32,350: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 19:54:35,046: INFO: *** IMPORTANT! ***
2017-11-06 19:54:35,047: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 19:54:35,049: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 19:54:35,049: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 19:54:35,049: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-06 19:55:39,014: INFO: *** IMPORTANT! ***
2017-11-06 19:55:39,014: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-06 19:55:39,014: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-06 19:55:39,014: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-06 19:55:39,014: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.

sillyfrog, 2017-11-07:
I would suggest doing a hard reboot (or just disconnected from power for a few minutes), and see if final stage comes back after that, as the flashing process looks like it has gone well.

djvandijk, 2017-11-08:
I had it disconnected for the entire night, but it doesn't reach final stage after coming back. 'Something' is flashed though, since I also can't get the Sonoff Touch into AP mode anymore.
When I turn the device on after about 20 seconds I see the wifi led blinking for 10 seconds, followed by 2 seconds of staying on and then the device seems to reboot and the sequence is repeated.
I can reboot it by pressing button 1 for 4 seconds, but this results in the same behaviour. Any more ideas?

sillyfrog, 2017-11-09:
Have you tried pressing the button 4 times, or setting up a default AP? See the end of the "Running" section at: https://github.com/mirko/SonOTA#running - it maybe one of the later runs worked

djvandijk, 2017-11-09:
I managed to get it into AP mode somehow (not sure how, but many presses and more presses, more then 4). It broadcasted an SSID of SonoffXXXX (not sure which number).
I entered my Wifi details, saved and restarted, but it doesn't seem to connect to my Wifi afterwards (I tried 2 times). 
Upon connecting to the Touch I see the landing page which tells me there is a minimalistic version installed and that I should upgrade tasmota.

djvandijk, 2017-11-09:
In addition to the previous the SSID is: sonoff-1970 and it does seem to be 4 presses, but I am unsure of the timing.

djvandijk, 2017-11-09:
That's sonoff-1790 (not 1970)

sillyfrog, 2017-11-10:
So it looks like Sonoff-Tasmota is installed OK, but just needs to be configured, I have some basic directions at https://github.com/mirko/SonOTA#running . That also includes some links to the Sonoff-Tasmota page.

The included binary is the sonoff.bin from https://github.com/arendst/Sonoff-Tasmota/releases , so it should match the Sonoff-Tasmota documentation.

Good luck!

djvandijk, 2017-11-11:
Problem is that I can enter the configuration of my home WiFi  when using AP mode this configuration doesn't seem to be saved as the device won't connect to the WiFi afterwards. The AP mode also doesn't provide me the option to update the firmware.

djvandijk, 2017-11-11:
Managed to get it working by setting up the temporary wifi again. Apparently the AP mode shows a different interface which doesn't allow you to upgrade the firmware. When connected to the temporary wifi I could upgrade the firmware. Case closed.

closed 2017-11-12 by sillyfrog