mirko/SonOTA #48: SonOff POW does not work after flashing

christiamdc, 2017-11-09:
- Operating System: Windows
- Python Version: Python 3.5
- Sonoff model: SOnOff POW

Hi.

I have successfully flashed SonOff basic and SonOff S20. I tried flashing SonOff POW, the process finished successfully but now the device does not respond, the wifi led just blink once when I powered it on but no more blinks after that. Do you know what could it be? I tried then with another SonOff POW device but with the same result.

Thanks you in advance for the help.

Christiam

sillyfrog, 2017-11-10:
If it has finished you should be able to put it into AP mode by pressing the button 4 times, the alternative is to configure an AP with the default credentials, see the end of the running section here: https://github.com/mirko/SonOTA#running

christiamdc, 2017-11-10:
Thanks for your reply. 

I did that as i did for the sonoff basic and S20, there was no problem putting those devices into AP mode. But it is not the case for these 2 SonOff POW. When I turn on the power the wifi led blink once and then it is dead. I tried pressing the button 4 times but still the led did not blink. In the other hand, the on off button does work.

turbolift, 2017-12-23:
I had the same problem. The parameter --legacy solved my problem.

Trago5, 2018-01-12:
I flashed today several Sonoffs and one POW ended up in this state after FinalStage. The _sonoff-####_ network didn't show up. However, when I later plugged it in and patiently waited over ten minutes, this network suddenly appeared and I managed to finalize the configuration.

This problematic unit was previously upgraded to firmware 2.0.4 in eWeLink. I also ran SonOTA without any parameters.

I was able to successfully flash another POW from firmware 2.0.2 in Linux using sudo and parameters --legacy and --slowstream. Now the sonoff-#### network appeared right after my laptop had disconnected from FinalStage network.

sillyfrog, 2018-01-28:
Thanks for the feedback - glad it worked!

closed 2018-01-28 by sillyfrog