mirko/SonOTA #50: Sonoff POW red light intermittent relay

adamrobinson361, 2017-11-11:
- Operating System: Mac
- Python Version: 3.6.0
- Sonoff model: Sonoff Pow

I have recently tried to flash the sonoff pow using this process. The script seemed to work fine, having completed the *final stage* but the device does not show up in wifi list nor will it join an indebuurt1 guest network I've set up with the credentials you outlined. 

The led just remains red with no flashing and the device intermittently clicks as if the relay is moving. I've tried all of the holding of buttons and button presses but the device doesn't seem to want to make the led flash on and off. 

Logs attached - any help appreciated. Project looks like a great idea!

[debug_1510270913.log](https://github.com/mirko/SonOTA/files/1463711/debug_1510270913.log)

adamrobinson361, 2017-11-13:
@sillyfrog - any ideas on this. Is the bold red light not flashing something you've come accross before?

Regards,

Adam

sillyfrog, 2017-11-16:
@adamrobinson361 hmmm, it's not something I have seen (that said, I don't have a POW either, but the other devices are similar).

Does a single button press flip the relay?

Also worth checking for the `sonoff-####` AP, as based on the timing it was about when I changed the default image to start with an AP rather than in the (default) WPS mode.

tali1974, 2018-04-09:
@adamrobinson361 Hi.. have the same issue on one of my POW, how did you fix this ?

mjhartley, 2018-12-02:
@adamrobinson361 Hi. I'm having the same issue. Did you get it resolved?