mirko/SonOTA #51: Final Stage does not show up

jbbr, 2017-11-11:
### What I have done
- put sonoff to pairing mode, connected to ITEAD- wifi
- Started `./sonota.py` without parameters, typed wifi SSID/password
- After provisioning switched back to home wifi, firmware upload seems to complete without errors

- **A few seconds after the last `WEBSOCKET INPUT` message without error I get some 404s:**
```
2017-11-11 14:55:41,703: WARNING: 404 GET /sockjs-node/info?t=1510408541677 (::1) 1.44ms
2017-11-11 14:55:41,706: WARNING: 404 GET /sockjs-node/info?t=1510408541678 (::1) 0.71ms
2017-11-11 14:55:41,708: WARNING: 404 GET /sockjs-node/info?t=1510408541676 (::1) 0.59ms
...
```

### Problem
- "Final Stage SSID" never shows up
- When keeping the sonota process running, the upload starts again with the same symptoms (probably related to 2c93c1994d2224ee5979114729b4a7d7f94414a5)

This _might_ be related to #21, but it looks different to me as I don't get any 404's during the image uploads

### System
- Operating System: Arch Linux x64
- Python Version: 3.6.3
- Sonoff model: S20
- SonOTA latest master 2c93c1994d2224ee5979114729b4a7d7f94414a5

### Attachments
[debug_1510408005.log](https://github.com/mirko/SonOTA/files/1463780/debug_1510408005.log)

jbbr, 2017-11-11:
I now checked out 736bd1df3ad9cb9a9f95baa825930a0de8ab4482 and flashing went fine without any issues.
Maybe 2c93c1994d2224ee5979114729b4a7d7f94414a5 or some commit in between broke the flashing process for the S20.

Debug log for the successful flash with 736bd1df3ad9cb9a9f95baa825930a0de8ab4482: [debug_1510411362.log](https://github.com/mirko/SonOTA/files/1463809/debug_1510411362.log)

sillyfrog, 2017-11-11:
Woah, that's really weird as the Sonoff is downloading `image_user2-0x81000.bin` twice at the same time.

The `404 GET /sockjs-node/info?t=1510408541676` can be ignored as that's just something on localhost (note the `(::1)`).

I have pushed an update that should help sort that sending issue (I was not flushing the packets correctly, and hardware acceleration may have kicked in.)

Thanks for sending me the logs, if you are doing any more devices it would be good to confirm if the latest build now works for your setup.

Cheers.

closed 2017-11-20 by sillyfrog

alfem, 2018-08-12:
I have flashed three Sonoffs Basic sucessfully, but the last one did not show the Final Stage SSID. After a long time I reset it and now do not show any SSID. I do not know what to do.

alfem, 2018-09-07:
Another try with a brand new Sonoff Basic. And another _brick_ in the wall

No "Final Stage SSID" again. Here is the log:

[flash-failed.log](https://github.com/mirko/SonOTA/files/2361829/flash-failed.log)

alfem, 2019-09-13:
I'm afraid SonOTA does not work anymore. It worked with old firmware, but Itead changed something in Sonoffs to avoid this 'trick'. Right now I only know of two alternatives:

- Flash your sonoff using [using wires](https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation) (soldering headers, or holding them carefully) 
- Buy one of the new R3 model with [DIY mode](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY) included