mirko/SonOTA #52: SONOF RF Bridge WiFi 433 MHz

wfa001, 2017-11-12:
- Operating System: Windows 10
- Python Version: 3.5.4
- Sonoff model: SONOFF RF Bridge WiFi 433 MHz 

While trying to flash the Sonoff RF Bridge Device the process will not start stage 2 - FinalStage never shows up. Has anyone already successfully tried to flash this type of device?

[debug_1510487306.log](https://github.com/mirko/SonOTA/files/1464584/debug_1510487306.log)

sillyfrog, 2017-11-12:
Looking at that log, my first guess would be the firewall - it's very common an issue on Windows. If you completely disable it for doing this flashing, things are likely to work.

Good luck!

wfa001, 2017-11-12:
with the given setup i successfully flashed a sonoff touch without any problems. Firewall is switched off. Maybe the rf bridge needs different handling?

sillyfrog, 2017-11-12:
I would suggest getting the latest build (as there have been a few important changes). But for this error try running with `--legacy` (and if that does not work, try updating to the latest Sonoff release using the eWeLink app first).

closed 2017-11-20 by sillyfrog