mirko/SonOTA #53: FinalStage does not show up

Siggemada, 2017-11-12:
Just trying this for the first time so excuse me if I did something simply wrong:

- Operating System: 
Ubuntu 17.10
- Python Version: 
3.5.3
- Sonoff model: 
model: "PSA-BHA-GL",
version: "23.42.5"

Please also drag and drop the log file to the issue so it is attached.

[sonota_error.log](https://github.com/mirko/SonOTA/files/1464711/sonota_error.log)

Siggemada, 2017-11-12:
And the relay starts clicking every loop. It sounds like a double click....

Siggemada, 2017-11-12:
Model marking inside device:
Sonoff TH10/16
Ver 2.1
2017-6-8

rlust, 2017-11-12:
I am having the same issue:
OS OSX
Python 3.6
SonOff Basic

Never get Final Version, looped during instal and never stopped
[debug_1510501076.log](https://github.com/mirko/SonOTA/files/1464866/debug_1510501076.log)
l

Siggemada, 2017-11-12:
I tried to OTA wirh Windows 7 instead. No luck:

[sonota_error_win.txt](https://github.com/mirko/SonOTA/files/1464919/sonota_error_win.txt)

sillyfrog, 2017-11-12:
I have pushed a change that will revert how files are served, which should fix this. I need to figure out a way to make it automatic (one day). If you can pull the latest release and try again, that would be great.

NameLessJedi, 2017-11-13:
Not sure if that's same problem but FinalStage never appeared for me. And in the process I might bricked two Sonoff Basic's :/ (I'll try with serial cable later but at the moment both look like dead ; no LED blinking, no reaction to push button). I had problem yesterday and after today's fix one Basic flashed perfectly. Couple of hours later tried again on another and waited for FinalStage for like 20min and it didn't show up
[debug_1510584791.log](https://github.com/mirko/SonOTA/files/1467457/debug_1510584791.log)

Siggemada, 2017-11-13:
I am sorry but I went ahead and flashed with cables. I have one more device will try with that one.

sillyfrog, 2017-11-13:
@NameLessJedi That's weird, it stopped downloading after the first chunk. You could try a power cycle of the Sonoff, then run again with `--no-prov`.

NameLessJedi, 2017-11-14:
Unfortunately no dice :/ There is no FinalStage SSID and sonota.py just waits.
iwscan does not report any ITEAD-xxx, FinalStage or any new SSIDs when I power on Sonoff

Siggemada, 2017-11-14:
If you pull down GPIO0 (in my case by simply pushing the button) to earth before applying 3,3 v and keeping it to ground, the esp8266 boots in flash mode. This should not be affected by this upload. Simply upload the new sketch at this point.

wavemaking, 2017-11-14:
Also for me FinalStage does not appear. Everything else before seems to work without trouble. Debug log is attached. I noted that while waiting for the FinalStage network to appear, I can actually see the old "ITEAD-100007c38d" network again. What does that mean? I also tried waiting much longer, or power cycling the Sonoff with no visible changes.

- Operating System: Ubuntu 16.04
- Python Version: 3.5.2
- Sonoff model: POW (Not sure which version; but I hope it can be easily identified with the attached photo)

[debug_1510702658.log](https://github.com/mirko/SonOTA/files/1472924/debug_1510702658.log)

[image]

sillyfrog, 2017-11-15:
Looking at that log, it appears that the Sonoff has not reconnected to download.
Couple of things to try:

- Ensure all firewalls are off when back on your LAN
- Try removing the second IP address thats on the same LAN
- Try running with `--legacy`

Getting a `tcpdump` will also be of value if all of the above don't work.

NameLessJedi, 2017-11-15:
@Siggemada thanks. This did the trick. Soon I'll have POW version and I'll try again :)

wavemaking, 2017-11-15:
sillyfrog, a combination of bullets 2 (disabling the wired ethernet connection) and 3 (--legacy) did it for me. Thanks!

closed 2017-11-15 by sillyfrog