mirko/SonOTA #54: Sonoff Touch rebooted during flash

mikeS7, 2017-11-12:
- Operating System: 
Mac OS X 10.12.6
- Python Version: 
Python 3.6.3
- Sonoff model: 
Sonoff Touch Wall Switch
Please also drag and drop the log file to the issue so it is attached.

I successfully flashed Sonoff 4CH Pro.
But Sonoff Touch rebooted itself during flashing and uploading started again, after several restarts I got message about FinalStage, but never got such SSID.

Unfortunately I don't have logs from first attempt.

Now Sonoff is connected to my network.
I tried to restart the process but got the exception.

`Slow Sending file: /ota/image_user2-0x81000.bin (This may take several minutes)
Uncaught exception GET /ota/image_user2-0x81000.bin?deviceid=10000e6c51&ts=561035572&sign=655579ab8f760c8eade3d0303b86f2deeb9bf3d2c152003c4c18e5cd0d69d898 (192.168.7.35)
HTTPServerRequest(protocol='http', host='dl.itead.cn', method='GET', uri='/ota/image_user2-0x81000.bin?deviceid=10000e6c51&ts=561035572&sign=655579ab8f760c8eade3d0303b86f2deeb9bf3d2c152003c4c18e5cd0d69d898', version='HTTP/1.1', remote_ip='192.168.7.35', headers={'Host': 'dl.itead.cn', 'Range': 'bytes=0-4095', 'User-Agent': 'itead-device'})
Traceback (most recent call last):
  File "/Users/mike7/Library/Python/3.6/lib/python/site-packages/tornado/web.py", line 1511, in _execute
    result = yield result
  File "/Users/mike7/Library/Python/3.6/lib/python/site-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/Users/mike7/Library/Python/3.6/lib/python/site-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "<string>", line 4, in raise_exc_info
  File "/Users/mike7/Library/Python/3.6/lib/python/site-packages/tornado/gen.py", line 1063, in run
    yielded = self.gen.throw(*exc_info)
  File "sonota.py", line 152, in get
    yield self.flush()
  File "/Users/mike7/Library/Python/3.6/lib/python/site-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/Users/mike7/Library/Python/3.6/lib/python/site-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "<string>", line 4, in raise_exc_info
tornado.iostream.StreamClosedError: Stream is closed
Cannot send error response after headers written
200 GET /ota/image_user2-0x81000.bin?deviceid=10000e6c51&ts=561035572&sign=655579ab8f760c8eade3d0303b86f2deeb9bf3d2c152003c4c18e5cd0d69d898 (192.168.7.35) 14562.80ms`

log is attached
[SonoffTouch-flash.txt](https://github.com/mirko/SonOTA/files/1464791/SonoffTouch-flash.txt)

As I understand SonOTA can't store firmware

mikeS7, 2017-11-12:
Relay clicks on every store attempt, may be the same way as #53

sillyfrog, 2017-11-12:
@mikeS7 Can you try with the latest build, there have been a few changes to how things are served. If that does not work, also try running with `sudo ./sonota.py --legacy`.

mikeS7, 2017-11-13:
Got it working now, thanks a lot!
Log is attached
[SonoffTouch-flash-good.txt](https://github.com/mirko/SonOTA/files/1465795/SonoffTouch-flash-good.txt)

closed 2017-11-13 by mikeS7