mirko/SonOTA #56: Sonoff POW 2.0.2

Xeroxxx, 2017-11-13:
For whatever reason I'm not allowed to edit this page: https://github.com/mirko/SonOTA/wiki

Sonoff POW / 2.0.2 / requires --legacy and --slowstream

Cheers

christiamdc, 2017-11-15:
Hi.

I ran the sonota process for POW 2.04. The process finished ok, but something it is wrong because the wifi led (blue light) only blink once when I turn on the device but after that it seems dead, the on/off button works though.

Do you have any idea what could be happening?. Maybe I need to run the process without upgrading the Sonoff firmware to 2.04.

I'll appreciate any advice.

regards

sillyfrog, 2017-11-15:
Can you please post the log?  Did you download the SonOTA build in the last 24 hours? If so, you should have a `sonoff-####` AP that you can connect to and configure the device - in which case it worked :).

christiamdc, 2017-11-15:
Thanks for answer.

Sorry, I don't have the log because all the steps well fine until de final message "all done" :(. The problem is that the device doesn't work after flashing it. I can turn it on and off using the button but the wifi lead does not blink. The SonOff never enter in configuration mode and the AP never shows.

I ran SonOta for S20 and basic models without no problem. The AP sonoff-### is visible and works fine.

Regards

sillyfrog, 2017-11-15:
Sounds like it's working if you have the AP :) . If you connect to the AP, you can configure Sonoff-Tasmota to connect to your network.

Should it be needed, there is a log file saved for each run in the same directory.

closed 2017-11-15 by sillyfrog

christiamdc, 2017-11-15:
Sorry for the misunderstanding. I meant that the AP showed for basic and S20 models. But it is not the case for the POW model. For this model, the AP never shows. I press the button 4 times but the wifi led does not blink and the AP never appears. I also tried configuring the default SSID but the device never connected.

Thanks for your instructions. I found the log. Here it is:

2017-11-09 18:04:11,624: DEBUG: Current IPs: ['192.168.1.6']
2017-11-09 18:04:28,886: INFO: Using the following configuration:
2017-11-09 18:04:28,888: INFO: 	Server IP Address: 192.168.1.6
2017-11-09 18:04:28,889: INFO: 	WiFi SSID: Bettycl
2017-11-09 18:04:28,890: INFO: 	WiFi Password: ********
2017-11-09 18:04:28,892: INFO: Platform: win32
2017-11-09 18:04:29,037: INFO: ** Now connect via WiFi to your Sonoff device.
2017-11-09 18:04:29,039: INFO: ** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
2017-11-09 18:04:29,040: INFO: To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
2017-11-09 18:04:29,041: INFO: ** This application should be kept running and will wait until connected to the Sonoff...
2017-11-09 18:05:11,892: DEBUG: Current IPs: []
2017-11-09 18:05:12,234: DEBUG: Current IPs: ['10.10.7.2']
2017-11-09 18:05:12,236: DEBUG: ~~ Connection attempt
2017-11-09 18:05:12,239: DEBUG: >> HTTP GET /10.10.7.1/device
2017-11-09 18:05:12,343: DEBUG: << {
2017-11-09 18:05:12,344: DEBUG:     "apikey": "bc447220-4f43-42b3-8f98-da421df18761",
2017-11-09 18:05:12,347: DEBUG:     "accept": "post",
2017-11-09 18:05:12,348: DEBUG:     "deviceid": "10000bad11"
2017-11-09 18:05:12,349: DEBUG: }
2017-11-09 18:05:12,350: DEBUG: >> HTTP POST /10.10.7.1/ap
2017-11-09 18:05:12,351: DEBUG: >> {
2017-11-09 18:05:12,352: DEBUG:     "port": 8443,
2017-11-09 18:05:12,353: DEBUG:     "password": "********",
2017-11-09 18:05:12,354: DEBUG:     "version": 4,
2017-11-09 18:05:12,356: DEBUG:     "serverName": "192.168.1.6",
2017-11-09 18:05:12,357: DEBUG:     "ssid": "Bettycl"
2017-11-09 18:05:12,358: DEBUG: }
2017-11-09 18:05:12,465: DEBUG: << {
2017-11-09 18:05:12,466: DEBUG:     "error": 0
2017-11-09 18:05:12,467: DEBUG: }
2017-11-09 18:05:12,468: INFO: ~~ Provisioning completed
2017-11-09 18:05:12,470: INFO: Starting stage2...
2017-11-09 18:05:12,547: INFO: ** The IP address of <serve_host> (192.168.1.6) is not assigned to any interface on this machine.
2017-11-09 18:05:12,548: INFO: ** Please change WiFi network to Bettycl and make sure 192.168.1.6 is being assigned to your WiFi interface.
2017-11-09 18:05:12,550: INFO: ** This application should be kept running and will wait until connected to the WiFi...
2017-11-09 18:05:18,953: DEBUG: Current IPs: ['192.168.1.6']
2017-11-09 18:05:18,955: INFO: ~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
2017-11-09 18:05:18,966: INFO: ~~ Waiting for device to connect
2017-11-09 18:05:19,066: INFO: *** IMPORTANT! ***
2017-11-09 18:05:19,067: INFO: ** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
2017-11-09 18:05:19,069: INFO: ** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
2017-11-09 18:05:19,070: INFO: This server should automatically be allocated the IP address: 192.168.4.2.
2017-11-09 18:05:19,072: INFO: If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
2017-11-09 18:05:32,025: DEBUG: << HTTP POST /dispatch/device
2017-11-09 18:05:32,026: DEBUG: >> /dispatch/device
2017-11-09 18:05:32,027: DEBUG: >> {
2017-11-09 18:05:32,028: DEBUG:     "error": 0,
2017-11-09 18:05:32,029: DEBUG:     "port": 8443,
2017-11-09 18:05:32,030: DEBUG:     "reason": "ok",
2017-11-09 18:05:32,031: DEBUG:     "IP": "192.168.1.6"
2017-11-09 18:05:32,032: DEBUG: }
2017-11-09 18:05:32,034: INFO: 200 POST /dispatch/device (192.168.1.9) 10.00ms
2017-11-09 18:05:32,261: INFO: 101 GET /api/ws (192.168.1.9) 1.00ms
2017-11-09 18:05:32,262: DEBUG: << WEBSOCKET OPEN
2017-11-09 18:05:32,268: DEBUG: << WEBSOCKET INPUT
2017-11-09 18:05:32,269: DEBUG: << {
2017-11-09 18:05:32,270: DEBUG:     "ts": 703,
2017-11-09 18:05:32,272: DEBUG:     "model": "PSC-B01-GL",
2017-11-09 18:05:32,273: DEBUG:     "apikey": "bc447220-4f43-42b3-8f98-da421df18761",
2017-11-09 18:05:32,274: DEBUG:     "deviceid": "10000bad11",
2017-11-09 18:05:32,275: DEBUG:     "userAgent": "device",
2017-11-09 18:05:32,277: DEBUG:     "action": "register",
2017-11-09 18:05:32,278: DEBUG:     "version": 2,
2017-11-09 18:05:32,278: DEBUG:     "romVersion": "2.0.4"
2017-11-09 18:05:32,280: DEBUG: }
2017-11-09 18:05:32,281: DEBUG: ~~~ device sent action request, acknowledging / answering...
[... 296 lines cut ...]
2017-11-09 18:05:42,568: DEBUG:     ]
2017-11-09 18:05:42,568: DEBUG: }
2017-11-09 18:05:42,568: DEBUG: ~~~ device sent action request, acknowledging / answering...
2017-11-09 18:05:42,569: DEBUG: ~~~~ query
2017-11-09 18:05:42,570: DEBUG: >> {
2017-11-09 18:05:42,570: DEBUG:     "error": 0,
2017-11-09 18:05:42,571: DEBUG:     "apikey": "8f73bdab-6c8c-4148-976c-50f91f27e077",
2017-11-09 18:05:42,571: DEBUG:     "params": 0,
2017-11-09 18:05:42,572: DEBUG:     "deviceid": "10000bad11"
2017-11-09 18:05:42,572: DEBUG: }
2017-11-09 18:06:21,364: INFO: *** IMPORTANT! ***
[... cut ...]
2017-11-09 18:06:36,140: INFO: Once "FinalStage" has gone away, you can stop this program
2017-11-09 18:06:39,593: DEBUG: Sending file: /ota/image_user2-0x81000.bin
2017-11-09 18:06:47,121: INFO: 200 GET /ota/image_user2-0x81000.bin (192.168.4.1) 7531.43ms
2017-11-09 18:06:59,769: DEBUG: Sending file: /ota/image_arduino.bin
2017-11-09 18:07:08,710: INFO: 200 GET /ota/image_arduino.bin (192.168.4.1) 8944.51ms
2017-11-09 18:07:24,085: DEBUG: Current IPs: ['192.168.1.6']
2017-11-09 18:07:24,086: INFO: No longer on "FinalStage" SSID, all done!

sillyfrog, 2017-11-15:
I'm assuming a long press (7-10 seconds) does not do anything?

That's weird because as you say, it all looks to have completed OK.