mirko/SonOTA #57: OSError: [Errno 0] Error

domidollar, 2017-11-17:
- Operating System: Windows 7
- Python Version: 3.5
- Sonoff model: Basic
[log.txt](https://github.com/mirko/SonOTA/files/1483619/log.txt)

File "C:\Users\doqu\AppData\Local\Programs\Python\Python37-32\lib\ssl.py", lin
e 1075, in do_handshake
    self._sslobj.do_handshake()
  File "C:\Users\doqu\AppData\Local\Programs\Python\Python37-32\lib\ssl.py", lin
e 696, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error

Any idea?

domidollar, 2017-11-17:
Okay i think i know it. I updated the Sonoff to 3.6.1. 

It is possible to downgrad the Sonoff?

sillyfrog, 2017-11-17:
I’m not not sure how Python does it’s folders on Windows, but I would suggest moving Users\doqu\AppData\Local\Programs\Python\Python37-32 sideways and try again. I *think* it’s trying to use a Python 3.7 library in 3.5.

domidollar, 2017-11-17:
I uninstalled  the Python 3.7. but same isue.

Then i take another sonoff basic with FW3 .5.5 and it works without isues.

Thank you for your help and great work!!!

sillyfrog, 2017-11-18:
Do you have the logs from the successful run? I'd be keen to see it as the v3 firmware is very new and I don't have access to it (yet at least), it would be good to get a feel if it will be an issue in the future.
Thanks!

domidollar, 2017-11-18:
I'm sorry. I mean Fw 1.5.5, not 3.5.5 🙈 

In my first comment ist it the Fw 1.6.1...

sillyfrog, 2017-11-19:
Ahhh! :) That makes more sense. Cheers!

closed 2017-11-19 by sillyfrog