mirko/SonOTA #59: Confused about the steps

jonasbager, 2017-11-22:
- Operating System: OSX
- Python Version: 3.6.3
- Sonoff model: Basic

I can't seem to make this process work. Don't quite know if I'm doing it wrong. I installed python and got the script running fine, but it seems to mess up before the step of the "Final Stage".

What I did was, I ran the script, Selected my Mac's IP as the IP, set the wifi details of the same network and then set the Sonoff into pairing. Then connected to the ITEAD wifi, and from there on I get stuck. 

Anything I missed?

If someone could make a step-by-step, it would be sooo awesome! Apologies for being a rookie in this area

Thanks!

sillyfrog, 2017-11-22:
Can you attach the .log file from the run (drag and drop it onto the text field here). It should be promoting you as to what to do for each step.

jonasbager, 2017-11-23:
[debug_1511304227.log](https://github.com/mirko/SonOTA/files/1498571/debug_1511304227.log)

sillyfrog, 2017-11-23:
This is the important part (the last log line keeps repeating in another thread).

```2017-11-21 23:45:51,109: DEBUG: ~~~ device acknowledged our action request (seq 1511304347621) with error code 404
2017-11-21 23:45:51,110: ERROR: *************************************************
2017-11-21 23:45:51,110: ERROR: Received a 404 error, try running with '--slowstream' option.
2017-11-21 23:45:51,110: ERROR: *************************************************```

The Sonoff you have needs an alternative method of receiving the file, so if you do exactly the same thing again, but run it with `--slowstream`, it should work.

Good luck!

jonasbager, 2017-11-23:
IT WORKED!!! THIS METHOD IS SO FUCKING AMAZING. You won't believe how long I've been trying to get Sonoff flashed the normal way, without any luck. 

Does this method work, if I have some Sonoff where I've already tried the normal way, but failed, since the now probably won't have the standard firmware on them?

sillyfrog, 2017-11-25:
Unfortunately not, if you can't connect to an AP/get in on your WiFi, you are up the creek unfortunately. I would suggest checking out the various pages here: https://github.com/arendst/Sonoff-Tasmota/wiki as there is a lot of info there for the serial way.

Good luck!

closed 2017-11-25 by sillyfrog