mirko/SonOTA #60: Stopped sending image_user2-0x81000.bin

jchidley, 2017-11-22:
I can't do anything with it now.  

Using the latest versions of your software and a Sonoff basic sealed with "2017 8 20"

I was using Windows - having turned off all the firewalls and using it's Ethernet connected address for serving, whilst using everything else over WiFi.

````
Sending file: /ota/image_user2-0x81000.bin
206 GET /ota/image_user2-0x81000.bin?deviceid=100015be37&ts=2002583103&sign=9f63bb03cc5e84b762f65c6ed44a6c4264770b103194c00c3cd1ac03ed2e6fc0 (192.168.1.31) 2.71ms
<< WEBSOCKET INPUT
<< {
    "apikey": "54061ec0-dc79-4e66-af89-244502a5040d",
    "userAgent": "device",
    "error": 0,
    "deviceid": "100015be37",
    "sequence": "1511348990410"
}
~~~ device acknowledged our action request (seq 1511348990410) with error code 0
........................
*
````

It is difficult to program these things.  I have tried to program them over serial too.  I wonder if they've changed some of the components - like memory?

I have ordered a new USB/UART cable (http://www.ftdichip.com/Support/Documents/DataSheets/Cables/DS_C232HD_UART_CABLE.pdf ) to eliminate that.  

Is there a way to specify everything at the command line and avoid all the checks?  

I use a Windows main machine (including ubuntu on it), several Raspberry Pis and an Arch Linux box I can boot up too.  It's a great way to eliminate the various problems you get...

sillyfrog, 2017-11-23:
Do you have the `debug_######.log` file? There maybe something else there as well.

There are various options `--no-prov` and `--no-check-ip` to get things going at different stages / skip checks. See the output of `./sonota.py -h` for more details.

jchidley, 2017-11-23:
No, stupidly I deleted those.

jchidley, 2017-11-23:
My new UART arrived today, and I managed to flash the firmware onto it using that.  One problem solved but there's a problem booting it.

jchidley, 2017-11-23:
I think I found the OTA device: it's booted up and this is where it's got to:

````
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

00:00:00 CFG: Use defaults
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_57E9F5, GroupTopic sonoffs) Version 5.9.1
00:00:00 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-2549...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-2549...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute
00:00:33 APP: Restarting
````

Which means, I think, that I must have missed a step of the process somewhere, or that it hasn't been configured correctly to look for my own network.

jchidley, 2017-11-23:
I created one of the access points (indeburrt1) and then connected to the device's web page (because I could see the IP address from the serial output:  

````
00:00:07 HTP: Web server active on sonoff-2549 with IP address 192.168.43.235
````

The web page tells me that it's a minimal configuration and needs an upload.  It's not clear what files I should be uploading to it, perhaps it is on a Wiki somewhere?  I also tried to do the whole button press thing to configure via ESP8266 SmartConfig but that didn't work.

sillyfrog, 2017-11-24:
So that all looks like what I would expect from an older release. When did you last download? Did you replace the `image_arduino.bin` file by chance?

It will not keep the configuration as it is completely rewriting the flash, so it does need to be configured. However it should be a full build (certainly has everything I need), so not sure what that is about.

The [Sonoff-Tasmota Initial-Configuration](https://github.com/arendst/Sonoff-Tasmota/wiki/Initial-Configuration) page has some overview as to how to configure it for the first time now it's installed.

closed 2017-11-24 by sillyfrog