mirko/SonOTA #61: Beginner Problems and Fix

reloxx13, 2017-11-26:
- Operating System: Windows 10 x64
- Python Version: Python 3.6.2
- Sonoff model: sOnOff Basic

Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.

Hey,

im new to this project and this is my beginner expierience and how i fixed them:
Python was installed already on my machine, so i just started your install command and got following output:

```
C:\xampp\htdocs\SonOTA>pip3 install --user -r requirements.txt
Requirement already satisfied: httplib2>=0.10.3 in c:\users\xxx\appdata\roaming\python\python36\site-packages (from -r requirements.txt (line 1))
Collecting netifaces>=0.10.6 (from -r requirements.txt (line 2))
  Using cached netifaces-0.10.6.tar.gz
Collecting tornado>=4.5.1 (from -r requirements.txt (line 3))
  Using cached tornado-4.5.2-cp36-cp36m-win_amd64.whl
Installing collected packages: netifaces, tornado
  Running setup.py install for netifaces ... error
    Complete output from command c:\python36\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\Xxx\\AppData\\Local\\Temp\\pip-build-miff9toj\\netifaces\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\Nils\AppData\Local\Temp\pip-bqll4rwm-record\install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_ext
    building 'netifaces' extension
    error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": http://landinghub.visualstudio.com/visual-cpp-build-tools

    ----------------------------------------
Command "c:\python36\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\Nils\\AppData\\Local\\Temp\\pip-build-miff9toj\\netifaces\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\Nils\AppData\Local\Temp\pip-bqll4rwm-record\install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in C:\Users\Nils\AppData\Local\Temp\pip-build-miff9toj\netifaces\
```

I had to install Visual C++ 2015 Build Tools from here: http://landinghub.visualstudio.com/visual-cpp-build-tools
While installing, i choosed everything to install from this setup (9gb space req.).

After that i could instal netifaces successfully.

Next i got this error message:

```
Starting stage2...
** The IP address of <serve_host> (192.168.178.30) is not assigned to any interface on this machine.
** Please change WiFi network to HomeZone and make sure 192.168.178.30 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
...........Current IPs: ['192.168.178.30', '192.168.52.1', '192.168.142.1']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Traceback (most recent call last):
  File "sonota.py", line 723, in <module>
    main()
  File "sonota.py", line 719, in main
    stage2()
  File "sonota.py", line 642, in stage2
    app.listen(DEFAULT_PORT_HTTP)
  File "C:\Users\Nils\AppData\Roaming\Python\Python36\site-packages\tornado\web.py", line 1943, in listen
    server.listen(port, address)
  File "C:\Users\Nils\AppData\Roaming\Python\Python36\site-packages\tornado\tcpserver.py", line 142, in listen
    sockets = bind_sockets(port, address=address)
  File "C:\Users\Nils\AppData\Roaming\Python\Python36\site-packages\tornado\netutil.py", line 197, in bind_sockets
    sock.bind(sockaddr)
OSError: [WinError 10048] Normalerweise darf jede Socketadresse (Protokoll, Netzwerkadresse oder Anschluss) nur jeweils einmal verwendet werden
```
OSError: [WinError 10048] Normalerweise darf jede Socketadresse (Protokoll, Netzwerkadresse oder Anschluss) nur jeweils einmal verwendet werden
Translated: Usually every Socket (Protocol, Networkaddress or Port) can only be used once.

This was caused by my xampp running on my local machined as service on 8080 and 80.
After is Stoped them i could run through the setup.

i would suggest to add some hints in the readme for this problems or chec the ports at the beginning.

Thanks for this OTA Project guys, saved a lot of time for me :)

pavelbacka, 2017-12-30:
I am having same issue? Any hint?

pavelbacka, 2017-12-30:
So I changed DEFAULT_PORT_HTTP 80 from 8080 and commented out app.listen(DEFAULT_PORT_HTTPS). So I am not getting this message anymore, but still no success.

sillyfrog, 2018-01-28:
Thanks for the feedback - hopefully helpful for others (and your exe will help as well of course :) )

Cheers.

closed 2018-01-28 by sillyfrog