mirko/SonOTA #65: Endless Finalstage

JC38, 2017-12-01:
- Operating System: Win 10 x64
- Python Version: 3.5.x
- Sonoff model: S20 
[debug_1512163013.log](https://github.com/mirko/SonOTA/files/1523008/debug_1512163013.log)

Hello,

I've tried to upload your custom firwmare in many steps and S20 reboot.
Now, the S20 always shows the Finalstage SSID, i launch the script with --no-prov option while my Windows firewall is stopped for every kind of network (domain, private and public). And i also set the Finalstage into a private network.

Now, when the S20 starts, the led is flashing green 20-30 times , then switch off for 3-4 seconds and switch on green for 3-4 seconds. This 2 last steps are cycling.

Is it possible to restart from scratch in OTA ? 

Thank you for your time and your work

sillyfrog, 2017-12-02:
That sounds like it should be trying... My suggestion of things to test are:

- Try disabling/removing the device with the IP address `192.168.0.181`.
- Once your PC is connected to FinalStage, try connecting another device (eg: a phone), and browsing to http://192.168.4.2:8080/ - It should give a 404 error (not a connection error or similar), there should also be the attempt in the logs on screen. If there is a connection error, I would try again to turn off the firewall completely in the control panel.

JC38, 2017-12-02:
I've tried disabling all the lan connections except the Wireless, the problem is the same.
When my PC was connected as 192.168.4.2 (and pinging 192.168.4.1 to be sure), I was not able to connect to the FinalStage Wlan from my Android phone. The phone said 'unable to get an ip address' for this Wlan.

sillyfrog, 2017-12-02:
It will reboot every minute or so, I would suggest trying again joining with your phone, and see if it gets an IP address (I have tested this just now, and it does work with a couple of devices connected to the FinalStage SSID).
Good luck!

jorisc90, 2017-12-02:
Running into the same thing, also with an S20 (EU version). All went well until in the final stage. I quit the script after ~5mins of trying. The output was the same as the replicated one underneath.

Debug after re-plugging the device; we see it tries to download the correct bin. Afterwards we see me trying the HTTP server using a browser (favicon, then downloading the actual image file). 

Device button remains lit green steadily after trying to load the image. 

`> sonota.py --no-prov                                                                                                                                                            
Platform: win32                                                                                                                                                                  
Starting stage2...                                                                                                                                                               
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)                                                                                                                        
~~ Waiting for device to connect                                                                                                                                                 
Current IPs: ['10.195.1.93', '192.168.56.1']                                                                                                                                     
                                                                                                                                                                                 
*** IMPORTANT! ***                                                                                                                                                               
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.                                           
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.                                                                                       
This server should automatically be allocated the IP address: 192.168.4.2.                                                                                                       
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.       
...Current IPs: ['192.168.4.2', '192.168.56.1']                                                                                                                                  
                                                                                                                                                                                 
                                                                                                                                                                                 
The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed.                                                            
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all firewalls have been COMPLETELY disabled on your system.                                              
Sending file: /ota/image_arduino.bin                                                                                                                                             
.........200 GET /ota/image_arduino.bin (192.168.4.1) 19463.82ms                                                                                                                 
..404 GET /ota (192.168.4.2) 1.00ms                                                                                                                                              
404 GET /favicon.ico (192.168.4.2) 1.00ms                                                                                                                                        
..........Sending file: /ota/image_arduino.bin                                                                                                                                   
200 GET /ota/image_arduino.bin (192.168.4.2) 30.02ms                                                                                                                             
.........                                                                                                                                                                        
                                                                                                                                                                                 
The "FinalStage" SSID will disappear when the device has been fully flashed and image_arduino.bin has been installed.                                                            
If there is no "Sending file: /ota/image_arduino.bin" log entry, ensure all firewalls have been COMPLETELY disabled on your system.                                              
..                                                                                                                                                                               
                                                                                                                                                                                 `

sillyfrog, 2017-12-02:
@jorisc90 To confirm, each time you try you have something like this in the logs?
`200 GET /ota/image_arduino.bin (192.168.4.1) 19463.82ms`
If so, this looks like a different issue to this (it's trying to get the image, but not actually installing it). If you do a full reset of the device (ie: disconnect from power for a few minutes), does it still do the same? (If so, can you please create a new issue, as the problem appears to be something different).

jorisc90, 2017-12-02:
@sillyfrog that is indeed what happens, after each reset I see a succesful GET complete in the logs. For now, I'm trying to flash this particular device using the UART to see if all is well. I've got a newly boxed one I will try tomorrow, and if needed will create a new issue for it.

sillyfrog, 2017-12-03:
Hmmm... bummer... If you have a chance to get the serial output before you flash that would be great. I think someone else had a similar issue, but I have never been able to see the serial output to try and do something about it. The FinalStage will print a bunch of debug info via serial as it's running - so that _may_ shed some light as to what is happening.

Good luck!

jorisc90, 2017-12-03:
Ah. Didn't get to see your post before I flashed. Device works normally after the reflash. Created the new issue with according logs.

JC38, 2017-12-04:
Hello,

My problem is solved but not with OTA as expected.
Using a wired connexion, I was able to upload a custom firmware.
Thanks for your job and for your help.

closed 2017-12-04 by JC38