mirko/SonOTA #66: S20 EU does not move from "FinalStage" firmware to final firmware

jorisc90, 2017-12-03:
- Operating System: Windows 10 1709
- Python Version: 3.6 x64
- Sonoff model: S20 EU

Please also drag and drop the generated `debug_########.log` file onto the issue so it is attached.
[debug_1512295526.log](https://github.com/mirko/SonOTA/files/1524662/debug_1512295526.log)

When trying to use SonOTA to flash a "freshly" unboxed S20, it seems that the unit does not want to boot from FinalStage (minimal) firmware image to the regular firmware. 

After replugging the device, the FinalStage SSID appears again, thus indicating that the regular firmware was not flashed after it was downloaded from the server (see log). 

I had the same issue with another unit yesterday, this one I was able to flash with the UART like normal afterwards (even though I half-destroyed it with crappy screwdrivers in the progress 👎 ; note to self: invest in quality equipment).

Hoping that we can fix this one with debug collected using the script, or collected over IP. I can ping the 192.168.4.1 IP when connected to the FinalStage SSID (although I'm not sure which services are exposed over IP). Otherwise it might take me some time to follow up, as I want to keep this one physically more intact.

sillyfrog, 2017-12-05:
I'm unsure on this one. If you do get stuck in this state again, it would be great to get the output from the UART (as that is the only place FinalStage logs anything). FinalStage is supposed to do the bare minimum (ie: be pretty dumb), so I'm not sure what else it could be.

Thanks for the debug log, it looks like as you have suggested - unfortunately I'm not sure what else it could be at this stage, sort of some different type of Flash requirement or something...

klein0r, 2017-12-06:
Same issue with an Sonoff DUAL. Everything was working fine. I've received this message before sonota.py quitted.

```
2017-12-06 00:16:51,529: INFO: No longer on "FinalStage" SSID, all done! Now connect to the sonoff-#### SSID and configure for your WiFi (it will not be configured).
2017-12-06 00:16:51,531: INFO: Quitting.
```

But: FinalStage is still present. How can I fix this, without flashing the sonoff by serial? Is it possible? How can I provide more information to help you?

jorisc90, 2017-12-06:
I'll try to get a good screwdriver set and read out what is happening. Will
update the thread as soon as I get some output.

jorisc90, 2017-12-06:
Hi all. Just opened up my second unit (this time without breaking the case :-) ) and collected the logs. They are attached. Seems the unit hangs in the bootloader section.

[putty.log](https://github.com/mirko/SonOTA/files/1536510/putty.log)

klein0r, 2017-12-06:
How do you got that log? Have you attached a serial connection to the sonoff?

jorisc90, 2017-12-06:
By attaching the USB --> serial (UART) converter to the S20 (soldered on the pins), then opening a Putty session with logging on the USB COM port; baud rate set to 115200.

Connect the GND, TX and RX pins first, then power on with the 3.3V pin.

klein0r, 2017-12-06:
Mh thats really strange. I am getting the exactly same output as yours. But: The script says it has finished with the "no longer connected to FinalStage Wifi", but the serial connection is still printing hundreds and hundreds of dots.

It looks like the Wifi Connection is interrupted and the script is thinking everything is finished, but the flash isn't completed at that time completely.

jorisc90, 2017-12-06:
Same behavior as here. Dots just keep going.

klein0r, 2017-12-06:
The "FinalStage" WiFi network is disappearing for a second during flash after exactly 5 minutes. Looks like this interrupts the flash script.

klein0r, 2017-12-06:
Is it normal that 2. Stage requires 5+ minutes?!

sillyfrog, 2017-12-06:
It should be a lot faster. That dump is helpful in that it'll at least help narrow down where to look in the code, specifically:
https://github.com/sillyfrog/Espressif2Arduino/blob/f6f9a8dc095db11298bd7a2ac253cc8cc63fea97/Espressif2Arduino/Espressif2Arduino.ino#L256

I'm away at the moment, but will try and see if I can get another FinalStage built so we can test next week if someone is up for it? It will require the FinalStage to be reflashed via serial, but it should be a good test if this will fix it for others.

sillyfrog, 2017-12-06:
Given the number of dot's - I'm assuming this has gone bad, and the 5 minutes is just it rebooting and trying again.

jorisc90, 2017-12-06:
I'll keep mine delidded for now. No final use case for it anyways :-) Will flash the new FinalStage when ready.

klein0r, 2017-12-08:
Same here, just waiting for another image to flash via serial ;)

sillyfrog, 2017-12-10:
Attached is an updated bin file (I needed to zip it up so I could upload, so please unzip first). 

To flash, running something like: `esptool.py -p /dev/cu.usbserial-A50285BI -b 921600 write_flash  0x81000 Espressif2Arduino.ino-0x81000.bin` should do it. Obviously replace the serial port with your serial port, and if you have any issues, adjust the baud rate to something that works (probably 115k). This will only flash the ROM 2 that was having issues as per the serial dump. This has a single change that just provides more debugging - the more I look the more confusing it is, but this may give  a pointer.

To get valuable debugging, run `sonota.py --slowstream --no-prov` once it's flashed. If you can then post the serial output, I may give a further pointer.

[Espressif2Arduino.ino-0x81000.bin.zip](https://github.com/mirko/SonOTA/files/1545355/Espressif2Arduino.ino-0x81000.bin.zip)

jorisc90, 2017-12-10:
Just uploaded the new firmware and tried again, to no avail. The output from the serial changed a bit, but the loop encountered remains for now. Attached the logs from esptool, putty and SonOTA while perfoming the tests:

[esptool.log](https://github.com/mirko/SonOTA/files/1545490/esptool.log)
[putty.log](https://github.com/mirko/SonOTA/files/1545492/putty.log)
[debug_1512900347.log](https://github.com/mirko/SonOTA/files/1545493/debug_1512900347.log)

sillyfrog, 2017-12-12:
It matches what I would have guessed, for some reason it's not reading the entire 4096 bytes of data. I'm going to guess that for some reason the S20 has less RAM, or maybe something else (not sure). It'll be a little while before I can spend more time on this, but my next test will be enabling debugging in the HTTP client to see why the buffer is so small (this is the source for it: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h ).

The source of what I uploaded is here if you wanted to have a poke around yourself: https://github.com/sillyfrog/Espressif2Arduino

I'm keen to try and get this resolved, but unfortunately it'll be longer than I would like (and I don't have an S20 for testing).

If you do give it a go, let me know what you find, and I'll also update when I get a chance to have a go again.

jorisc90, 2017-12-12:
Thanks for your update and the time spent :-)

I'm on training this week so don't have physical access to the S20 either. I don't have the time resources to find out how to poke around (debugging and finding out how to compile) this week.

If you would be able to get a debug firmware compiled (or a quick step-by-step of how to do this), I could do it this weekend.

maumi, 2017-12-20:
Hey guys, any progress on this?
I could test OTA on the S20 and if it helps I also can flash classic per cable.
But I would need some guidance with classic flashing :)

sillyfrog, 2017-12-21:
Practically I won't be able to look at this for about a month (time off). If you were were looking to flash it manually, I would be suggesting you look here https://github.com/arendst/Sonoff-Tasmota/wiki as this is how it's installed typically.

From the debugging @jorisc90 helped do I can see what they problem is - just not sure why... :-/ 

I don't have an S20 of my own to debug further (but we'll accept donations :) - something mirko have started discussing) - but the next steps (as per above) involve looking at the HTTP library and trying to figure out why it's not buffering more data.

Good luck!

maumi, 2017-12-21:
Thanks. good to know. Yesterday I flashed it wired. Now that I know I need to wait I will flash my second also classic. 
Nice work anyway, works flawless for POW and RF.

jorisc90, 2017-12-21:
Going to flash wired as well. Thanks for updating.

@sillyfrog please send me your address details for a belated christmas gift :-)

sillyfrog, 2017-12-21:
@jorisc90 Thanks! Drop me an email and I'll follow up directly. Cheers.

NameTakenAgain, 2018-01-20:
I wish I had read this thread before I attempted an OTA update of one of my S20s. I saw fw 1.5.5 was listed as a Known Working Configuration. Updated the wiki entry for S20 with a pointer to this thread.

As the others in the thread I got through the first phase correctly and I see the FinalStage SSID. I also so a "Sending file: /ota/image_arduino.bin" log entry, but gave up after 30 minutes. After rebooting the S20 it tries to download the file again.

alanvanwyk, 2018-01-20:
Got an S20 coming from Amazon on Monday.
Firmware update fail will mean an immediate return.

Also had to remove a basic I had cables for the same reason.

They’re great devices, but they either need to allow personalised MQTT in their default firmware or allow OTA updates of 3rd party firmware. If not - they’ve lost my business - I have Lightwaverf and Zigbee around the house anyway, so will be the nominal premium and go that way instead.

Does anyone know of an alternatively branded unit that still allows OTA and supports Sonata? (And is CE approved)

klein0r, 2018-01-21:
Since firmware 1.6 is getting popular, i'll keep flashing my devices wired. Would be great if OTA would work for every firmware version. You don't need to solder the header pins - just hold them tight to the connectors while flashing.

It's a pity that this awesome project will come to an end 😢 

If I was ITEAD, I would not prevent my customers from exchanging the firmware. Instead, I would do anything to make that progress as easy as possible. It's a huge market for those components...

ElecTriX666, 2018-01-22:
No update on this? I have a S20 with 1.5.5 and for some reason flashing with wires don't work on mine, it doesnt enter flash mode..

ElecTriX666, 2018-01-22:
If it is a factor I have the v1.3 EU board!

alanvanwyk, 2018-01-22:
Amazon S20 arrived.
It’s running 1.55 - will feedback on OTA flash.

Intention was 6 more, but going to send this back in revolt, until Sonoff sort this rubbish out.

I have an extensive home automation setup - Sonoff was filling a gap. I spend enough time writing code against openhab and various APIs . I don’t have time for soldering.

Mqtt was a godsend - sonata made it happen,

Sonoff, for now you’ve lost my business.

@mirko etc - thanks for your hard work

maumi, 2018-01-22:
I also would enjoy it if I could flash the S20 OTA, but it isn‘t such a drama to flash it wired. You Even don‘t need to solder. You just Need to hold the Pins for some seconds while flashing.

alanvanwyk, 2018-01-22:
Managed my 1.55 firware S20 OTA.

@maumi - a photo showing the "hold wires" technique would be handy.

Does the same apply for the Sonoff basic etc?
sadly, I'll need to buy one of the USB adapters too.
I am sure I used to have a couple from my arduino / netduino days, but I
thought those days were past :-(

Still not convinced I want to continue supporting a company who made their
start of "gadgeteers" like me that like to push the boundries a little, but
now want to make things more difficult without just cause.

It's not as if they have even bothered responding on their forums!

ElecTriX666, 2018-01-23:
@alanvanwyk 

Wire holding works the same for the Basic, I did it. 5 holes but only 4 are used.

etpedro, 2018-01-27:
Hi Everyone,

I bought an S20 that comes with fw 1.6.0 and OTA doesn't work also. debug attached.
[debug_1517062176.log](https://github.com/mirko/SonOTA/files/1670135/debug_1517062176.log)

sillyfrog, 2018-01-28:
@etpedro Please see #58 (and now home page) re v1.6 - unfortunately there is no way around this. Please post feedback to ITEAD, I have started a thread here: http://disq.us/p/1oqbm8m - the more comments the better I figure.

@jorisc90 I'm finally back and have been testing the S20 I have now received! Thanks again!

It came with v1.5.2, and worked for me first go - so not really helping to debug the issue unfortunately. Of course the app only gives me the option to upgrade to v1.6 - which did as expected (blocked while accessing the SSL port). I'll post here if I find anything else.

Cheers.

Professor-Mayhem, 2018-02-11:
I just received a batch this week. This is what the serial console reports during "FinalStage"
.Waiting for connection...
.Connected
Web server accessible...
Flash Mode: 0
ROM or Flash Mode not recognized

pplucky, 2018-03-08:
Hello everyone. 

Does anyone know of a way (without going through the whole SonOTA process), of knowing the current original firmware installed on an S20. Going throught Ewelink app device settings doesn't work, as device appears offline all the time.

Thanks in advance.

sillyfrog, 2018-03-09:
@pplucky In the EWeLink app, under the device, click on the ellipsis (`...`) in the top right and click "Setting".

pplucky, 2018-03-09:
@sillyfrog That didn't work because device always appeared offline in the app and nothing appeared in Settings/Firmware.

I ended up trying SonOTA anyway but as it didn't work, I just opened the plug, plugged FTDI into it and flashed Tasmota and now it works properly.

Thanks anyway.

Trippersgarage, 2018-08-11:
Any updates? The world waits with bated breath. :)

rradar, 2018-08-30:
How can I know the software version running on my s20-clone? There is a hardware revision written on the board... but software wise?

rradar, 2018-09-07:
I have this one here:

[image]

No idea about the software version or how to test it

And here the accident

[image]

mahockey, 2018-10-03:
Sonoff have released the S26 which seems to be a miniature version of the S20 and these things are selling like hotcakes, so would love to have a way to flash OTA.
Any update to the issues above with the S20? so we could proceed with the S26?

The S26 board is so much smaller than the S20 so lot harder to solder onto the pin outs, so for the novice its just got a hell of a lot harder - this is where an OTA solution would be great :)

here is an image of the new smaller S26 board and pin outs... https://user-images.githubusercontent.com/288709/43617653-96d3c05c-96bb-11e8-9c1b-2c4442eeeda1.jpeg

Klimbim1, 2018-10-17:
Aloah, any news concerning this topic? 

I tried to flash my S20 OTA, but the final stage WiFi did not show up... very sad, the soldering is just wasting time...

peterhoeg, 2018-11-09:
I had the same issue with an S20 being stuck on FinalStage - unfortunately I do not know what version of the original firmware was running at the time but the device was bought about 5 months ago.

I then tried running this with 192.168.4.1 being the address used during FinalStage:

``` sh
pio run -e itead-s20-ota -t upload --upload-port=192.168.4.1
```

And I was then greeted with 

``` sh
[INFO]: Sending invitation to: 192.168.4.1
Authentication...FAIL
[ERROR**: Authentication Failed```.
*** [upload] Error 1
```

So let's try *with* a password:
``` sh
export ESPURNA_AUTH=fibonacci
pio run -e itead-s20-ota -t upload --upload-port=192.168.4.1
```

Lo and behold, that worked and my S20 is now running espurna 1.13.3 without having opened it.

Maybe somebody else can try this as well?

Klimbim1, 2018-11-10:
> I then tried running this with 192.168.4.1 being the address used during FinalStage:
> 
> ```shell
> pio run -e itead-s20-ota -t upload --upload-port=192.168.4.1
> ```
> 
> And I was then greeted with
> 
> ```shell
> [INFO]: Sending invitation to: 192.168.4.1
> Authentication...FAIL
> [ERROR**: Authentication Failed```.
> *** [upload] Error 1
> ```
> 
> So let's try _with_ a password:
> 
> ```shell
> export ESPURNA_AUTH=fibonacci
> pio run -e itead-s20-ota -t upload --upload-port=192.168.4.1
> ```
> 
> Lo and behold, that worked and my S20 is now running espurna 1.13.3 without having opened it.
> 
> Maybe somebody else can try this as well?

I am stuck as well and would like to try something without soldering if possible. Did you try that in visual studio code or in which program? What would be the command to upload the sonOTA fw instead of espurna?

Regards!

peterhoeg, 2018-11-12:
> Did you try that in visual studio code or in which program? 

platformio from the command line.

> What would be the command to upload the sonOTA fw instead of espurna?

I don't know. Espurna supports a two-step update process where it pushes a small "core" version first which is then updated to the proper version using its OTA update mechanism. I think you should be able to upload another firmware later.

Ganfoud, 2020-06-25:
hi all
has there been any news on this since 2018? I just bought some S20 EU plugs...
Is this an issue with Tasmota only and could it work properly with ESPurna for example?
what is the situation after the failure to upload the firmware, is it still usable with the original firmware?
thanks

sillyfrog, 2020-06-27:
@Ganfoud I'm guessing at this stage it's more likely the issue as mentioned on the home page. Sonoff have "fixed" the loophole we were using to put on custom firmware in the current releases :(
So this project is unlikely to be of any value. Personally I have moved to using Tuya based devices, and using Tuya-Convert to update them.