mirko/SonOTA #67: Compatible with Sonoff RF Bridge

pplucky, 2017-12-03:
- Sonoff model: Sonoff RF Bridge

Although I didn’t find it on the Wiki, I was wondering if this update method is compatible with the Sonoff RF bridge (https://www.itead.cc/wiki/Sonoff_RF_Bridge_433).

Thanks in advance for any insights.

sillyfrog, 2017-12-03:
It should, but I have not tested personally. 

Worst case it will need to be flashed by serial anyway. 

Please let me know/update the Wiki if it does work. 

Cheers.

projekti, 2017-12-13:
I'm trying to do this right now.
I'm using ArchLinux and Python 3.6.3.
I get the following Error.

`Exception in callback None   
Traceback (most recent call last):    
  File "/usr/lib/python3.6/site-packages/tornado/ioloop.py", line 888, in start
    handler_func(fd_obj, events)
  File "/usr/lib/python3.6/site-packages/tornado/stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/tornado/iostream.py", line 519, in _handle_events
    self._handle_read()
  File "/usr/lib/python3.6/site-packages/tornado/iostream.py", line 1390, in _handle_read
    self._do_ssl_handshake()
  File "/usr/lib/python3.6/site-packages/tornado/iostream.py", line 1310, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1068, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
`
Issue #57 looks similar.
I also tried '--slowstream' and '--legacy'

[debug_1513202640.log](https://github.com/mirko/SonOTA/files/1557286/debug_1513202640.log)

sillyfrog, 2017-12-14:
@projekti the `OSError: [Errno 0]` is a bad thing... I never really knew what caused it previously. I think a mismatch in libraries, or if the Sonoff is on v 1.6.x, then it is likely #58

MACscr, 2017-12-29:
Did you get this working? Im wanting to do the same.

pelson, 2018-01-13:
Just to say, I'm just doing a Sonoff RF Bridge OTA installation now. In order to get the ITEAD-XXXXXX wifi network to appear, I needed to follow the advice in the wiki. Specifically:

> Getting the ITEAD-###### SSID
> On various devices it appears that a 2nd 7 second button press is required sometimes to put the Sonoff into Configuration mode so it will broadcast the ITEAD-###### SSID. If after pressing the button for 7 seconds you have a pattern such as blink-blink-blink-pause-blink-blink-blink-pause, press the button again for another 7 seconds until you have a steady blink-blink-blink...

With my python 3 environment (provisioned from conda-forge) and pip install of requirements file, I was able to get to the FinalStage:

```
python sonota.py 
Current IPs: ['192.168.1.5']
Select IP address of the WiFi interface:
    0: 192.168.1.5
Select IP address [0]: 0
WiFi SSID: PLUSNET-XXXX
WiFi Password: XXXX

Using the following configuration:
	Server IP Address: 192.168.1.5
	WiFi SSID: PLUSNET-XXXX
	WiFi Password: ****
Platform: darwin
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
............................................................................................................................................................................................................................................................................Current IPs: []
...Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100026XXX",
    "apikey": "bf78ec7f-XXXX-4509-XXXX-69e87XXXX11e",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "PLUSNET-XXXX",
    "password": "****",
    "serverName": "192.168.1.5",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.1.5) is not assigned to any interface on this machine.
** Please change WiFi network to PLUSNET-XXXX and make sure 192.168.1.5 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
......Current IPs: []
.............................Current IPs: ['192.168.1.5']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
```

Sadly, I didn't ever get any download logs, nor did the FinalStage SSID appear. I'm going to move on to programming serially.

tyjtyj, 2018-01-14:
I m stuck here. No final stage ssid.. Please remove compatible from the readme page

oglodyte, 2018-01-15:
@tyjtyj  as you can see from [this thread](https://github.com/mirko/SonOTA/issues/20), looks like few of us having problem flashing RF Bridge or B1 with SonOTA. Might be that latest Sonoff firmware is checking for valid certificate.  I used to flash B1 earlier successfully, but no success with recent RF Bridge I got.

EliasKotlyar, 2018-05-07:
Can confirm that i am also having the following error on Sonoff RF with Firmware 1.60:
```
OSError: [Errno 0] Error
Exception in callback None()
```

sillyfrog, 2018-05-07:
@EliasKotlyar Can you upload the full logs and show the full error, that looks more like a OS type issue than a Sonoff side issuea.

afaucogney, 2018-07-06:
Here is my log

..............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................Current IPs: []
.....Current IPs: ['10.10.7.2']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100018XXXX",
    "apikey": "2e8a8038-XXXX-XXXX-XXXX-1eaa616e39bc",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "Freebox-Syrah",
    "password": "************************",
    "serverName": "192.168.1.41",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.1.41) is not assigned to any interface on this machine.
** Please change WiFi network to Freebox-Syrah and make sure 192.168.1.41 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
......Current IPs: []
....Current IPs: ['192.168.1.41']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
Using selector: KqueueSelector
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
.Uncaught exception, closing connection.
Traceback (most recent call last):
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()
handle: <Handle cancelled>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/platform/asyncio.py", line 117, in _handle_events
    handler_func(fileobj, events)
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 657, in _handle_events
    self._handle_read()
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
......................Uncaught exception, closing connection.
Traceback (most recent call last):
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 657, in _handle_events
[... 48 lines cut ...]
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
.....
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
......Uncaught exception, closing connection.
Traceback (most recent call last):
[... cut ...]
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1514, in _handle_read
    self._do_ssl_handshake()
  File "/Users/anthonyfaucogney/Library/Python/3.7/lib/python/site-packages/tornado/iostream.py", line 1434, in _do_ssl_handshake
    self.socket.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
Exception in callback None()

LizMyers, 2018-08-19:
Has this been resolved? I'm having same issue - pass Stage 2, then never get to final stage.

heiderich, 2019-03-21:
Looks like this might be due to the following issue in tornado: https://github.com/tornadoweb/tornado/issues/2504

Stoatwblr, 2020-04-26:
Some testing using other hosts on the same network seem to indicate that the RF bridge never attempts to make a connection to the webserver on the SONOTA system

Testing meaning that I can connect from other hosts and it's logged - it really does look like SONOTA is simply waiting forever for a connection that never comes.

closed 2022-05-07 by pplucky