mirko/SonOTA #68: Using with ESP Easy?

glsf91, 2017-12-03:
Can I also use SonOTA to flash ESP Easy on the Sonoff?

If yes, is it just replacing the image_arduino.bin with the bin (546KB) of ESP Easy?

sillyfrog, 2017-12-05:
Yes, that's correct, it's designed so you can drop in any (Arduino) based imaged and it will get flashed on. Be sure it has been built with the correct flags for the flash type.

That said, it needs to be firmware that can run with OTA, and given the size (>512KB), I'm concerned you may have issues with it. So I would suggest confirming that ESP Easy is designed to work with OTA updates on a 1MB flash.

Sonoff-Tasmota also supports uploading via it's web UI, so if you were to update to Sonoff-Tasmota, you can then replace it using the web interface if you wanted (again, subject to that size limit).

closed 2017-12-05 by sillyfrog

glsf91, 2017-12-06:
ESP Easy can also use OTA on 1M. There is also a minimal version, which you need to flash before flashing the "bigger" one. Tasmota is using the same approach.
But the ESP easy minimal version needs SPIFFS.
I do not know when SPIFFS is "placed" on the device. Is it included in the firmware "image_arduino.bin" or is this done separately?
Normally SPIFFS survives flashes (as far as I know), so I expect it will be outside the firmware file.

So can a firmware file which needs SPIFFS be flashed with SonOTA if the SPIFSS is not there already (which I think is the case)?

glsf91, 2017-12-09:
To answer my own question:
It is indeed possible to flash ESP Easy through Tasmota firmware upgrade.
First install Tasmota minimal firmware. Then you flash the ESP Easy firmware.
So now I have Sonoff S20 OTA flashed with Tasmota en then ESP Easy without soldering. Thanks to SONOTA