mirko/SonOTA #69: Failure on Windows 10

ConfusedTA, 2017-12-05:
- Operating System: Windows 10 Enterprise Build 1709
- Python Version: Python 3.6.3 (v3.6.3:2c5fed8, Oct  3 2017, 18:11:49) [MSC v.1900 64 bit (AMD64)] on win32
- Sonoff model: Sonoff Basic (firmware 1.6.0)

When attempting to perform Stage1, immediately after connecting to the device, the following errors occur.

```
C:\Users\Garry\Desktop\SonOTA-master>sonota.py` -s
Current IPs: ['192.168.10.46', '192.168.204.1', '192.168.193.1']
Select IP address of the WiFi interface:
    0: 192.168.10.46
    1: 192.168.204.1
    2: 192.168.193.1
Select IP address [0]: 0
WiFi SSID: Coggles End
WiFi Password: *********

Using the following configuration:
        Server IP Address: 192.168.10.46
        WiFi SSID: Coggles End
        WiFi Password: *********
Platform: win32
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
...Current IPs: ['10.10.7.2', '192.168.204.1', '192.168.193.1']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "10001f400e",
    "apikey": "88e252e0-5ca2-46d3-92eb-7bb324b5c17c",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "Coggles End",
    "password": "*********",
    "serverName": "192.168.10.46",
    "port": 8443
}
Traceback (most recent call last):
  File "C:\Users\Garry\Desktop\SonOTA-master\sonota.py", line 723, in <module>
    main()
  File "C:\Users\Garry\Desktop\SonOTA-master\sonota.py", line 718, in main
    stage1()
  File "C:\Users\Garry\Desktop\SonOTA-master\sonota.py", line 601, in stage1
    "http://10.10.7.1/ap", "POST", json.dumps(data))
  File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\site-packages\httplib2\__init__.py", line 1322, in request
    (response, content) = self._request(conn, authority, uri, request_uri, method, body, headers, redirections, cachekey)
  File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\site-packages\httplib2\__init__.py", line 1072, in _request
    (response, content) = self._conn_request(conn, request_uri, method, body, headers)
  File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\site-packages\httplib2\__init__.py", line 1025, in _conn_request
    response = conn.getresponse()
  File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 1331, in getresponse
    response.begin()
  File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\http\client.py", line 258, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out
```

[debug_1512481278.log](https://github.com/mirko/SonOTA/files/1531291/debug_1512481278.log)

ConfusedTA, 2017-12-05:
To add, I have also tried with --legacy and -s and a combination of both.

ConfusedTA, 2017-12-05:
Just took my 2nd newly delivered Sonoff Basic out of the box, and didn't do anything else with it - didn't hook it up with eWeLink, didn't update the firmware (so sorry I don't know what version of the firmware it was running on) - it's now up and running perfectly with Tasmota however.

sillyfrog, 2017-12-05:
That's... a really weird error for that stage of things - the only thing I can these is the Sonoff went out of AP mode the instant it tried to configure it - I'm assuming you got the same error more than once?

ConfusedTA, 2017-12-05:
Yes I've had that error multiple times on Windows, and also just had it looping over and over, in both Windows and Linux.

reloxx13, 2017-12-06:
did u configure the failed device with the eWeLink app before?

did u try to reset it to factory before running sunota?

> Factory Reset
> 
> You can reset the devices wifi & network configuration settings by holding the switch down on the device for 10 – 20 seconds. You can reset all of the settings on the device (useful if you forget the configuration password) by holding down the switch for 20 – 60 seconds.

ConfusedTA, 2017-12-06:
Yes, I did set it up via eWeLink, and updated the firmware to 1.6.0

Pressing and holding the button did not appear to do anything, I'm sure I held it down for well over a minute - I will give it a try again tomorrow though.

sillyfrog, 2017-12-06:
@ConfusedTA I have found the reset does not always work, sometimes you need to do it a second time, sometimes a power cycle is required.

binford6000, 2018-01-26:
Hi everybody,
i got exactly the same issue, two sonoff basic and windows 10 and 
`...File "C:\Users\Garry\AppData\Local\Programs\Python\Python36\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out`
One is configured in eWeLink, the other is not. Both having firmware 1.6.0. Both have been resetted several times. Nothing changes. Kind of frustating...
[debug_1516988385.log](https://github.com/mirko/SonOTA/files/1668630/debug_1516988385.log)

closed 2018-01-28 by sillyfrog