mirko/SonOTA #70: Sonoff basic 1.6.0 firmware OTA update not working

xyleth, 2017-12-05:
- Operating System: MacOSX
- Python Version: 3.6.3
- Sonoff model: Basic
[debug_1510154892.log](https://github.com/mirko/SonOTA/files/1532299/debug_1510154892.log)

Hello,

I recently got a pair of Sonoff Basic devices (having previously used this to flash fw to TH16 and 4ch devices).  With the first unit I mistakenly upgraded it to the latest 1.6.0 firmware via the Sonoff app and now it won't complete the provisioning.  It just skips from the ITEAD-1000001xxxxx SSID provisioning step straight to the final download step without performing the main download in the middle sequence.

The other device, which I left on 1.5.5, flashed fine.

I've attached the debug log.  Is there anything else I can do to help troubleshoot this with you?

ConfusedTA, 2017-12-05:
This must be a firmware related issue - as I have just made the same mistake as you in updating one to 1.6.0 which I have incorrectly attributed to being a Windows issue - I've just fired up a Linux box and am experiencing similar issues which I recorded in #69 - as these are seemingly related to the same issue, hopefully someone can combine the two.

sillyfrog, 2017-12-05:
@xyleth That's... weird... To clarify, are you seeing the "FinalStage" SSID? (I don't see that happening in the logs).

It can take a minute or two for the Sonoff to retry after your PC connects back to the normal WiFi, so maybe worth leaving it there a bit longer.

xyleth, 2017-12-05:
Hi - thanks for the quick response.

I never see the "FinalStage" SSID with this unit.  I tried again and left it for 10 minutes to no avail - the script just cycles through its final stage warning and the device never seems to connect for firmware upgrade.

If there's anything else you'd like me to do  / check let me know.  I don't have an immediate need for this second Basic so I'm happy to use it for testing.

ConfusedTA, 2017-12-05:
Same result here - and likewise am in no immediate need for it, so can do some testing on Windows too.

sillyfrog, 2017-12-05:
I think the first thing to check is that it's getting a DHCP lease.

But (more likely), it could be #58  - I have a Basic that is not updating via the app for me (and I'm away this week). But I'll try again next week and see if I can get it to update, and replicate this issue.

ConfusedTA, 2017-12-06:
Yes mine was connecting to the internal network and was pingable.

It would suck if they're clamping down on the SSL certificate - they make some reasonable hardware, and the fact we could do this to it probably did boost their sales quite a bit!!

agidlund, 2017-12-07:
- Operating System: Windows 10
- Python Version: 3.6.3
- Sonoff model: S20
- Sonoff factory firmware version: 1.5.5

I don't want to open a new issue, as I have the exact same behavior as described here. I upgraded two S20 to firmware version 1.6.0 and can't flash any of them. The process stops when the device is to download the new firmware in step 2. The other S20 I didn't upgrade are all able to flash with sonota.

If it is the case that they are now verifying the ssl certificate, isn't it possible to use something like http://docs.mitmproxy.org/en/latest/mitmproxy.html and tamper the response?

andyjenkinson, 2017-12-07:
Another report here, stupidly allowed it to update to 1.6.  tcp dump attached.

[dump.pacp.zip](https://github.com/mirko/SonOTA/files/1540832/dump.pacp.zip)

andyjenkinson, 2017-12-07:
@agidlund I don't see how one would get the sonoff to trust the CA certificate to do a man in the middle attack, if that is indeed the problem.

agidlund, 2017-12-08:
@andyjenkinson True, I didn't realize that aspect. That would require a change of its firmware and that's what we are trying to do in the first place...

jebo01, 2017-12-09:
I did the same mistake to upgrade to 1.6 version and getting the same problem. Would be great if anyone could come up with a solution. I take it that its not possible to restore the device to original FW version?

krasatos, 2017-12-09:
So, is this confirmed that there is no option to flash v1.6.0 yet?

sillyfrog, 2017-12-10:
@krasatos If it's the same as #58, then it won't brick it, just reset the configuration worst case, and you'll need to reconfigure for your LAN.

I'm closing this issue, and going to put all related comments on #58.

closed 2017-12-10 by sillyfrog

mirko, 2017-12-11:
@andyjenkinson @agidlund just for the record: neither the Arduino TLS nor the Espressif Update stack implement functionality for either certificate pinning nor CA cert checking. Happy MITM'ing!

NiekN, 2018-02-26:
Are there any updates regarding to firmware 1.6

Gramps, 2018-06-16:
Anyone know if the 1.8.1 update makes this functional again?

jgyoldi, 2018-06-18:
I have the same question, I have two new modules that came wit 1.6 version and I don´t know if it is a good idea to upgrade them to 1.8.1 version. 
Another question, is there any possibility to downgrade this sonoff modules to 1.55 firmware?

Gramps, 2018-06-18:
I'm going to upgrade one to 1.8.1 and try it out.  I also have a 1.5.5 but can't get that to work.  Nor can I find patch notes for 1.8.1.

I believe to downgrade you have to use the USB method.

jgyoldi, 2018-06-18:
Ok. Let us now what you get.

About downgrading: to use USB  I have to solder pines and if I solder them I can flash it no matter what FW I have...