mirko/SonOTA #73: Never reaches "FinalStage"

MTrab, 2017-12-09:
- Operating System: Windows 10
- Python Version: 3.5.4
- Sonoff model: Sonoff Touch, EU model, 1 gang
[debug_1512838320.log](https://github.com/mirko/SonOTA/files/1544954/debug_1512838320.log)

When trying to flash a Sonoff Touch it never show up on FinalStage (don't think it should take 20 minutes to flash), but rather seems to reboot during the flash, and then start over.

What could be the issue for this?

neuman1812, 2017-12-10:
I would also like to add to this.  I was running on an Ubuntu 17.10 System.     The process seems to stop at step 4.

` Step 4: Once the firmware has been downloaded, there will be a new FinalStage SSID that you can connect to. Do this, and the device will then download the final stages of the firmware, including the Arduino image, replacing the default boot loader.`

My terminal brings up the message about waiting for the download to finish and look for the SSID to connect to,  but it just keeps posting that message.  Approx every 20 seconds.  With little dots counting down?   I let it run for 30 min with no sign of that FINALSTAGE ssid.

sillyfrog, 2017-12-10:
Nothing jumps out as it appears to be connecting. My suggestion would be to try running without `--slowstream` - and if you are still having issues, if you can take a traffic dump and post it, that may also help. I would also suggest double checking there is nothing blocking incoming port 8080.

MTrab, 2017-12-10:
I have tried with/without both slowstream and legacy - same outcome.
All firewalls are disabled on the machine during this.

But one thing I noticed late last night was that it appears to stop transferring the bin files in the mid 20%

petjek, 2017-12-10:
Same problem here. Everything runs smoothly until stage 4. I used a basic that was connented using the app before. So maybe it installed a newer firmware during that time? It took quite a while until it finally showed up online within the app.

adam-goddard, 2017-12-10:
Operating System: MacOS 10.12.6
Python Version: 3.6.3
Sonoff model: Sonoff Basic firmware 1.6.0

Same issue. Just keeps downloading, no "FinalStage".

neuman1812, 2017-12-11:
I just checked one of th Sonoff devices I had.  It was running firmware version 1.5.5.  I ran the script on that and it worked.  Connected and updated.   The one for 1.6.0  Failed.   Looks like its a Cert Issue based on this https://github.com/mirko/SonOTA/issues/58#issue

petjek, 2017-12-11:
@neuman1812 Sorry for my dumb question but did you check the firmware version?

HendrikRoth, 2017-12-11:
Operating System: MacOS 10.13
Python Version: 3.6.3
Sonoff model: Sonoff Basic firmware 1.6.0

Same issue.

HendrikRoth, 2017-12-11:
Sonoff Version 1.5.5 also.

neuman1812, 2017-12-11:
@petjek   Yes, as stated in my comment.,  The firmware version on sonoff 1.5.5 allowed me to run the script.  The Sonoff that was running 1.6.0  did not allow me to finish the script.

adam-goddard, 2017-12-12:
I have another Sonoff Basic that hasn't been configured via eWelink so should be on v1.5.5, another Sonoff Basic on v1.6.0 and a Sonoff Dual R2 on v2.0.1 that are both configured on eWelink. I'll try them all tonight and let you know.

petjek, 2017-12-12:
OK, finally I figured out that all my devices are on 1.5.5. 4x Sonoff Basic ans 1x Sonoff Pow. None of them ever reached FinalStage.

sillyfrog, 2017-12-14:
@petjek A Basic running v1.5.5 is known to work. Given it has failed on everything, it looks like there is a firewall blocking inbound traffic (this is very common on Windows). If you are still having issues, please attach the logs as well.

petjek, 2017-12-14:
@sillyfrog I did set my firewall off but made another mistake. I entered credentials of my 5GHz-Wifi first that is not supported by Sonoff devices. I changed that to a 2.4GHz-Wifi and first it seems to work. The binary file was uploaded and right when upload was finished I could hear the Basic klick 4 times. After that the upload started again. And thats where I'm stuck now. Every minute I hear a click-clack click-clack and the upload starts again. See attached logfiles.
Btw I'm on MacOS, not Windows.
[debug_1513234010.log](https://github.com/mirko/SonOTA/files/1558207/debug_1513234010.log)
[debug_1513234139.log](https://github.com/mirko/SonOTA/files/1558208/debug_1513234139.log)

Glotzon, 2017-12-15:
Same issue here:
Operating System: Archlinux
Python Version: 3.6.3
Sonoff model: Sonoff Basic and Sonoff Touch

No firewall involved and stage 3 ended with error code 0.
After waiting > 5min and a power off/on of the device, still no Finalstage SSID.

sillyfrog, 2017-12-17:
@petjek In that latest log it looks like it is dropping off the network during the download, which is weird to be doing it so many times.

I would suggest rebooting everything (leaving the Sonoff disconnected for a full 60 seconds), then try again. If that does not work, try again with `--slowstream` (although I'm not sure that will help in this instance).

Glotzon, 2017-12-17:
Is it possible the issue is triggered by the minimum install image which is included in SonOTA?

After stage3 completed with error code 0, on my switches no finalstage came up. 
Connecting one switch to my PC, using my UART and starting a serial monitor I saw the switch trying to connect to "indebuurt1", "indebuurt2" respectively, as we would expect from any fresh tasmota install. Aparently the minimum base image was installed correctly. 

As soon as you provide the desired wifi connection you are able to connect to the webinterface of the mininum base image and flash your final image from there. 

My conclusion would be, and please call me out if I'm completely off track here, the minimum base image has the issue of not opening "finalstage" at all, maybe currently there is an incorrect base image being pushed?

sillyfrog, 2017-12-19:
@Glotzon Sounds like your install completed all the way through automatically (this can happen if you are lucky and your PC already knows and the `FinalStage` SSID). The "FinalStage" image is very very simple, and has no indebuurtX SSID's etc, it's goal is to get things into such a state that it can rewrite the boot loader, and install an Arduino image. See the actual source code here: https://github.com/sillyfrog/Espressif2Arduino/tree/master/Espressif2Arduino

Glotzon, 2017-12-19:
Yes I agree, hence my thought the minimum image might be lacking something. 
Let me start reading the code and see if I have a glaw in my logic on how the process works, as due to step 4 in the documentation I understood the flash of the final firmware to happen automatically, as I was left with the minimum version only.

LMey, 2017-12-31:
Same Issue here. 
My Output looks like this

`Lucass-MBP:SonOTA-master meyer_lucas$ ./sonota.py 
Current IPs: ['192.168.2.105']
Select IP address of the WiFi interface:
    0: 192.168.2.105
Select IP address [0]: 0
WiFi SSID: WLAN-Home
WiFi Password: 5p!d3rmaN

Using the following configuration:
	Server IP Address: 192.168.2.105
	WiFi SSID: WLAN-Home
	WiFi Password: *********
Platform: darwin
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
.....Current IPs: ['10.10.7.3']
~~ Connection attempt
>> HTTP GET /10.10.7.1/device
<< {
    "deviceid": "100012d9a0",
    "apikey": "c7faec82-6148-4040-9945-b05e686ee5bc",
    "accept": "post"
}
>> HTTP POST /10.10.7.1/ap
>> {
    "version": 4,
    "ssid": "WLAN-Home",
    "password": "*********",
    "serverName": "192.168.2.105",
    "port": 8443
}
<< {
    "error": 0
}
~~ Provisioning completed
Starting stage2...
** The IP address of <serve_host> (192.168.2.105) is not assigned to any interface on this machine.
** Please change WiFi network to WLAN-Home and make sure 192.168.2.105 is being assigned to your WiFi interface.
** This application should be kept running and will wait until connected to the WiFi...
....Current IPs: []
...Current IPs: ['192.168.2.105']
~~ Starting web server (HTTP port: 8080, HTTPS port 8443)
~~ Waiting for device to connect

*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................
*** IMPORTANT! ***
** AFTER the first download is COMPLETE, with in a minute or so you should connect to the new SSID "FinalStage" to finish the process.
** ONLY disconnect when the new "FinalStage" SSID is visible as an available WiFi network.
This server should automatically be allocated the IP address: 192.168.4.2.
If you have successfully connected to "FinalStage" and this is not the IP Address you were allocated, please ensure no other device has connected, and reboot your Sonoff.
..............................`

No "FinalStage" SSID comes up :-/ 

I am working on an Mac and trying to flash a Sonoff POW. 
What I acctually didn't tryied is to replace the static/image_arduino.bin with a newer one... no idea if this might help?!

If anybody can help, please let me know :)

flaviodexter, 2018-01-13:
same issue here:
win10
sonoff pow 2.0.4 firmware
[debug_1515861048.log](https://github.com/mirko/SonOTA/files/1628525/debug_1515861048.log)
wireshark shows tcp traffic between my laptop (192.168.2.153) anche the pow (192.168.2.154) but no final stage appears.

I've tryed both wothout parameters and with --legacy --slowstream with no luck, hope someoune can help me because I'm stuck with my openhab progect if I cannot use mqtt on sonoff devices

tks

3ricj, 2018-03-18:
Hi there, I just wanted to report, that my sonoff S31 firmware version "1.0.0" as reported by eweklink never reaches the final stage ESSID.  No errors, the logs look the same as folks report above.

alfem, 2018-09-07:
Same here. Second "bricked?" sonoff basic, stuck waiting for Final Stage.

[flash-failed.log](https://github.com/mirko/SonOTA/files/2361905/flash-failed.log)

And no more led blinking either. I hope I can flash it using the 'hard' way.

sillyfrog, 2018-09-09:
@alfem If you get a chance to monitor the Serial port before you flash it, it _may_ give some indication as to what state it's in. Maybe useful for debugging (not anything else that can be done unfortunately). I have pushed some minor updates recently that try to address one of the potential issues we have come across (power getting cut during the flashing).

alfem, 2018-09-10:
No, @sillyfrog, no wifi ssid comes up. I haved power cycled this sonoff a dozen times, but I can not see any life signals (led, wireless activity in my router...) 
 
I think I will forget the SonOffs for a while, and buy a couple of  [ElectroDragon hacker-friendly Relay boards](https://www.electrodragon.com/product/wifi-iot-relay-board-based-esp8266/)

tv09, 2018-10-20:
Hi,

My situation also never get to the "FINAL STAGE" either so after restarted and try to go through the procedure again SONOTA.EXE exit as soon as I enter the Wifi and password info.   Here is the log : 
-------------------------------------------------
2018-10-19 18:30:02,401: DEBUG: Current IPs: ['10.10.7.2']
2018-10-19 18:30:03,704: CRITICAL: It looks like you are already on a Sonoff/Final stage WiFi network, please change to your normal WiFi network. If this has already been done, you may need to modify the IP range of your LAN to use this tool safely. If this is what you intended run again with --no-prov to skip this step.
2018-10-19 18:30:03,705: INFO: Quitting.
-------------------------------------------------
Any pointer to the solution is greatly appreciated.  Thanks.

TV.

tchatard, 2018-11-10:
Hi
I have tried SonOTA.exe on Win10 with Sonoff S31 with 3 firmware versions: 1.0.0, 2.6.0 and 2.7.0.
In all cases I get the same behavior where the provisioning step works fine but the Sonoff never issues any HTTP request to retrieve the firmware from my laptop (stage 2). I know the SonOTA webserver is running and accessible because I see the messages printed on the console when I wget any URL from it, including from a different machine (to eliminate any possibility of firewall interfering). So, same thing, the Sonoff never reaches the FinalStage AP .
Has anyone been successful with a Sonoff S31?
Thanks

pabatribick, 2018-11-21:
same problem here. only see dots and a few minutes later this info messages in a loop. No WiFi to find to connect... That's Bad. I've wasted today 4 hours of my life... -.-

closed 2019-05-03 by MTrab